hbldh/bleak #491: BlueZDBus: AttributeError on task callback

devbis, 2021-03-22:
* bleak version: 0.11.0
* Python version: 3.7.9
* Operating System: OpenWrt
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

After some time working, bleak can raise AttributeError:
```
ERROR:asyncio:Exception in callback BleakClientBlueZDBus._parse_msg.<locals>.<lambda>(<Task finishe...> result=None>) at /usr/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py:1045
handle: <Handle BleakClientBlueZDBus._parse_msg.<locals>.<lambda>(<Task finishe...> result=None>) at /usr/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py:1045>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/events.py", line 88, in _run
  File "/usr/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 1045, in <lambda>
    lambda _: self._disconnecting_event.set()
AttributeError: 'NoneType' object has no attribute 'set'
```

Could you please consider moving check for `self._disconnecting_event` in the callback? It seems that it becomes None before callback is called.
https://github.com/hbldh/bleak/blob/develop/bleak/backends/bluezdbus/client.py#L1045

hbldh, 2021-03-23:
@devbis You did not try to disconnect here? It just happened?

In that case it seems like your device disconnected from BlueZ and BlueZ sent Bleak sent an update regarding this disconnection.
From what I can see, there are two possible solutions here:

1. Set `self._disconnecting_event = asyncio.Event()` in `__init__` or `connect`
2. Check event existence at line 1045:
```
if self._disconnecting_event:
    task.add_done_callback(
        lambda _: self._disconnecting_event.set()
    )
```
@dlech Thoughts? The second one is more correct, right? Since the `_disconnecting_event ` is used to complete the called disconnection procedure?

dlech, 2021-03-23:
I was thinking of something more like this:

```python
disconnecting_event = self._disconnecting_event
if disconnecting_event:
    task.add_done_callback(
        lambda _: disconnecting_event.set()
    )
```

This way, if `self._disconnect_event` is set to `None` or a new event before the done callback is called, it is essentially a no-op instead of potentially setting the wrong event.

devbis, 2021-03-24:
> You did not try to disconnect here? It just happened?

My app connects and disconnects from various devices in loops, so highly likely this error occurs on disconnect. But I cannot provide a sample that reproduces this error.

A similar `AttributeError` occurs in https://github.com/hbldh/bleak/blob/develop/bleak/backends/bluezdbus/client.py#L353
```
AttributeError: 'NoneType' object has no attribute 'wait'
```
But maybe it is better to create one more issue.

dlech, 2021-03-24:
> But maybe it is better to create one more issue.

Yes, that looks like a separate issue.

commit ea298f0b (2021-03-31): Async methods for scanner detection callbacks (#497)

Async methods for scanner detection callbacks
Also made change to fix #491.
closed 2021-03-31 by hbldh