hbldh/bleak #506: Add option to not enumerate all characteristcs for all serices on Windows

dlech, 2021-04-03:
* bleak version: 0.11.0
* Python version: 3.8.x
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: N/A

### Description

> Describe what you were trying to get done.

Trying to connect to LEGO Technic hub bootloader using Bleak.

> Tell us what happened, what went wrong, and what you expected to happen.

This is described in more detail at https://github.com/pybricks/pybricksdev/issues/15

The short story is that we get an AccessDenied error from `GetCharacteristicsAsync()` due to the fact that the device claims that it has a Service Changed characteristic that supports indications. Apparently Windows sees this and tries to automatically enable indications for this characteristic. But the device replies with an error.

This device works with Web Bluetooth on Windows, so I'm hoping that there is something in Bleak that can be tweaked.

For example, we might be able to use `GetGattServicesForUuidAsync()` and `GetCharacteristicsForUuidAsync()` instead of  `GetGattServicesAsync()` and `GetCharacteristicsAsync()` to only enumerate the services and characteristics we are actually interested in.

Or we could separate out discovering characteristics from discovering services. But this would mean that this would have to be manually called by users later.

Actually, I suppose we could do all of this and keep things backwards compatible by adding a keyword argument to connect (and the object constructors for use with async with) to disable scanning for services/characteristics on connect. Then add new API to get services that takes an optional list of UUIDs as an argument. Likewise, APIs would be needed to get characteristics and descriptors with optional UUIDs. Windows and Mac have OS APIs for this, but we would probably just have to fake it on BlueZ.

dlech, 2021-04-06:
It looks like rebooting fixed my problem since I can no longer reproduce the issue today.

dlech, 2021-05-16:
This happened again. I think this is the way it happened. I was using the Chrome web browser to connect to a device. Then disconnected it in the browser. Then I tried to connect to the same device with Bleak and got the AccessDenied error. After closing the web browser, I was able to connect to the device using Bleak without rebooting the computer. So, I think that the web browser was still holding on to a handle to the device somehow even though the device was no longer connected.

Also, [this chromium bug](https://bugs.chromium.org/p/chromium/issues/detail?id=1058849&q=bluetooth%20accessdenied&can=2) mentions a similar problem. In particular, it sounds like HID services will cause an AccessDenied error since they are used by the OS. It looks like they were able to work around the issue to some extent.

hbldh, 2021-05-18:
It does not seem to be anything Bleak can do to remedy this, is there? Would the "Get already connected devices" functionality help here? Is it registered as connected in Windows Bluetooth, or is it locked in some other way?

dlech, 2021-05-18:
I think the original problem I was encountering might be a Chromium bug that it doesn't release the Bluetooth device when it is no longer being used without closing all browser windows. However, in the course of researching the issue, I found the bug that I just linked above that describes the same error that can be caused by other means. 

So, I've changed the intent of this issue to be "maybe we can work around the AccessDenied error in some cases but probably not all cases".

I don't think getting connected devices would help in the case of the original issue since the device is disconnected but Chromium still seems to have a handle on it.