hbldh/bleak #507: Bleak error for kernel 5.4.0-70

mitulty, 2021-04-04:
* bleak version: 0.11.0
* Python version: python 3.8.5
* Operating System: ubuntu 20.04 with linux kernel 5.4.0-70-generic
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.53

### Description
I am trying to connect the esp32 module to the laptop using ble. The code for esp32 is as follows:
`
#include <BLE2902.h>
#include <Arduino.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLEDevice.h>

#define LED 2
#define BUTTON 0
#define SERVICE_UUID "---------------------------------------------"
#define LED_CHARACTERISTIC_UUID "---------------------------------------------"
#define BUTTON_CHARACTERISTIC_UUID "---------------------------------------------"

BLECharacteristic *ledCharacteristic;
BLECharacteristic *buttonCharacteristic;

bool deviceConnected = false;
volatile int buttonState = HIGH;

void pin_ISR();

class ServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      Serial.println("Central connected XD");
      deviceConnected = true;
    };
 
    void onDisconnect(BLEServer* pServer) {
      Serial.println("Central dis-connected :(");
      deviceConnected = false;
    }
};

class ControlSwitch: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic* pCharacteristic) {
      std::string value = pCharacteristic->getValue();
      Serial.printf("Received '%s'\n", value.c_str());
      if (value.length() > 1) {
        Serial.println("Value should be 0 or 1");
        return;
      }
      long switchState = std::strtol(value.c_str(), NULL, 10);
      if (switchState == 0l) {
        digitalWrite(LED, LOW);
      }
      else {
        digitalWrite(LED, HIGH);
      }
    }
};

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  // set up pin
  pinMode(LED, OUTPUT);
  pinMode(BUTTON, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(BUTTON), pin_ISR, CHANGE);

  // set up ble
  BLEDevice::init("esp32-two-way");
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new ServerCallbacks());

  BLEService *lightSwitchService = pServer->createService(SERVICE_UUID);
  ledCharacteristic = lightSwitchService->createCharacteristic(
                            LED_CHARACTERISTIC_UUID,
                            BLECharacteristic::PROPERTY_READ |
                            BLECharacteristic::PROPERTY_WRITE
                          );
  BLEDescriptor* descriptor = new BLEDescriptor(BLEUUID((uint16_t) 0x2901));
  descriptor->setValue("Set LED");
  ledCharacteristic->addDescriptor(descriptor);
  ledCharacteristic->setCallbacks(new ControlSwitch());
  ledCharacteristic->addDescriptor(new BLE2902());

  buttonCharacteristic = lightSwitchService->createCharacteristic(
                          BUTTON_CHARACTERISTIC_UUID,
                          BLECharacteristic::PROPERTY_NOTIFY
                        );
  // client charactersitic descriptor: required for notify
  buttonCharacteristic->addDescriptor(new BLE2902());
  lightSwitchService->start();

  // off initially
  ledCharacteristic->setValue("0");
  buttonCharacteristic->setValue("0");

  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(false);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  delay(300);
  if (buttonState == LOW) {
    Serial.println("Button pressed!");
    if (buttonCharacteristic->getValue() == "0") {
      // Serial.println("Button pressed!");
      buttonCharacteristic->setValue("1");
      buttonCharacteristic->notify();
    }
    else {
      buttonCharacteristic->setValue("0");
      buttonCharacteristic->notify();
    }
  }
}

void pin_ISR() {
  buttonState = digitalRead(BUTTON);
}
`

The code for python script on laptop is as follows:

`
import asyncio
import platform

from bleak import BleakClient

LED_UUID = "---------------------------------------"
BUTTON_UUID = "------------------------------------"

address = (
    "----------------------"  # <--- Change to your device's address here if you are using Windows or Linux
    if platform.system() != "Darwin"
    else "----------------------------"  # <--- Change to your device's address here if you are using macOS
)

async def notify(client):
    def keypress_handler(sender, data):
        print(f"BUTTON: {sender} {data}")
    await client.start_notify(BUTTON_UUID, keypress_handler)
    while True:
        await asyncio.sleep(1)

async def run(client):
    while True:
        switchState = await client.read_gatt_char(LED_UUID)
        switchState = switchState.decode('utf-8')
        if switchState == "0":
            value = bytearray(b'1')
            # value = value.extend(map(ord, "1"))
        else:
            value = bytearray(b'0')
            # value = value.extend(map(ord, "0"))
        await client.write_gatt_char(LED_UUID, value)
        print(f'LED {value}')
        await asyncio.sleep(4.0)

async def main(address):
    async with BleakClient(address) as client:
        await asyncio.gather(notify(client), run(client))

if __name__ == "__main__":
    asyncio.run(main(address))
`

Both the codes are running fine on ubuntu 18.04 and kernel 5.4.0-52. But in my system after some time(18sec to 20 sec) the laptop automatically disconnects from the esp32 and gives the following exception:

raise Bleakerror("Not Connected")
bleak.exec.Bleakerror: Not Connected

I have tried changing the kernel parameters related to timeout as mentioned in other issues but nothing worked.

dlech, 2021-04-05:
I would suspect that this doesn't have anything to do with Bleak. BlueZ will disconnect if the remote device sends something unexpected, for example I have seen this if you do a write without response from Bleak and the remote device sends a response anyway.

mitulty, 2021-04-05:
The code is running in other distros(ubuntu 18 kernel 5.4.0-52) very well. No timeout and no connection lost even for 45 mins.

hbldh, 2021-04-05:
It really does not sound like this is a bug in Bleak. There is nothing I can change to improve things for you.

If it works in older setups than your current one, then it probably means something has changed in BlueZ or Ubuntu.

dlech, 2021-04-05:
Is this all on the same hardware?

mitulty, 2021-04-06:
Yes. The code is the same for all the systems I tried with. It works for some time and then disconnects. Only thing I can relate it is to the kernel image version. Worked on one but not on others.

dlech, 2021-04-06:
If the kernel version is really the only difference (not a different Bluetooth chip?), then I would suggest filing a bug with Ubuntu.

mitulty, 2021-04-07:
Thank you. Will see to it.

closed 2021-04-07 by mitulty