hbldh/bleak #512: bleak.exc.BleakDBusError: org.bluez.Error.NotPermitted

iot49, 2021-04-11:
* bleak version: 0.11.0
* Python version: 3.7.3
* Operating System: Linux pi4server 5.10.17-v7l+ #1403
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

Running example `examples/uart_service.py`. Receiving works, but calling send throws an error.

Both sending and receiving work fine using an app (Adafruit bluefruit connect), hence I know that the peripheral is working.

### What I Did

Code is example from this site: `examples/uart_service.py`. 

It is also preferable if that example is not dependent on a specific peripheral device, but can be run and
reproduced with other BLE peripherals as well (I've wrapped`client.write_gatt_char` in the example code in a `try` block to get this error message).

```
Traceback (most recent call last):
  File "<ipython-input-8-365825f9a599>", line 61, in uart_terminal
    await client.write_gatt_char(UART_RX_CHAR_UUID, b"abc\n")
  File "/home/pi/iot49/.direnv/python-3.7.3/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 812, in write_gatt_char
    assert_reply(reply)
  File "/home/pi/iot49/.direnv/python-3.7.3/lib/python3.7/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name)
bleak.exc.BleakDBusError: org.bluez.Error.NotPermitted
```

As already mentioned, connecting to and receiving data from the peripheral work fine.

JacobMaldonado, 2021-04-14:
Hello, I had a similar issue trying to write characters over GATT. 
Updating bluez to 5.51 solved most of my problems.
I tried diferent versions 5.47, 5.50, 5.56 5.58 and for some reason 5.51 was the one that worked like a charm.
Hope this helps.

angel-ricardo, 2021-04-15:
> Hello, I had a similar issue trying to write characters over GATT.
> Updating bluez to 5.51 solved most of my problems.
> I tried diferent versions 5.47, 5.50, 5.56 5.58 and for some reason 5.51 was the one that worked like a charm.
> Hope this helps.

Thx, It works!

iot49, 2021-04-18:
I confirm that updating bluez to 5.51 solved my problem.

closed 2021-04-18 by iot49

dlech, 2021-04-19:
If versions newer than 5.51 don't work, then I would suggest reporting the regression to BlueZ.