hbldh/bleak #526: How can I activate notify if both notify and indicate are in the characteristic at the same time?

chandong83, 2021-04-24:
* bleak version: 0.11.0
* Python version: 3.7.8
* Operating System: windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

i'm using bleak to control BLE(bluetooth low energy) devices in Python. 
and i was trying to control a BLE device that included notify and indicate properties in one characteristic, 
so there was a problem that only indicate is activated.

so when i checked the bleak source code, it was found that if indicate exists, activate it as indicate, and if not, check notify and activate it as notify, and start_notify works.

the following is the relevant source code and link.

https://github.com/hbldh/bleak/blob/develop/bleak/backends/winrt/client.py

        async def start_notify(
                self,
                char_specifier: Union[BleakGATTCharacteristic, int, str, uuid.UUID],
                callback: Callable[[int, bytearray], None],
                **kwargs
            ) -> None:
                ...
                ...
                characteristic_obj = characteristic.obj
                if (
                    characteristic_obj.characteristic_properties
                    & GattCharacteristicProperties.INDICATE
                ):
                    cccd = GattClientCharacteristicConfigurationDescriptorValue.INDICATE
                elif (
                    characteristic_obj.characteristic_properties
                    & GattCharacteristicProperties.NOTIFY
                ):
                    cccd = GattClientCharacteristicConfigurationDescriptorValue.NOTIFY
                else:
                    cccd = GattClientCharacteristicConfigurationDescriptorValue.NONE
        
                fcn = _notification_wrapper(bleak_callback, asyncio.get_event_loop())
                event_handler_token = characteristic_obj.add_value_changed(fcn)
                self._notification_callbacks[characteristic.handle] = event_handler_token
                status = await characteristic_obj.write_client_characteristic_configuration_descriptor_async(
                    cccd
                )
        

### What I Did

what is the best way to deal with it in this case? 
so i implemented the code below to use "notify", but it doesn't look good.

        async with BleakClient(address, timeout=5.0) as client:
            client.set_disconnected_callback(on_disconnect)                        
            services = await client.get_services()        
            for service in services:
                for characteristic in service.characteristics:
                    if characteristic.uuid == read_write_charcteristic_uuid:  
                        if 'notify' in characteristic.properties:
                            # if "indicate" is exist remove it
                            if 'indicate' in characteristic.properties:
                                characteristic.properties.remove('indicate')
                            # then "notify" becomes active.
                            await client.start_notify(notify_charcteristic, callback)

thank you

dlech, 2021-04-24:
Apple doesn't allow selecting notifications or indications separately and doesn't specify what happens if a characteristic allows both. 

- https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518949-setnotifyvalue?language=objc
- https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/PerformingCommonCentralRoleTasks/PerformingCommonCentralRoleTasks.html

BlueZ is similar in that the D-Bus API doesn't allow specifying which to enable. But since it is open source, we can see that it will prefer notifications over indications.

- https://github.com/bluez/bluez/blob/7c3ca2a6b940d36c553fabe38066fabc66530dc9/src/shared/gatt-client.c#L1594-L1602

So it would be nice if someone could test the behavior on Apple to see what it does. But I would suggest change the Windows backend to prefer notifications over indications just like BlueZ. My experience has been that indications aren't really used much and tend to have long timeouts. (I would be interested to know if others have had a different experience.)

chandong83, 2021-04-24:
@dlech Thank you for answer.

I didn't doubt it because on android, esp32 and nrf52, i can choose notify and indicate.

thanks for the good information.
and thanks to you, I learned a lot.

hbldh, 2021-04-24:
The reason I chose indication over notification is because that is the way it is done in the MS UWP samples: https://github.com/microsoft/Windows-universal-samples/blob/b1cb20f191d3fd99ce89df50c5b7d1a6e2382c01/Samples/BluetoothLE/cs/Scenario2_Client.xaml.cs#L382

It seems like a `force_notify` keyword in the Windows backend `start_notify` might be called for.

chandong83, 2021-04-25:
@hbldh 

Thank you for answer.

Yes I understand what you mean (for window sample code). But I didn't understand the content well below it.

"It seems like a force_notify keyword in the Windows backend start_notify might be called for."

Does that mean that I can directly add the force_notify keyword in the library to change notify first? Or is there another way?

Thank you

commit c3afa7f1 (2021-08-10): Added `force_notify` keyword argument for WinRT backend client's `start_notify` method.

Fixes #526
commit 6c958462 (2021-08-12): Added `force_indicate` keyword argument for WinRT backend client's `start_notify` method.

Fixes #526
PR #620 (2021-10-06): Added keyword argument option to force using Notifications instead of Indications in WinRT backend
Added `force_notify` keyword argument for WinRT backend client's `start_notify` method.

Fixes #526
closed 2021-10-06 by dlech