hbldh/bleak #532: Wrong error message printed for ATT error codes

rtoijala, 2021-05-05:
* bleak version: 0.10.0
* Python version: 3.8.1
* Operating System: Windows 10 20H2

### Description

When a GATT server replies to a read or write of a characteristic with an ATT error code, the wrong string is printed for that code. Other GATT operations may also be affected, based on a quick look at the code.

Example
```
await client.write_gatt_char('eeeeeeee-aaaa-bbbb-cccc-dddddddddddd', b'foo', response=True)
```
results in
```
bleak.exc.BleakError: Could not write value b'foo' to characteristic eeeeeeee-aaaa-bbbb-cccc-dddddddddddd: ProtocolError (Error: 0x07: Memory Capacity Exceeded)
```
The correct error string would be e.g. "Invalid Offset" instead of "Memory Capacity Exceeded".

The root cause is using the error strings for Bluetooth controllers instead of the error string for ATT errors in `https://github.com/hbldh/bleak/blob/develop/bleak/backends/dotnet/client.py#L613`. The correct errors are those listed in `https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattprotocolerror`, though I am not sure whether there is a built-in way to get the strings from those. See also Bluetooth Core Specification v5.2, volume 3, part F, section 3.4.1.1.

dlech, 2021-05-05:
I think we will just have to make another lookup table like `CONTROLLER_ERROR_CODES`.

dlech, 2021-05-05:
Also related: #527 

It could be useful to add the error code numeric value as an argument to the exceptions so we don't have to match on strings that are just meant to be human readable and may change (e.g to fix a typo).

commit a8fe0153 (2021-10-06): winrt: fix protocol error descriptions

The wrong set of error messages was being used for protocol errors in
the WinRT backend. This adds a new lookup table with the correct short
error descriptions.

Fixes #532.