hbldh/bleak #535: Program disconnects before connection established. Locks up Bleak

Dallin-Lloyd, 2021-05-12:
* bleak version: 0.11.0
* Python version: 3.9
* Operating System: MacOS Big Sir
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Describe what you were trying to get done.
Trying to connect and disconnect from an array of devices. trying to connect and then disconnect to make sure devices under test are working properly. I am iterating through the devices and cycling the connect/disconnect N number of times. 
(devices have firmware that sends device to sleep. sometimes wont wake up)

Tell us what happened, what went wrong, and what you expected to happen.
Program would freeze after 20-30 connect / disconnect cycles. 

### What I Did
i added print statements to client.py and scanner.py to monitor in the terminal where it was freezing.
This was to help me visualize. It appears that sometimes the disconnect callback gets called and locks up the program. its call back to disconnect before it has established a connection.

It is preferable if an issue contains a [Miminal Workable Example](https://stackoverflow.com/help/minimal-reproducible-example).
This will otherwise be one of the first questions you will get as a response.

It is also preferable if that example is not dependent on a specific peripheral device, but can be run and
reproduced with other BLE peripherals as well.

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here as well.
```
    def _disconnected_callback_client(self):
        """
        Callback for device disconnection. Bleak callback sends one argument as client. This is wrapper function
        that gets called from the CentralManager and call actual disconnected_callback by sending client as argument
        """
        logger.debug("Received disconnection callback...")
        print("received disconnect callback")
        if self._disconnected_callback is not None:
            self._disconnected_callback(self)

    async def disconnect(self) -> bool:
        """Disconnect from the peripheral device"""
        print("hit disconnect function")
        manager = self._central_manager_delegate
        if manager is None:
            return False
        await manager.disconnect()
        self.services = BleakGATTServiceCollection()
        # Ensure that `get_services` retrieves services again, rather than using the cached object
        self._services_resolved = False
        self._services = None
        return True

my output normally looks like this: 
await manager.connect
await self.get_services()
returning
connection successful with device: F53E1F1D-2DCE-44F1-ACEC-737FF821AE07
hit disconnect function
received disconnect callback
connection with device 47 was successful

When the program freezes, it looks like this:
await manager.connect
await self.get_services()
received disconnect callback

dlech, 2021-05-12:
Thanks for reporting the issue. It looks like we need to at least add a timeout while waiting for services to be enumerated. Even better would be to allow it to be interrupted by the disconnect event.

Dallin-Lloyd, 2021-05-26:
Are there any updates on this? is there anything i can do to help with this?

dlech, 2021-06-03:
Hi @Dallin-Lloyd, can you please test #557?

PR #564 (2021-06-08): Mac cleanup and fixes
This is a bit much all at once, but one thing led to another... and another...

Commit messages have more details, but the basic gist is:
- Improved type hints (includes adding typings for CoreBluetooth and other Obj-C APIs)
- Fixed a bunch of bugs
  - Fixes #535 
  - Fixes #538
  - And more
- Removed some unused and overly complex code
commit e61b821f (2021-06-08): Fixed deadlock in CoreBluetooth backend when device disconnects

This fixes #535. In the CoreBluetooth backend, when a device disconnects,
none of the peripheral delegate callbacks will be called. So if any
futures are pending, they will be waiting forever. This adds a disconnect
hook that will raise an exception in all pending futures to avoid this
deadlock.
closed 2021-06-08 by dlech