hbldh/bleak #551: Notifications from 2 devices are blocked by asyncio.sleep

dlech, 2021-05-25:
I would like to restart this issue again. I am unable to read notifications from two devices simultaneuously. I am running:

- bleak==0.11.0
- linux kernel  5.4.0-72-generic
- Linux Mint (Ubuntu 20.04)

The code I am running is given below based on Issue [345](https://github.com/hbldh/bleak/issues/345):
```
import asyncio
from bleak import BleakClient

erc1address = "80:EA:CA:70:00:01"
erc2address = "80:EA:CA:70:00:02"
read_characteristic = '16005991-b131-3396-014c-664c9867b917'

with open('file_one.csv','w') as f:
    f.write("ERC-1\n")

with open('file_two.csv','w') as f:
    f.write("ERC-2\n")

def callback_one(sender, data):
     with open('file_one.csv','a') as f:
         for i in range(len(data)):
             f.write(f'{data[i]}')
         
def callback_two(sender, data):
     with open('file_two.csv','a') as f:
         for i in range(len(data)):
             f.write(f'{data[i]}')

def run(addresses):
    loop = asyncio.get_event_loop()

    tasks = asyncio.gather(connect_one(addresses[0]) ,connect_two(addresses[1]))

    loop.run_until_complete(tasks)

async def connect_one(address):
    print("starting", address, "loop")
    async with BleakClient(address, timeout=20.0) as client:

        print("connect to", address)
        try:
            await client.start_notify(read_characteristic, callback_one)
            while 1:
                await asyncio.sleep(1)
        except Exception as e:
            if e is KeyboardInterrupt:
                await client.stop_notify(read_characteristic)

    print("disconnect from", address)

async def connect_two(address):
    print("starting", address, "loop")
    async with BleakClient(address, timeout=20.0) as client:
        print("connect to", address)
        try:
            await client.start_notify(read_characteristic, callback_two)
            while 1:
                await asyncio.sleep(1)
        except Exception as e:
            if e is KeyboardInterrupt:
                await client.stop_notify(read_characteristic)

    print("disconnect from", address)

if __name__ == "__main__":
    run(
        [erc1address, erc2address]
    )
```

I have `while 1: await sleep(1)` because I need to read notifications from both devices. If I replace that with just `await sleep(10)` for instance, then after 10 seconds, the second connected device notifications are read. However, my devices send notifications continuously, therefore I need to read data from both.

It would be great if any suggestions/help can be provided.

_Originally posted by @mabhijithn in https://github.com/hbldh/bleak/issues/215#issuecomment-847858482_

mabhijithn, 2021-05-26:
I was able to reproduce the issue on Windows 10 too. The notifications are read from the last device connected.

dlech, 2021-06-03:
I think part of the issue might be that the way the program is written - it will try to start two scanners at the same time. It would probably work better to use `BleakScanner` to find the devices separately. Also, the BlueZ version could be an issue. 

Here is a program that is working for me on BlueZ 5.53:

```python
import asyncio
from typing import Set

from bleak import BleakScanner, BleakClient
from bleak.backends.scanner import BLEDevice, AdvertisementData

PYBRICKS_SERVICE_UUID = "C5F50001-8280-46DA-89F4-6D8051E4AEEF"
PYBRICKS_CHAR_UUID = "C5F50002-8280-46DA-89F4-6D8051E4AEEF"

async def monitor(device: BLEDevice) -> None:
    """
    Monitors notifications from a BLE device until it disconnects.
    """
    disconnect_event = asyncio.Event()

    def handle_disconnect(client: BleakClient) -> None:
        print(f"disconnected from {client.address}")
        disconnect_event.set()

    def handle_notification(handle: int, value: bytearray) -> None:
        print(f"received {value} from {device.address}")

    async with BleakClient(device, disconnected_callback=handle_disconnect) as client:
        print(f"connected to {client.address}")
        await client.start_notify(PYBRICKS_CHAR_UUID, handle_notification)
        await disconnect_event.wait()

async def scan() -> None:
    """
    Scans for devices for 10 seconds and initiates a ``monitor`` for each
    discovered device.
    """

    # keep track of detected devices to avoid creating duplicate connections
    detected: Set[str] = set()

    def handle_detection(device: BLEDevice, adv: AdvertisementData) -> None:
        if PYBRICKS_SERVICE_UUID.lower() not in adv.service_uuids:
            return

        if device.address not in detected:
            detected.add(device.address)
            asyncio.create_task(monitor(device))

    async with BleakScanner(detection_callback=handle_detection):
        print("scanning...")
        await asyncio.sleep(10)
        print("done scanning.")

async def test() -> None:
    asyncio.create_task(scan())
    # wait forever
    await asyncio.Event().wait()

asyncio.run(test())
```

asymptotic, 2021-06-11:
I've been following this since I've been having the same issue. The above code didn't work for me, it only connected to one device. For reference, it turns out I was using BlueZ 5.50 (Raspberry Pi 3) and upgrading it manually to 5.58 made it work, although not with the above code.

Using `BleakScanner` as above gives me timeouts, even if I up the timeout value to something huge:

```
scanning...
done scanning.
Task exception was never retrieved
future: <Task finished coro=<monitor() done, defined at test_bleak.py:24> exception=TimeoutError()>
Traceback (most recent call last):
  File "test_bleak.py", line 37, in monitor
    async with BleakClient(device, disconnected_callback=handle_disconnect) as client:
  File "/root/.local/lib/python3.7/site-packages/bleak/backends/client.py", line 61, in __aenter__
    await self.connect()
  File "/root/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 294, in connect
    timeout,
  File "/usr/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
Task exception was never retrieved
future: <Task finished coro=<monitor() done, defined at test_bleak.py:24> exception=TimeoutError()>
Traceback (most recent call last):
  File "test_bleak.py", line 37, in monitor
    async with BleakClient(device, disconnected_callback=handle_disconnect) as client:
  File "/root/.local/lib/python3.7/site-packages/bleak/backends/client.py", line 61, in __aenter__
    await self.connect()
  File "/root/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 294, in connect
    timeout,
  File "/usr/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
^CTraceback (most recent call last):
  File "test_bleak.py", line 72, in <module>
    asyncio.run(test())
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 571, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 539, in run_forever
    self._run_once()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1739, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.7/selectors.py", line 468, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
```

However, changing this to the `discover()` method of finding devices like in the readme works ok and I'm able to get the notifications from them:

```
devices = await BleakScanner.discover()
    for d in devices:
        if d.name == "Target Device":
            log.info("[BLE Discovery] {}".format(d.address))
            asyncio.create_task(monitor(d))
```

```
[BLE Discovery] CE:AE:EC:13:F8:05
[BLE Discovery] DD:62:84:86:9B:34
```

Any thoughts on what's going wrong here?

dlech, 2021-06-11:
Interesting, it seems that BlueZ 5.58 is having trouble connecting while it is still scanning. I will try to reproduce when I have time.

asymptotic, 2021-06-11:
I was getting the same error with 5.50, which is what first made me think of upgrading. Thanks, and I will keep you updated if I figure it out!

closed 2022-07-26 by dlech