hbldh/bleak #556: AttributeError in WinRT scanner _stopped_handler() callback

devanlai, 2021-06-02:
* bleak version: v0.12.0b1 (sha: ce63ed4)
* Python version: 3.9.5
* Operating System: Windows 10 20H2 (OS Build 19042.985)

### Description

I don't have a minimal example that reproduces the bug, but it appears that when scanning with the WinRT backend using `bleak.discover()`, I very rarely get exceptions in the scanner watcher stop callback:

```
Exception in callback BleakScannerWinRT._stopped_handler(<_winrt_Windo...0012FDA26F590>, <_winrt_Windo...0012FDA26F5B0>)

handle: <Handle BleakScannerWinRT._stopped_handler(<_winrt_Windo...0012FDA26F590>, <_winrt_Windo...0012FDA26F5B0>)>
Traceback (most recent call last):
  File "asyncio\events.py", line 80, in _run
  File "bleak\backends\winrt\scanner.py", line 118, in _stopped_handler
AttributeError: 'NoneType' object has no attribute 'status'
```

The `_stopped_handler()` callback that triggers the exception:
```python3
def _stopped_handler(self, sender, e):
    logger.debug(
        "{0} devices found. Watcher status: {1}.".format(
            len(self._devices), self.watcher.status
        )
    )
```

The `stop()` method that I think has a race condition:
```python3
async def stop(self):
    self.watcher.stop()

    try:
        self.watcher.remove_received(self._received_token)
        self.watcher.remove_stopped(self._stopped_token)
    except Exception as e:
        logger.debug("Could not remove event handlers: {0}...".format(e))

    self._stopped_token = None
    self._received_token = None

    self.watcher = None
```

I'm speculating that there might be a race condition where:
1. The scanner calls `stop()` to stop the watcher as part of its cleanup after the call to `bleak.discover()` returns.
2. The watcher stopped callback fires after we call `self.watcher.stop()` but before we remove the stopped callback, since it seems to run in a different thread.
3. The callback schedules our `_stopped_handler()` callback to run in our main thread/event loop via `call_soon_threadsafe()`.
4. Our main thread resumes and removes the callbacks successfully and sets `self.watcher = None`
5. `stop()` returns and eventually our `_stopped_handler()` callback is scheduled to run
6. `_stopped_handler()` attempts to access `self.watcher`, which is now None, triggering the exception.

From a quick attempt at inspecting the sender, it looks like one workaround would be for the `_stopped_handler()` callback to assume `sender` is always the watcher object and extract the status from the sender directly, rather than reading it from `self.watcher`. But I'm not intimately familiar with the WinRT callbacks dispatch, so I'm not sure if that always works.

devanlai, 2021-06-03:
By inserting a `time.sleep(1.0)` call in `stop()` in-between the call to `self.watcher.stop()` and `self.watcher.remove_stopped()`, I can reliably trigger this with `bleak-lescan`, so that seems to confirm that it's a race condition.

I think the best way to handle this would be for `_stopped_handler()` to check if `self.watcher is not None` before proceeding. I tried checking if `sender == self.watcher` like the dotnet backend does, but it seems like the two objects don't compare equal even when all of the critical values are equivalent (same status, same advertisement filter byte patterns / service UUIDs). I'm guessing the WinRT projection doesn't implement deep equality checks, just the default identity check.

Checking against None should handle the case where we've already manually stopped the scanner + watcher. It won't be as selective as the dotnet backend in the event that it's actually possible to receive a stopped event from some unrelated watcher object, but since it's only a log debug statement, I'm not too worried about that case.

dlech, 2021-06-03:
Since `stop()` is an async function, I think it would make sense to wait for the stopped event before removing the event handlers.

devanlai, 2021-06-03:
I'm not sure if that would always work. If the watcher stops outside of the call to `stop()` - which I'm assuming is the original situation the the handler wants to catch, since outside of this race condition it normally doesn't get called when you call `stop()` - then calling `watcher.stop()` inside of `stop()` might not generate a stopped event that we can wait on.

From testing, stopping the watcher, waiting for the stopped event, and then stopping the watcher again doesn't appear to generate a new stopped event.

hbldh, 2021-06-07:
I agree with @dlech, and I provided such an implementation in #571. @devanlai Try it out and see if it does what you need it to do.

devanlai, 2021-06-07:
Okay, now that I see how the asyncio stopped event would work in #571, I agree that's a better way to do it that covers the edge cases I was worried about.

PR #575 (2021-06-19): Release v0.12.0
If there aren't any glaring problems, I publish release tomorrow.

Shortlog
======

- New WinRT backend for Windows.
- Python 3.9 support.
- Overhaul of CoreBluetooth backend.
- Can now get MTU for connected devices.
- New, handy ``BleakScanner.find_device_by_filter()`` method.

Full Changelog
===========

Added
---------

* Added ``mtu_size`` property for clients.
* Added WinRT backend.
* Added ``BleakScanner.discovered_devices`` property.
* Added an event to await when stopping scanners in WinRT and pythonnet backends. Fixes #556.
* Added ``BleakScanner.find_device_by_filter` [... cut]
closed 2021-06-11 by hbldh