hbldh/bleak #559: CoreBluetooth: BleakClient(device_from_scan) fails with missing `address_or_ble_device.metadata["delegate"]`

jeffsf, 2021-06-03:
```
(venv) [jeff@mini ~/devel/pyDE1]$ python --version
Python 3.9.5
(venv) [jeff@mini ~/devel/pyDE1]$ pip list
Package                        Version
------------------------------ -------
aiologger                      0.6.1
bleak                          0.11.0
pip                            21.0.1
pyobjc-core                    7.1
pyobjc-framework-Cocoa         7.1
pyobjc-framework-CoreBluetooth 7.1
pyobjc-framework-libdispatch   7.1
setuptools                     56.0.0
```
macOS 10.14.6

### Description

Working toward being able to manage multiple devices without connectivity loss under Bluez (#361) and developing locally on my Mac, I found that trying to create a `BleakClient` from a `BLEDevice` returned from a scan failed. I expected that the `BLEDevice` passed to the scanner callback `is_skale(device: BLEDevice, adv: AdvertisementData)` would be able to be used in `skale = BleakClient(device)`. However, an error was reported under macOS (and not seen running Python 3.9.5 built for RPi Buster)
```
Traceback (most recent call last):
  File "/Volumes/miniup-users/jeff/devel/pyDE1/scale_direct.py", line 164, in is_skale
    skale = BleakClient(device)
  File "/Volumes/miniup-users/jeff/devel/pyDE1/venv/lib/python3.9/site-packages/bleak/backends/corebluetooth/client.py", line 51, in __init__
    self._central_manager_delegate = address_or_ble_device.metadata["delegate"]
KeyError: 'delegate'
```

### What I Did

Examination of the `device` using a debugger suggests that the error is "correct" and that the `device` is of the proper class and at least partially populated

[image]

Running the same code remotely on the RPi does not trigger the error, though also has an empty `metadata` dict.

[image]
```
(pyDE1_3.9) jeff@pi-walnut:~ $ python --version
Python 3.9.5
(pyDE1_3.9) jeff@pi-walnut:~ $ pip list
Package    Version
---------- -------
bleak      0.11.0
dbus-next  0.2.2
pip        21.1.2
setuptools 56.0.0
```

### Code To Replicate

Though tied to a specific device, I am more than willing to work to provide additional information as needed.

```
async def find_first_skale(timeout=5.0) -> BleakClient:

    scanner = BleakScanner(timeout=timeout)
    skale: Optional[BleakClient] = None
    found_skale = asyncio.Event()

    def is_skale(device: BLEDevice, adv: AdvertisementData):

        nonlocal skale

        if found_skale.is_set():
            logger.info(f"Skipping adv from {device} as already found skale")
            return
        if adv.local_name and adv.local_name.startswith('Skale'):
            found_skale.set()
            try:
                skale = BleakClient(device)
            except KeyError:
                logger.warning(
                    "Fallback to address on KeyError, "
                    "likely from metadata: {}\n{}".format(
                        device.metadata,
                        traceback.format_exc(),
                    ))
                skale = BleakClient(device.address)
            logger.info(f"Scale: {skale} at {skale.address}")

    scanner.register_detection_callback(is_skale)

    logger.info("Scan start")
    await scanner.start()

    logger.info("Waiting for Skale adv")
    try:
        await asyncio.wait_for(found_skale.wait(), timeout=timeout)
        logger.info("Found")
    except asyncio.TimeoutError:
        logger.info("Not found")

    await stop_scanner_if_running(scanner)
    return skale
```

In context:
[scale_direct.py.gz](https://github.com/hbldh/bleak/files/6593258/scale_direct.py.gz)

dlech, 2021-06-03:
Duplicate of #448?

jeffsf, 2021-06-03:
Closed as duplicate of #448

closed 2021-06-03 by jeffsf