hbldh/bleak #563: Philosophical question: are the type hints worth it?

jackjansen, 2021-06-04:
A question triggered by reading and trying to understand the bleak code.
Feel free to close if you don't think it is worth your time.

But my question is: are the type safety and early (as opposed to late) coding error detection that are possible because of the type hints used in bleak worth it, given that they make the code much more difficult to follow?

As an example, without the type hints and type checking `scanner.BaseBleakScanner.register_detection_callback` could be written as

```
    def register_detection_callback(self, callback):
        self._callback = callback
```

which is much more readable than the 13 lines it currently is (not counting the 10-line documentation).

And the various `__init__(self, **kwargs)` methods that cannot use keyword arguments but need to use `kwargs.get()` because of the type hints are another good example.

I have the feeling that the code would become much more readable and much more Pythonic if the type hinting was ripped out....

dlech, 2021-06-05:
I agree that the code would be much easier to read without things like arg checking and type hints. But I find when actually working on the code vs. just reading it, type hints boost my own productivity tremendously. I can just hover my mouse over an identifier and know what the intent was without having to backtrace through several function calls or having to refer to external documentation. So, for me, that outweighs the readability aspect.

As for the early failure vs. late failure, this only applies to the callback functions, like the ones for the advertisement data callback, the device disconnected callback and the characteristic notification callback. Due to async nature of this library and the fact that most users of Bleak don't seem to be experienced with that sort of program (or perhaps not very experienced with Python in general), I feel like having a clear error message right up front about what the user did wrong and how to fix it is more beneficial than keeping our internal code as simplistic as possible. When late failures happen with these callbacks, it is in a background task started by Bleak that is unconnected to the source of the problem, which can make it difficult to know where the problem was introduced.

jackjansen, 2021-06-05:
Fair enough. I think I sort-of agree with your reasoning too. I just raised the point for discussion, because to a Python old-timer like me the benefits don't seem to outweigh the disadvantages. But I'll stick to your conventions:-)

closed 2021-06-05 by jackjansen

bojanpotocnik, 2021-06-07:
> I agree that the code would be much easier to read without things like arg checking and type hints. But I find when actually working on the code vs. just reading it, type hints boost my own [...]

In addition to that, using advanced enough IDE (e.g. PyCharm) will use this info for code analysis and mark incorrect usages as errors while you type, eliminating most of the runtime errors normal user encounters. That also goes for properly handling/checking variables which can be `None` instead of e.g. empty list, marked as  `Optional[..]` or `Union[None, ..]` - similar to Java `@Nullable` annotation. I sometimes miss this in C for `NULL`.

In my opinion and experience, adding typing information where possible is very important - that it why it was so anticipated anyway, and that is why we also have e.g. TypeScript, Typed Lua,... looks like every language eventually tends to offer static typing. I count is as documentation - of course it is clearer and faster to write code without documentation... if you are the only developer and have no intentions to read the code ever again.

hbldh, 2021-06-07:
I agree with @dlech. I like the idea of typehints, they are in Python to stay and I think they actually add value in most libraries. They are in bleak to stay as well and given that we actually manage to set them correctly (which I have failed to do on multiple occasions) I envision them to be more helpful than not having anything. 

Given the multiple backends on different OS:es and their use of objects from other programming languages, it is a good thing to  have typehints to ensure that the internal types does not leak to the public API.