hbldh/bleak #576: Data upload from two device is not simultinuously

dinosauria123, 2021-06-19:
* bleak version: 0.11.0
* Python version: 3.7.3
* Operating System: Windows 10

### Description

I try to connect two M5StickC to PC via BLE to upload their sensor data.
Data upload from two devices are not simultaneously, the upload data is dominated first
connected M5StickC and secondly connected M5StickC data is little uploaded.

### What I Did

This script used to receive uploaded data from two M5StickC.
Both M5StickC has different UUID and address.  

```
from bleak import BleakClient
import asyncio

import bleak

address1 = "D8:A0:1D:55:EE:8A"
UUID1 = "beb5483e-36e1-4688-b7f5-ea07361b26a8"

address2 = "94:B9:7E:93:21:76"
UUID2 = "beb5483e-36e1-4688-b7f5-ea07361b26a2"

def run(addresses, UUIDs):
    loop = asyncio.get_event_loop()
    tasks = asyncio.gather(*(connect_to_device(address, UUID) for address, UUID in zip(addresses, UUIDs)))
    loop.run_until_complete(tasks)

async def connect_to_device(address, UUID):
    print("starting", address, "loop")
    try:
        async with BleakClient(address, timeout=5.0) as client:
            print("connected to", address)
            while(True):
                try:
                    print(await client.read_gatt_char(UUID))
                except Exception as e:
                    print(e)
    except bleak.exc.BleakError:
        pass

if __name__ == "__main__":
    run([address1, address2], [UUID1, UUID2])
```

dlech, 2021-06-19:
Interesting. I guess you will have to implement some kind of prioritization so that one device does not hog the asyncio event loop.

RIES-endo, 2021-06-20:
I think this may be help but I could not rewrite my script…

https://stackoverflow.com/questions/48361207/control-scheduling-priority-of-asyncio-coroutines-possible

RIES-endo, 2021-06-20:
I wrote this but upload data are never changed...

```
from bleak import BleakClient
import asyncio

address1 = "D8:A0:1D:55:EE:8A"
UUID1 = "beb5483e-36e1-4688-b7f5-ea07361b26a8"

address2 = "94:B9:7E:93:21:76"
UUID2 = "beb5483e-36e1-4688-b7f5-ea07361b26a2"

adresses = [address1, address2] 
UUIDs = [UUID1, UUID2]

def run(addresses, UUIDs):
    loop = asyncio.get_event_loop()
    task = asyncio.gather(*(connect_to_device(address, UUID) for address, UUID in zip(addresses, UUIDs)))
    loop.run_until_complete(task)

async def connect_to_device(address, UUID):
    print("starting", address, "loop")
    async with BleakClient(address, timeout=5.0) as client:
        print("connected to", address)
        tasks = {client.read_gatt_char(UUID)}
        while(True):
            done, not_done = await asyncio.wait(tasks, return_when=asyncio.FIRST_COMPLETED)
            for tasks in done:
                try:
                    print(await tasks)
                except Exception as e:
                    print(e)
                not_done.add(tasks)
            tasks = not_done

if __name__ == "__main__":
    run(adresses, UUIDs)

```

dlech, 2022-09-12:
In #970 this was shown to be a problem with the devices themselves (when to devices transmit at the same time, one can hog the connection), so closing.

closed 2022-09-12 by dlech