hbldh/bleak #579: Missing advertising data on Linux but working on Windows

FiachraBarry99, 2021-06-23:
* bleak version: 0.11.0
* Python version: 3.7.3
* Operating System: Windows 10/Raspberry Pi OS (Raspbian GNU/Linux 10 (buster))
* BlueZ version: 5.50

### Description

I am trying to use bleak to scan for advertising data from a Bluetooth peripheral. The advertising data in question is broadcast in the form of two bytes and is contained in the manufacturing data part of the advertising packet.  You can find the layout of the advertising data [here](https://usermanual.wiki/Document/TemperatureHumidityDataLoggerCommandsAPI24.2837071165/html) on page 6. I initially wrote the code on Windows where it worked fine. When I ran the exact same script on my Raspberry Pi it only picks up the second of the two bytes which is unfortunately the byte I am not interested in.

### What I Did

I included below the code that can be used to show the error. It is similar to the `detection_callback.py` example except filtered for only the manufacturer number I am looking for. I have tried this without the filter (similar to the original example) and I still cannot pick up the first byte. I am wondering if it is something to do with the fact it is broadcast as two bytes and not one. I have also included the output on Windows and the Pi to compare.

```python
import asyncio
from bleak import BleakScanner
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData
import logging

logging.basicConfig()

def simple_callback(device: BLEDevice, advertisement_data: AdvertisementData):
    data = advertisement_data.manufacturer_data.get(307)

    if not data:
        return
    else:
        print(data)

async def run():
    scanner = BleakScanner()
    scanner.register_detection_callback(simple_callback)

    while True:
        await scanner.start()
        await asyncio.sleep(5.0)
        await scanner.stop()

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```
#### Windows Output:
```
b'\x17d\x0e\x10\x12`\x01\x9e\x00\x84\x00P\x01\x00'
b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xbc\x00\x93\x01\x89\x00\x1d\x028\x00?\x00\x00\x00\x00\x00'
b'\x17d\x0e\x10\x12`\x01\x9e\x00\x84\x00P\x01\x00'
b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xbc\x00\x93\x01\x89\x00\x1d\x028\x00?\x00\x00\x00\x00\x00'
b'\x17d\x0e\x10\x12`\x01\x9e\x00\x84\x00P\x01\x00'
b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xbc\x00\x93\x01\x89\x00\x1d\x028\x00?\x00\x00\x00\x00\x00'
```
#### Raspberry Pi OS Output:
```
b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xbc\x00\x93\x01\x89\x00\x1d\x028\x00?\x00\x00\x00\x00\x00'
b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xbc\x00\x93\x01\x89\x00\x1d\x028\x00?\x00\x00\x00\x00\x00'
b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xbc\x00\x93\x01\x89\x00\x1d\x028\x00?\x00\x00\x00\x00\x00'
b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xbc\x00\x93\x01\x89\x00\x1d\x028\x00?\x00\x00\x00\x00\x00'
b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xbc\x00\x93\x01\x89\x00\x1d\x028\x00?\x00\x00\x00\x00\x00'
```
I also tried to connect to the BLE peripheral in question as a `BLEDevice` and looking in `BLEDevice.metadata` but it still only contained the second byte. Also, rather annoyingly, if I leave it to run for a very long time on the Pi, it will very occasionally pick up the first byte (I'm talking like once every half an hour and not predictably).

I don't think this is an issue with hardware as I have tested the Raspberry Pi with other advertising data sources and it works fine. It is also not a problem with the peripheral as the script runs perfectly on Windows.

dlech, 2021-06-23:
FYI, we have just released Bleak v0.12.0 (although I don't expect that to fix the issue).

I'm guessing that this device sends different manufacturer-specific data in both the advertising data packet and the scan response packet.  Windows treats these as totally separate events, which is why it works with no problem there.

Hopefully we just have a race condition in Bleak that we can fix instead of there being a problem in BlueZ. You can try enabling logging in Bleak to see if we are getting the correct information in the first place.

FiachraBarry99, 2021-06-23:
Yes, according to their docs the first byte is the standard advertising payload and the second is the scan response payload. I have included below the logging that I think is relevant to the BLE device I am trying to connect to. Unfortunately I cannot see any trace of the first byte (standard payload) only the second (scan response).

This is the initial relevant log:
```
2021-06-23 17:08:02,125 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_C0_3D_AA_ED_C6_39', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', C0:3D:AA:ED:C6:39)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Name': <dbus_next.signature.Variant ('s', C03DAAED)>, 'Alias': <dbus_next.signature.Variant ('s', C03DAAED)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -75)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', [])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {307: <dbus_next.signature.Variant ('ay', b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xb7\x00\x93\x01\x89\x00\x1f\x01\xf6\x00Z\x00\x00\x00\x00\x00')>})>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
DEBUG:bleak.backends.bluezdbus.scanner:received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_C0_3D_AA_ED_C6_39', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', C0:3D:AA:ED:C6:39)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Name': <dbus_next.signature.Variant ('s', C03DAAED)>, 'Alias': <dbus_next.signature.Variant ('s', C03DAAED)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -75)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', [])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {307: <dbus_next.signature.Variant ('ay', b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xb7\x00\x93\x01\x89\x00\x1f\x01\xf6\x00Z\x00\x00\x00\x00\x00')>})>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
```
This is the relevant log that appears every time after that:
```
2021-06-23 17:08:10,391 bleak.backends.bluezdbus.scanner DEBUG: cached devices: {'/org/bluez/hci0/dev_C0_3D_AA_ED_C6_39': {'Address': 'C0:3D:AA:ED:C6:39', 'AddressType': 'random', 'Name': 'C03DAAED', 'Alias': 'C03DAAED', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {307: b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xb7\x00\x93\x01\x89\x00\x1f\x01\xf6\x00Z\x00\x00\x00\x00\x00'}, 'ServicesResolved': False}}
DEBUG:bleak.backends.bluezdbus.scanner:cached devices: {'/org/bluez/hci0/dev_C0_3D_AA_ED_C6_39': {'Address': 'C0:3D:AA:ED:C6:39', 'AddressType': 'random', 'Name': 'C03DAAED', 'Alias': 'C03DAAED', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {307: b'\x02\xcc\x02\x8d\x00\x9b\x00\x0e\x02\xb7\x00\x93\x01\x89\x00\x1f\x01\xf6\x00Z\x00\x00\x00\x00\x00'}, 'ServicesResolved': False}}
```

FiachraBarry99, 2021-06-23:
Ok, thank you for the help. I presume to turn logging off again I just set `BLEAK_LOGGING=0` yes? Also I am going to close this issue if it is not an issue with Bleak but rather the BlueZ backend.

dlech, 2021-06-23:
If you did `export BLEAK_LOGGING=1`, you can do `unset BLEAK_LOGGING` to disable it in that shell.

FiachraBarry99, 2021-06-23:
Ok, thank you.

closed 2021-06-23 by FiachraBarry99