hbldh/bleak #580: Can't work with tkinter.filedialog.askdirectory

lin1987www, 2021-06-24:
* bleak version: 0.12.0
* Python version: 3.9.5
* Operating System: Win10 2004 (OS Build 19041.1052)

### Description

when using  import bleak and tkinter.filedialog.askdirectory will make program stop respond

### What I Did

```
import bleak  # Even I didn't use anything, just import bleak, still cause problem
import tkinter as tk
from tkinter import filedialog

root = tk.Tk()
dest_path = filedialog.askdirectory()
```

If remove the line

> import bleak

 it will work just fine.

dlech, 2021-06-24:
Given that you are using Python 3.9, I assume you are using the WinRT backend. This could be caused by https://github.com/microsoft/xlang/issues/690 since both Bleak and Tkinter need to initialize the thread to use Windows APIs.

dlech, 2021-06-24:
Does this work?

```python
import bleak  # Even I didn't use anything, just import bleak, still cause problem
import tkinter as tk
from tkinter import filedialog
from winrt import _winrt

_winrt.unint_apartment()

root = tk.Tk()
dest_path = filedialog.askdirectory()
```

lin1987www, 2021-06-25:
Yes! It works. Thanks you very much!

```
import bleak
import tkinter as tk
from tkinter import filedialog
from winrt import _winrt

_winrt.uninit_apartment()  # This line make filedialog.askdirectory() work

root = tk.Tk()
dest_path = filedialog.askdirectory()
```

But I still confuse, without `_winrt.uninit_apartment()` these similar methods below still work. 

filedialog.asksaveasfile()
filedialog.asksaveasfilename()
filedialog.askopenfile()
filedialog.askopenfiles()
filedialog.askopenfilename()

What make `filedialog.askdirectory()` so special? Or maybe this is tkinter's issue? 

If you have any idea, I would like to try.

dlech, 2021-06-25:
`import winrt` calls [RoInitialize](https://docs.microsoft.com/en-us/windows/win32/api/roapi/nf-roapi-roinitialize) with the default `RO_INIT_MULTITHREADED`. Most GUI toolkits on Windows need a single thread apartment instead. I think it is just by luck that the other dialogs worked without locking up.

closed 2021-06-25 by lin1987www

hbldh, 2021-06-28:
This shed a lot of light on problems using GUIs with Bleak. This should be addressed in #266.

Good work finding `uninit_apartment`! That is many steps closer to a root cause than I have found so far.