hbldh/bleak #581: First example results in "Abort trap: 6" in Mac OS

szepi1991, 2021-06-25:
* bleak version: 0.12.0
* Python version: Python 3.7.3
* Operating System: MacOS 11.3.1
* BlueZ version (`bluetoothctl -v`) in case of Linux: -

### Description

Tried to run the first [usage example](https://github.com/hbldh/bleak#usage), scanning for devices. Only output is
```
Abort trap: 6
```

### What I Did

```
import asyncio
from bleak import BleakScanner

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

 in bluetooth_list_2.py file, then run it:
```
$ python bluetooth_list_2.py
Abort trap: 6
```

dlech, 2021-06-25:
The terminal where this is running needs to have permission to use Bluetooth. See https://github.com/hbldh/bleak/issues/438#issuecomment-787125189.

szepi1991, 2021-06-25:
Thank you, this resolved the issue!

closed 2021-06-25 by szepi1991