hbldh/bleak #585: Write Large amounts of data fails

mmartinm1992, 2021-06-30:
* bleak version: 0.12.0
* Python version: 3.9.5
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Hi there, I am developing a BLE GUI in wondows, I have been using btstack in C and calling it from python. Then I came up with this amazing library and I decided to try to port my application to use bleak, however when I was performing a long write, (which I have been using with btstack and the prepare write, and execute commands) I realized it is not working with bleak.

I just tried to write to a characteristic using write_gatt_char with large (2Kbytes) and short data (50 bytes). when sending large ammounts of data I got the following error:

>   File "c:\Users\Martín Martín\Desktop\PythonVscWorkspace\Industrial_App.py", line 185, in set_factory_settings_ble
>     loop.run_until_complete(self.bleClient.write_gatt_char(self.SET_FACTORY_CHAR_UUID, factory_info_bytes.raw, response = False))
>   File "C:\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
>     return future.result()
>   File "C:\Python39\lib\site-packages\bleak\backends\winrt\client.py", line 642, in write_gatt_char
>     write_result = await characteristic.obj.write_value_with_result_async(
> RuntimeError

and when I did it with 50 bytes (MTU is set to 500), everything worked just fine.
 
Is this a known bug or a missing future feature?

I also shaw this issue that might be related, but on linux. (https://github.com/hbldh/bleak/issues/263 ). Any help will be appreciated.

### What I Did
Here is my code :

```
import asyncio
from ctypes import *
from bleak import BleakClient

address = "18:FE:34:02:01:FE"
MODEL_NBR_UUID = "4F4A4554-4520-4341-4C4F-520001000002"

async def run(address):
    async with BleakClient(address) as client:
        #info = "{\"response_code\": 0,\"result\":12345}" # put short data here
        info = "{\"response_code\": 0,\"result\":\" This is a larger amount of data that I want to send: In a village of La Mancha, the name of which I have no desire to call to mind, there lived not long since one of those gentlemen that keep a lance in the lance-rack, an old buckler, a lean hack, and a greyhound for coursing. An olla of rather more beef than mutton, a salad on most nights, scraps on Saturdays, lentils on Fridays, and a pigeon or so extra on Sundays, made away with three-quarters of his income. The rest of it went in a doublet of fine cloth and velvet breeches and shoes to match for holidays, while on week-days he made a brave figure in his best homespun. He had in his house a housekeeper past forty, a niece under twenty, and a lad for the field and market-place, who used to saddle the hack as well as handle the bill-hook. The age of this gentleman of ours was bordering on fifty; he was of a hardy habit, spare, gaunt-featured, a very early riser and a great sportsman. They will have it his surname was Quixada or Quesada (for here there is some difference of opinion among the authors who write on the subject), although from reasonable conjectures it seems plain that he was called Quexana. This, however, is of but little importance to our tale; it will be enough not to stray a hair's breadth from the truth in the telling of it.\"}" # put your large data here
        length = len(info)
        factory_info_bytes = create_string_buffer(info.encode('utf-8'), length)
        await client.write_gatt_char(MODEL_NBR_UUID, factory_info_bytes.raw)

loop = asyncio.get_event_loop()
loop.run_until_complete(run(address))
```

dlech, 2021-06-30:
`write_gatt_char` by default does write without response, which is limited to the MTU size - 3 bytes. Maybe with btstack it did write with response or just took care of chopping up the data for you.

You could wrap `factory_info_bytes` in `io.BytesIO` then make a loop to read and send 500 bytes at a time.

mmartinm1992, 2021-07-01:
The write with response or not is irrelevant in this case, it is a different command upon the hci layer I think, instead of a normal write, it sendes a prepare write to say to the device that the data it is receiving is incomplete, and then an execute write when it is complete.

Sending chunks of 500 bytes won't help as the device I am talking to will try to interpret it as a complete json each time it receives the data. Any idea if this is supported on the lowest layers of the library? I saw that it is using Winrt underneath how can I modify the code?

dlech, 2021-07-01:
WinRT docs are here: https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile?view=winrt-20348

Bleak WinRT code is here: https://github.com/hbldh/bleak/tree/develop/bleak/backends/winrt

Bleak only does write with response and write without response and not prepared writes, authenticated writes, etc. since as far as we know those aren't available on all supported platforms - happy to be proved wrong though.

dlech, 2022-09-12:
According to the Bluetooth spec the max attribute value size is 512 bytes (for write with response), so writing 2K at one go is not possible. We recently added a new attribute to get the max write without response size at runtime.

closed 2022-09-12 by dlech