hbldh/bleak #586: Using write without response causes exception on disconnect

adamincera, 2021-07-02:
* bleak version: 0.12.0
* Python version: 3.9.5
* Operating System: macOS 11.4
* BlueZ version (`bluetoothctl -v`) in case of Linux: N/A

### Description

It seems that calling `write_gatt_char` with `response` set to False results in an exception being thrown when `disconnect()` is called.

### What I Did
Here is an example program that reproduces the issue:
```
#!env/bin/python

import asyncio
from bleak import BleakScanner, BleakClient, BleakError

uuid = '4831911B-DE54-409F-8750-0172C5A43BEF'
write_char = 'CED94322-6692-4A12-87D5-6F2764762B2A'

test_data = b'\x00' * 20

def callback(handle, value):
    print("received data: " + str(value))

async def find_device(device_uuid):
    target_device = None
    tries = 0
    while target_device == None and tries < 5:
        devices = await BleakScanner.discover()
        for device in devices:
            if "uuids" in device.metadata and device_uuid.lower() in device.metadata["uuids"]:
                target_device = device

        tries = tries + 1

    return BleakClient(target_device)

async def reproduce_exception():
    device = await find_device(uuid)
    await device.connect()
    await device.start_notify(write_char.lower(), callback)
    await device.write_gatt_char(write_char.lower(), test_data, False)
    await device.stop_notify(write_char.lower())
    await device.disconnect()

if __name__ == '__main__':
    event_loop = asyncio.get_event_loop()
    event_loop.run_until_complete(reproduce_exception())

```
When I run that script, I get the following output:
```
¯\_(ツ)_/¯ ble_testing:./minimal_example.py
Future exception was never retrieved
future: <Future finished exception=BleakError('disconnected') created at /usr/local/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py:424>
source_traceback: Object created at (most recent call last):
  File "/Users/adamincera/code/ble_testing/./minimal_example.py", line 38, in <module>
    event_loop.run_until_complete(reproduce_exception())
  File "/usr/local/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 629, in run_until_complete
    self.run_forever()
  File "/usr/local/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 596, in run_forever
    self._run_once()
  File "/usr/local/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 1882, in _run_once
    handle._run()
  File "/usr/local/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/adamincera/code/ble_testing/./minimal_example.py", line 32, in reproduce_exception
    await device.write_gatt_char(write_char.lower(), test_data, False)
  File "/Users/adamincera/code/ble_testing/env/lib/python3.9/site-packages/bleak/backends/corebluetooth/client.py", line 319, in write_gatt_char
    success = await self._delegate.write_characteristic(
  File "/Users/adamincera/code/ble_testing/env/lib/python3.9/site-packages/bleak/backends/corebluetooth/PeripheralDelegate.py", line 166, in write_characteristic
    future = self._event_loop.create_future()
  File "/usr/local/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 424, in create_future
    return futures.Future(loop=self)
bleak.exc.BleakError: disconnected
¯\_(ツ)_/¯ ble_testing:
``` 
Running the same script with `response` set to True results in no error message.

dlech, 2021-07-03:
Can you please try #588 to see if it fixes the problem?

adamincera, 2021-07-03:
Fixed! Thanks for the quick response.

Do you know when I could expect this patch to be available through pip? I'm not really familiar with the release process for that.

dlech, 2021-07-03:
This seems worth of a 0.12.1 bug fix release. Maybe next week if I have time.

commit 405351de (2021-07-03): CoreBluetooth: fix unawaited future for write without response

In CoreBluetooth, there is no feedback for write without response, so
we just have to send it and hope for the best. In this case we should
not create a future since it gets registered with the event loop but it
is never awaited.

Fixes #586
PR #588 (2021-07-07): CoreBluetooth: fix unawaited future for write without response
In CoreBluetooth, there is no feedback for write without response, so we just have to send it and hope for the best. In this case we should not create a future since it gets registered with the event loop but it is never awaited.

Fixes #586
PR #592 (2021-07-07): Release/0.12.1

Changed
------

* Changed minimum ``winrt`` package version to 1.0.21033.1. Fixes #589.

Fixed
-----

* Fixed unawaited future when writing without response on CoreBluetooth backend.
  Fixes #586.
closed 2021-07-07 by dlech