hbldh/bleak #587: MacOS, BLEDevice from detection_callback without 'metadata' and 'delegate'

pawelm87, 2021-07-02:
* bleak version: 0.11.0
* Python version:  3.7
* Operating System: 10.15
* BlueZ version (`bluetoothctl -v`) in case of Linux: None

### Description

When I use the example "detection_callback.py" I get a BLEDevice object but with incomplete data, the 'metadata' property is missing among others. If I want to use it later to connect to the device I get an error that there is no 'delegate' property in the object.
When I use the example "connect_by_bledevice.py", function "find_device_by_address" finding the device and return a BLEDevice object and from it I can connect to the device after BLE because it has more properties including 'metadata'

### What I Did

connect_by_bledevice.py
```
...
async def print_services(ble_address: str):
    device = await BleakScanner.find_device_by_address(ble_address, timeout=20.0)
    print("MY DEVICE DATA", device.metadata)
    if not device:
        raise BleakError(f"A device with address {ble_address} could not be found.")
    async with BleakClient(device) as client:
        svcs = await client.get_services()
        print("Services:")
        for service in svcs:
            print(service)
...
```

detection_callback.py
```
....
def simple_callback(device: BLEDevice, advertisement_data: AdvertisementData):
    print(device.address, "RSSI:", device.rssi, advertisement_data)

async def run():
    scanner = BleakScanner()
    scanner.register_detection_callback(simple_callback)

    while True:
        await scanner.start()
        await asyncio.sleep(5.0)
        await scanner.stop()
...
```

Analyzing the library it seems that 'find_device_by_address' function uses 'find_device_by_filter' and this one uses 'callback_detection'. Therefore both solutions should give the same result, two same objects but it doesn't happen.

I need to find a device that has a random MAC address but has some information encoded in the broadcast frame and I need to decode it first to know if this is the device I am looking for. On windows 10 everything works fine using 'callback_detection' mechanism but on MacOs10.15 it does not (in BLEDevice it baks metadata and detegate).

I normally work on Windows 10 but I want to adapt the code for macOS and I have this weird problem.

dlech, 2021-07-02:
Please try with Bleak v0.12.0.

pawelm87, 2021-07-05:
thank you for your reply.
I have looked at the changelog of the new version and other forum discussions regarding my problem. I will try the new version and post soon or just close the topic if it is resolved.

pawelm87, 2021-07-06:
updating the library to version 0.12.0 solved the problem

closed 2021-07-06 by pawelm87