hbldh/bleak #594: WIN10 NotImplementedError error during connect "connect_by_bledevice.py"

pawelm87, 2021-07-08:
* bleak version: 0.12.1
* Python version: 3.7.4
* Operating System: win 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: none

### Description

I continue the problem with #589 
I updated lib to the new version 0.12.1 and run the previous script.

### What I Did

```
"""
Connect by BLEDevice
"""

import asyncio

from bleak import BleakClient, BleakScanner
from bleak.exc import BleakError

async def print_services(ble_address: str):
    print("1--------------------------")
    device = await BleakScanner.find_device_by_address(ble_address, timeout=20.0)
    print("2--------------------------", device)
    if not device:
        print("3--------------------------")
        raise BleakError(f"A device with address {ble_address} could not be found.")
    async with BleakClient(device) as client:
        print("4--------------------------")
        svcs = await client.get_services()
        print("Services:")
        for service in svcs:
            print(service)

loop = asyncio.get_event_loop()
loop.run_until_complete(print_services("E4:D0:AE:38:71:7E"))
```
Logs:
```
PS C:\Users\p\Desktop\PM\workspace\python\cli-tool> & C:\Users\p\Desktop\PM\workspace\python\VENV-cli-tool\Scripts\python.exe .\bleak_scan.py
1--------------------------
2-------------------------- E4:D0:AE:38:71:7E: DEV01
Traceback (most recent call last):
  File ".\bleak_scan.py", line 27, in <module>
    loop.run_until_complete(print_services("E4:D0:AE:38:71:7E"))
  File "C:\Users\p\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 579, in run_until_complete
    return future.result()
  File ".\bleak_scan.py", line 18, in print_services
    async with BleakClient(device) as client:
  File "C:\Users\p\Desktop\PM\workspace\python\VENV-cli-tool\lib\site-packages\bleak\backends\client.py", line 61, in __aenter__
    await self.connect()
  File "C:\Users\p\Desktop\PM\workspace\python\VENV-cli-tool\lib\site-packages\bleak\backends\winrt\client.py", line 227, in connect
    await self.get_services(use_cached=use_cached)
  File "C:\Users\p\Desktop\PM\workspace\python\VENV-cli-tool\lib\site-packages\bleak\backends\winrt\client.py", line 495, in get_services
    str(characteristic.uuid),
NotImplementedError
PS C:\Users\p\Desktop\PM\workspace\python\cli-tool> & C:\Users\p\Desktop\PM\workspace\python\VENV-cli-tool\Scripts\pip.exe show bleak
Name: bleak
Version: 0.12.1
Summary: Bluetooth Low Energy platform Agnostic Klient
Home-page: https://github.com/hbldh/bleak
Author: Henrik Blidh
Author-email: henrik.blidh@nedomkull.com
License: MIT
Location: c:\users\p\desktop\pm\workspace\python\venv-cli-tool\lib\site-packages
Requires: pythonnet
Required-by:
```
The script starts to connect to the device, but an Exception is raised.

hbldh, 2021-07-08:
Your script works fine in my installation with winrt. But I think I know the reason.

Could you do a `pip show winrt`? Because you use Python 3.7, Bleak will install only the 3.7-default `pythonnet` backend and this will mean that your `winrt` is probably still at `1.0.20330.1` instead of the required `1.0.21033.1`. 
WinRT is the default installed backend for Python 3.9+, but if you have WinRT installed separately in your environment, Bleak will use it anyway regardless of Python version.

Do a `pip install winrt==1.0.21033.1` and this problem will probably go away.

pawelm87, 2021-07-08:
@hbldh you were right the problem lay in the winrt library.

I was installing it manually from dependencies and check if BLE modem is available  on my side.

I update the winrt library to the version you provided and everything is working.

@hbldh I have a question if your library now checks for BLE modem availability on WIN and possibly raise an Exception that I can catch and handle ?

Thank you for your help 
Best regards Pawel

dlech, 2021-07-10:
> I have a question if your library now checks for BLE modem availability on WIN and possibly raise an Exception that I can catch and handle ?

Please see #320

closed 2021-07-10 by dlech