hbldh/bleak #597: Cannot connect to device on macOS

KeKsBoTer, 2021-07-12:
* bleak version: 0.12.1
* Python version: 3.7 / 3.9
* Operating System: macOS BigSur 11.2.2

### Description

I am trying to connect to a BLE device (Arduino Nano IoT 33) with a python script. I use the `connect_by_bledevice.py` example and only changed the address.
On Linux I was able to connect to the BLE device with no problem.

On my Mac Book Pro 2017 I cannot connect to the device.
I get the following error:
`A device with address <device_id> could not be found` where <device_id> is my device id.

With the `discover.py` example I can find the device.
I tried different BLE devices and got the same error.

Any suggestions how I can debug this?

KeKsBoTer, 2021-07-12:
After I restarted my MacBook the error does not occur anymore.
Any idea what caused this?

dlech, 2021-07-16:
> Any idea what caused this?

Not without additional logging.

> Any suggestions how I can debug this?

For future reference: https://bleak.readthedocs.io/en/latest/troubleshooting.html

Feel free to re-open if the issue happens again.

closed 2021-07-16 by dlech