hbldh/bleak #601: Problem with data from 5 ble modules in bleak

sd1026, 2021-07-16:
I have been able to connect to my 5 BLE modules, and I am receiving notifications from all of them. But I have a doubt that when I pass data through the callback function, it takes (sender, data) as parameters. Now, here I am confused as to what is this "sender" here. When i debug the code, it always prints 13 as sender.

So, is it the case that multiple ble modules are connected but, the notifications are coming from only one of them.
Or, is it something that the code already has.

dlech, 2021-07-16:
`sender` is a unique identifier for the attribute handle that sent the data. This is probably only useful when using the same callback for multiple characteristics on a single device. It would have been nice if the callback also included the `BleakClient` object as well.

As a workaround you could use [functools](https://docs.python.org/3.8/library/functools.html) to create a unique callback for each device.

sd1026, 2021-07-19:
what if i assume that the data i got is from all of the devices simultaneously. But just to differ that data by device, what if I add a BleakClient as a parameter to the function in the library.
Or you may suggest some good method, as I have not been clear with the functools approach.

sd1026, 2021-07-19:
I tried to store the data from two sensor modules in different files, while subscribing to different characteristics at the same time. I now found that both the devices are giving data in parallel. But how can i find the difference, if the characteristics are same for all BLE sensor modules.

dlech, 2021-07-19:
Does this help? https://github.com/hbldh/bleak/discussions/467

sd1026, 2021-07-19:
Thanks sir, I'll give it a try first.

sd1026, 2021-07-19:
Sir I want to know one thing. As I am using this python code on windows as well. The problem here is that with the bluetooth 4.0 adapter, the code accepts two ble devices perfectly. But when I try to give it a third address. It says unreachable.

Moreover, when I try with the Raspberry Pi 3b+, with the bluetooth 4.2 on chip adapter. I get to connect 5 ble devices at same time. 

So, the thing is that I want to know is there such difference in handling of multiple slaves based on version of Adapters.

dlech, 2021-07-19:
AFAIK, the number of connections depends on the Bluetooth chipset, so you would have to consult documentation from the manufacturer to find out what it is capable of.

sd1026, 2021-07-21:
One more thing, may I know the reason why after some time the data from multiple sensors slows down. And if all the devices are connected the data comes very slowly. 

The sensors are sending the data at a very high speed, and we need to capture that for a real time scenario following some calculations as well.

So is there a way to have a fast growing dataset that never slows down irrespective of number of connections.

sd1026, 2021-07-22:
Also, I have seen that sometimes an "Unreachable exception occurs" that says **bleak.exc.BleakDotNetTaskError: Could not get GATT services: Unreachable**. Sometimes it does not appear and the script works very well.

Is there any way that I can solve it, and whenever I start the script it must connect to devices, and work well without this exception.

sd1026, 2021-07-22:
> 
> 
> Does this help? #467

This worked very well for me. Thanks sir

sd1026, 2021-07-22:
Can we make a scneraio, where the script holds until I connect all my BLE devices with the script, and when all are connected then start_notify all of them at once.

sd1026, 2021-07-22:
I also want to know that does Windows keep the memory of the connected devices for bluetooth. If yes, then how to check and dispose that memory before starting to connect with the devices.

sd1026, 2021-07-23:
Why i am getting this error everytime on Windows 10 20H2 version. I am unable to resolve this issue. Please help me fix this. The script connects to a few devices at the same time and works fine, but this issue arises each time I start it. Why is it so?

`Task exception was never retrieved
future: <Task finished name='Task-5' coro=<connect_to_device() done, defined at C:\Users\BIOMOD\AppData\Roaming\JetBrains\PyCharmCE2021.1\scratches\scratch_2.py:27> exception=BleakDotNetTaskError('Could not get GATT services: Unreachable')>
Traceback (most recent call last):
  File "C:\Users\BIOMOD\AppData\Roaming\JetBrains\PyCharmCE2021.1\scratches\scratch_2.py", line 28, in connect_to_device
    async with BleakClient(address, timeout=90.0) as client:
  File "C:\Python39\lib\site-packages\bleak-0.13.0a1-py3.9.egg\bleak\backends\client.py", line 60, in __aenter__
    await self.connect()
  File "C:\Python39\lib\site-packages\bleak-0.13.0a1-py3.9.egg\bleak\backends\winrt\client.py", line 227, in connect
    await self.get_services(use_cached=use_cached)
  File "C:\Python39\lib\site-packages\bleak-0.13.0a1-py3.9.egg\bleak\backends\winrt\client.py", line 417, in get_services
    raise BleakDotNetTaskError(
bleak.exc.BleakDotNetTaskError: Could not get GATT services: Unreachable`

dlech, 2021-07-23:
Probably scheduled a task without awaiting it. Hard to say without seeing the code.

sd1026, 2021-07-26:
Sir, i have this code to connect 5 BLE devices. There are two scenarios, where i use the same code in Windows & raspberry pi 3b+. But the problem is that when i run the script on raspberry pi, it connects all five devices (error probability is less in raspberry pi).

And in Windows, I use the same code. But the above error, or and timeout error occurs each time. (error probability is really high). Now, when I want to connect all five, and give their address, it connects to 3 or 4 devices arbitrarily.

Now, it works fine on raspberry pi. But i have used that only for testing purposes. So, I want this script for Windows. But I have some points as doubt, which I want to clear.

Doubts in this case:

1. The adapter I have on Windows 10 is a Tp-Link UB400 Bluetooth 4.0 adapter. (does even know that 4.0 can handle the parallel devices and lots of data simultaneously).
2. The raspberry pi script is successful as it has a bluetooth 4.2 adapter on-board.
3. Does that makes it clear to use a blueooth 4.2 adapter on Windows as well.
4. Following the point 1. I ran the script a few days back where it Bluetooth 4.0 adapter gets connected to all 5 devices, and runs for around 30 minutes.
5. But this scenario does not follow each time.

Code I have used on Windows and raspberry pi 3b+:
`from bleak import BleakClient, discover, BleakError, BleakScanner
from functools import partial
import asyncio

CHAR_UUID = "00000080-0001-11e1-ac36-0002a5d5c51b".format(0xFFE1)
connected_devices = []

def callback(client: BleakClient, sender: int, data: bytearray):
    hex_string = data.hex()
    print(f"{client.address}", "".join([hex_string[x:x + 2] for x in range(0, len(hex_string), 2)]))

def run(addresses):
    loop = asyncio.get_event_loop()

    tasks = asyncio.gather(*(connect_to_device(address) for address in addresses))

    loop.run_until_complete(tasks)

async def connect_to_device(address):
        async with BleakClient(address, timeout=20.0, use_cached=False) as client:
            print(f"Connected: {address} {client.is_connected}")
            connected_devices.append(client.address)
            try:
                await client.start_notify(CHAR_UUID, partial(callback, client))
            except Exception as e:
                print(f"Exception: {e}")

if __name__ == "__main__":
    run(["26:80:E1:4E:F8:B6", "26:80:E1:AD:99:34", "26:80:E1:C1:90:51", "26:80:E1:C1:11:D0", "26:80:E1:16:E4:F2"])`

sd1026, 2021-07-26:
@dlech 
Sir awaiting a response from your side. Suggest me some corrections, or suggest me to change my hardware, if that's required.

dlech, 2021-07-26:
There is example code and several issues here that show how to connect to multiple devices at the same time.

sd1026, 2021-07-26:
I am not clear about what you just said

sd1026, 2021-08-09:
Sir, i am unable to get the device connections with Bluetooth 4.0 adapter to 5 ble devices, it randomly picks up some address from the address list, and start to notify. It would be really helpful if you could help me to resolve the issue. I have shared the code above as well.

hbldh, 2021-08-09:
As already stated, the issues are not a help line where we will debug your code for you. We will not `suggest changes` for your code, but we try to address actual errors in the Bleak package. We are glad to address things that you have discovered and can show are wrong in the Bleak source code. We are not that delighted to be asked fix your code.

Most probably, your issue is due to some limitation on the number of BLE devices Windows+Tp-Link UB400 is allowed to connect to compared to Rpi+Onboard chip. It has nothing (I strongly believe, but I may be wrong) to do with the Bluetooth version. If the code works on your RPi, but cannot connect to the same number of devices in Windows, I would first try to find the lowest number of devices that gets a working result in Windows and accept that there apparently is a limit there. Bleak cannot help you there.

I have however seen multiple weak points in your code and I will point them out to you, so you can fix it yourself and learn something in the process:

```python
import asyncio
from bleak import BleakClient
from functools import partial

CHAR_UUID = "00000080-0001-11e1-ac36-0002a5d5c51b".format(
    0xFFE1
)  # The format here does not do anything

def callback(client: BleakClient, sender: int, data: bytearray):
    hex_string = data.hex()
    print(
        f"{client.address}",
        "".join([hex_string[x : x + 2] for x in range(0, len(hex_string), 2)]),
    )

def run(addresses):
    # These loops will be the same loop. Use only one loop.
    # Ref: https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.get_event_loop
    loop = asyncio.get_event_loop()
    loop1 = asyncio.get_event_loop()

    for address in addresses:
        try:
            loop.create_task(connect_to_device(address))
        except Exception as e:
            # No. Do not stop a loop with potentially multiple long time running jobs on it because one job failed.
            loop.stop()
            address = address
            # This adds the new task to the same loop you just stopped.
            loop1.create_task(connect_to_device(address))

    loop.run_forever()
    loop1.run_forever()

    # This method is problematic. My advice is to use the `asyncio.gather` and `loop.run_until_complete`
    # pattern in the two_devices.py example in the Bleak repo.

async def connect_to_device(address):
    async with BleakClient(address, timeout=20.0, use_cached=False) as client:
        print(f"Connected: {address} {client.is_connected}")
        await asyncio.sleep(0.8)
        # Look closely at what the loop below says. It says Do `start_notify` forever. This is not what you want...
        # This will essentially stop and restart your notifications in very short time intervals forever 
        # leading to unpredictable behaviour.
        while True:
            await client.start_notify(CHAR_UUID, partial(callback, client))
        # Use something like
        #
        # client.start_notify(CHAR_UUID, partial(callback, client))
        # while True:
        #     await asyncio.sleep(1.0)
        #
        # This is because you want the client to be kept is a connected state, but you defer all processing time to
        # be used by the callback instead. If you are using Bleak 0.12+ then you can make your `callback` method async
        # as well to be even more async compliant.

if __name__ == "__main__":
    run(
        [
            "26:80:E1:4E:F8:B6",
            "26:80:E1:AD:99:34",
            "26:80:E1:C1:90:51",
            "26:80:E1:C1:11:D0",
            "26:80:E1:16:E4:F2",
        ]
    )
```

Practice using `asyncio` from online tutorials and I think you will soon see what I mean. Maybe it turns out that there was no limitation, but that the restarting of notifications was the problem. 

Good luck!

sd1026, 2021-08-17:
Can this code work with BLED112 dongle

closed 2021-11-26 by hbldh

wahadatjan, 2024-07-08:
for me on raspberry pi getting below error
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress