hbldh/bleak #602: Bleak dbus client opens bluetooth process on each instanciation

elupus, 2021-07-19:
* bleak version: 0.12.1
* Python version: 3.9
* Operating System: Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

With this type of workflow: Open connection, give device command, close connection using a async context manager the dbus bluez backend spawns subprocesses on each iteration here: https://github.com/hbldh/bleak/blob/7e0fdae6c0f6a78713e5984c2840666e0c38c3f3/bleak/backends/bluezdbus/client.py#L88

Feel like this should be a cached check.

### What I Did

```python
while True:
  async with BleakClient(ble_address) as client:
    client.read..
  await asyncio.sleep(10)
```

I suppose i could do as this:

```python
client = BleakClient(ble_address)
while True:
  async with client:
    client.read..
  await asyncio.sleep(10)
```

But it's less clean.

dlech, 2021-07-19:
We also get the BlueZ version when the `bleak` module is imported, so we could just save that and not each time we create a `BleakClient` instance.

---

Another way to write the loop (makes it slightly more obvious what the intention is but behaves the same as the 2nd alternative above):
```python
async with BleakClient(ble_address) as client:
    while True:
        client.read...
        client.disconnect()
        await asyncio.sleep(10)
        client.connect()
```

hbldh, 2021-08-09:
Wrote a small fix with a PR for this.

commit aeb7a0a6 (2021-10-10): BlueZ: add lazy-get/check version method

This adds a new method to the BlueZ bakckend for checking the BlueZ
version. This lazy-gets the version from bluetoothctl so that the
subprocess only runs once.

Fixes #602.
closed 2021-10-10 by dlech