hbldh/bleak #606: Is passive scanning implemented in the Linux backend?

koenvervloesem, 2021-07-25:
* bleak version: 0.12.1
* Python version: 3.8.10
* Operating System: Ubuntu 20.04.2 LTS
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.53

### Description

I wanted to do a passive scan with Bleak. Issue https://github.com/hbldh/bleak/issues/500 seemed to suggest that this is possible with the keyword argument `scanning_mode='passive'` to the `BleakScanner` constructor. So I added this argument while initializing the `BleakScanner`. However, Wireshark still shows me SCAN_REQ packets coming from my computer and SCAN_RSP packets as replies from my BLE devices, which suggests that Bleak is still doing an active scan.

### What I Did

My Minimal Workable Example is:

```python
import asyncio
from bleak import BleakScanner

async def run():
    devices = await BleakScanner(scanning_mode='passive').discover()
    for device in devices:
        print(device)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

dlech, 2021-07-25:
I think Windows is the only OS that has an API to select passive scanning.

koenvervloesem, 2021-07-25:
You're right, I don't see any way to choose passive scanning in the [BlueZ D-Bus Adapter API description](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/adapter-api.txt), so apparently this isn't possible (yet).

bdraco, 2022-07-07:
Greetings

Home Assistant is currently overhauling our Bluetooth support and we are looking for a passive scanning option.  Our current recommendation is to use bleak for new integrations.  Thanks for that!

If this feature is viable soon we would want to only support bleak going forward, however if it’s something far future we likely need to consider supporting an additional Bluetooth library as well which we would really like to avoid.

If there is anything we can do to help move this forward, please let me know what is needed.  Thank you!

dlech, 2022-07-07:
If a company that uses Bleak want to sponsor me to take a week off of work, I can have a go at it.

balloob, 2022-07-07:
I sent you an email 👍

PR #884 (2022-07-25): Bluez advertisement watcher/passive scanning
This implements passive scanning in the BlueZ backend using the experimental [advertisement monitor api](https://github.com/bluez/bluez/blob/master/doc/advertisement-monitor-api.txt).

Fixes #606.

This PR also includes some improvements to the argument handling of the `BleakScanner` constructor that lays the groundwork for the new args required by this new feature. It also includes a new global BlueZ state manager which, in addition to making the passive scanning feature trivial to implement, will be useful in fixing other long-standing BlueZ issue [... cut]
commit a5b98c7a (2022-07-29): v0.15.0

Added
-----

* Added new ``assigned_numbers`` module and ``AdvertisementDataType`` enum.
* Added new ``bluez`` kwarg to ``BleakScanner`` in BlueZ backend.
* Added support for passive scanning in the BlueZ backend. Fixes #606.
* Added option to use cached services, characteristics and descriptors in WinRT backend. Fixes #686.
* Added ``PendingDeprecationWarning`` to use of ``address_type`` as keyword argument. It will be moved into the
  ``winrt`` keyword instead according to #623.
* Added better error message when adapter is not present in BlueZ backend. Fixes #889.

Changed
-------

 [... cut]
PR #911 (2022-07-29): Release/v0.15.0

Added
-----

* Added new ``assigned_numbers`` module and ``AdvertisementDataType`` enum.
* Added new ``bluez`` kwarg to ``BleakScanner`` in BlueZ backend.
* Added support for passive scanning in the BlueZ backend. Fixes #606.
* Added option to use cached services, characteristics and descriptors in WinRT backend. Fixes #686.
* Added ``PendingDeprecationWarning`` to use of ``address_type`` as keyword argument. It will be moved into the
  ``winrt`` keyword instead according to #623.
* Added better error message when adapter is not present in BlueZ backend. Fixes #889.

Changed
- [... cut]
closed 2022-07-25 by dlech