hbldh/bleak #614: Windows 10 - Error in start_notify

ArnauSoler, 2021-08-03:
* bleak version: 0.11.0
* Python version: 3.7.9
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: No idea

### Description

I'm trying to connect to a notify characteristic that I know it works because in the Xamarin Forms app for Android that I developed it works. Now I'm creating a Python app because I need to create a Windows version, but the notify characteristic always says the same error: bleak.exc.BleakDotNetTaskError: System.Exception: The attribute cannot be written. (Exception from HRESULT: 0x80650003)

### What I Did

I have a function that given a client tries to connect to my custom characteristic.

```python
async def start_notifies(client: BleakClientDotNet):
    print("Connecting to notify characteristics")
    x = await client.read_gatt_char(CHARACTERISTIC_UUID_MEASUREMENTS)
    print(x)
    print("Writing to the characteristic")
    await client.write_gatt_char(CHARACTERISTIC_UUID_MEASUREMENTS, bytearray("Hola",'utf-8'))
    print("Starting notify")
    await client.start_notify(CHARACTERISTIC_UUID_MEASUREMENTS, notification_handler)
```

The output of this fragment of code is:

> Connecting to notify characteristics
bytearray(b'Hola')
Writing to the characteristic
Starting notify

And then all the error message

```
Traceback (most recent call last):
  File ".\ANIME_Optris_BLE_Device.py", line 134, in <module>
    loop.run_until_complete(run())
  File "C:\Users\solerar\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 587, in run_until_complete
    return future.result()
  File ".\ANIME_Optris_BLE_Device.py", line 17, in run
    await connect_to_device(d.address)
  File ".\ANIME_Optris_BLE_Device.py", line 40, in connect_to_device
    await asyncio.gather(*my_tasks)
  File ".\ANIME_Optris_BLE_Device.py", line 92, in start_notifies
    await client.start_notify(CHARACTERISTIC_UUID_MEASUREMENTS, notification_handler)
  File "C:\Users\solerar\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\client.py", line 855, in start_notify
    return_type=GattCommunicationStatus,
  File "C:\Users\solerar\AppData\Local\Programs\Python\Python37\lib\site-packages\bleak\backends\dotnet\utils.py", line 79, in wrap_IAsyncOperation
    raise BleakDotNetTaskError(op.ErrorCode.ToString())
bleak.exc.BleakDotNetTaskError: System.Exception: The attribute cannot be written. (Exception from HRESULT: 0x80650003)
```

So I am completely lost, because I know that I can write to the characteristic but the error message says that I can't.

I have tried to change the custom UUID because another issue says that that is the problem, but no luck for me. Now the UUID is 00001001-c5f0-4582-9709-ad246b014c94.

If someone knows what is the mistake I'm doing, I will appreciate some help.

Thanks a lot.

dlech, 2021-08-03:
Does it work with Bleak v0.12.1?

ArnauSoler, 2021-08-09:
> Does it work with Bleak v0.12.1?

I just tried to upgrade Bleak to v0.12.1 but the error message is the same

dlech, 2021-08-09:
You could try using the `service_explorer.py` example to be sure you have the right UUID.

ArnauSoler, 2021-08-16:
> You could try using the `service_explorer.py` example to be sure you have the right UUID.

When using `service_explorer.py` I get 
```
[Characteristic] 00001001-c5f0-4582-9709-ad246b014c94 (Handle: 65): Browse Group Descriptor Service Class (read,write,notify), Value: b'0'
                [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 67): Characteristic User Description) | Value: b'Measured value(s)'```

I'm quite new in using Bluetooth, but I don't see where is the problem, it should allow me to call start_notify(), or I am making some mistake that I'm not seeing?

dlech, 2021-08-16:
Try calling `write_gatt_char()` with `response=True` (it is `False`) by default. 

The properties listed in the output of `service_explorer.py` only lists `write` and not `write_without_response`.

ArnauSoler, 2021-08-17:
> Try calling `write_gatt_char()` with `response=True` (it is `False`) by default.
> 
> The properties listed in the output of `service_explorer.py` only lists `write` and not `write_without_response`.

I tried  `write_gatt_char()` with `response=True` and it works.

The part that gives the error is in `client.py`:

```python
status = await wrap_IAsyncOperation(
    IAsyncOperation[GattCommunicationStatus](
        characteristic_obj.WriteClientCharacteristicConfigurationDescriptorAsync(
            cccd
        )
    ),
    return_type=GattCommunicationStatus,
)
```

If I comment that part it doesn't give me any error, but then the callback functions are never called.

Is there a way to know if the CharacteristicConfigurationDescriptor is writeable?

Thanks a lot for your help.

dlech, 2021-08-17:
>                 [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 67): Characteristic User Description) | Value: b'Measured value(s)'

Sorry, I didn't notice that before. For some reason, the device is sending wrong data about the descriptor. The value should be `0` or `1`, but it is `b'Measured value(s)'`. So apparently Windows has picked up the wrong attribute for the Characteristic Configuration Descriptor.

Maybe there is a way to remove the device in the Windows Settings? This should cause the device to be re-enumerated when it is connected again. Or the device could just be defective.

ArnauSoler, 2021-08-31:
> So apparently Windows has picked up the wrong attribute for the Characteristic Configuration Descriptor.

Yes! It was picking the wrong 2902 descriptor because the device was wrongly configurated.

Thanks a lot for your help!!

closed 2021-08-31 by ArnauSoler

hmake98, 2022-02-17:
Hey, I'm facing the same issue. I'm fetching data from the device using `start_notify` and it's logging the error:

`[WinError -2140864509] The attribute cannot be written`

```
import asyncio
from bleak import BleakScanner, BleakClient

scanner = BleakScanner()

def callback(sender: int, data: bytearray):
    print('in callback')
    print(sender, data.decode('base64'))

async def main():
    await scanner.start()
    await asyncio.sleep(10.0)
    await scanner.stop()

    for d in scanner.discovered_devices:
        if address == d.address:
            await connect(d.address)

async def connect(device_address):
    client = BleakClient(device_address)
    try:
        await client.connect()
        services = await client.get_services()
        for service in services:
            if main_service == service.uuid:
                chars = service.characteristics
                for char in chars:
                    await client.start_notify(char.handle, callback)
    except Exception as e:
        print(e)

asyncio.run(main())
```

ArnauSoler, 2022-02-18:
Hello @hmake98

> Hey, I'm facing the same issue. I'm fetching data from the device using `start_notify` and it's logging the error:
> 
> `[WinError -2140864509] The attribute cannot be written`
> 

Check if the characteristics you are starting the notify have the 2902 descriptor. If some of these characteristics don't have the descriptor, the program will try to write to the global 2902 descriptor, which is non writable.

At least that was my problem.

hmake98, 2022-02-18:
@ArnauSoler Hey, Thanks for the answer. Can you please give me some more details? I'm not able to understand CZ, I'm new to Bluetooth device connection. My target is to get data from the device every 2 seconds interval. 

I've used [this](https://www.npmjs.com/package/@abandonware/noble) package in Nodejs but It's not working with Bluetooth version 5. though it's working perfectly fine with the device. It doesn't show any errors while starting the notification.