hbldh/bleak #615: Windows 11 - Error in write_gatt_descriptor

theGENreel, 2021-08-03:
* bleak version: 0.12.1
* Python version: 3.9.2
* Operating System: Windows 11 Pro 22000.100

### Description

I'm trying to control my Mi Band 4. 
It seems that other people do not have this error. Therefore, I think that the problem is in the changes in the API. Perhaps the API is now returning int.

### What I Did

I have a function that writes data to descriptor.
```
async def aut():
    await client.write_gatt_descriptor(_desc_auth.handle, b"\x01\x00")
```

And here write_gatt_descriptor throws AttributeError.
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Users\Skummer\AppData\Local\Programs\Python\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "<input>", line 2, in aut
  File "C:\Users\Skummer\Documents\Projects\Python\miband4\venv\lib\site-packages\bleak\backends\winrt\client.py", line 690, in write_gatt_descriptor
    if write_result.status == GattCommunicationStatus.SUCCESS:
AttributeError: 'int' object has no attribute 'status'
```

I have verified that descriptor.obj.write_value_async in code below returns 0 (backends/winrt/client.py on 686 line).
```
write_result.status = await descriptor.obj.write_value_async(
            CryptographicBuffer.create_from_byte_array(list(data))
        )
```
And if you remove ".status" in the code below, the error goes away (line 690).
```
 if write_result.status == GattCommunicationStatus.SUCCESS:
            logger.debug("Write Descriptor {0} : {1}".format(handle, data))
```
It looks like there are some changes in the Windows 11 API, so the lib is not working now.

hbldh, 2021-08-04:
I do not think Bleak will support Windows 11 until it has been released for public use. I appreciate you raising the issue and making us aware of it, but solution to this will have to wait.

That said, you are free to delve deeper into the changes into the Bluetooth API of Windows 11 and make pull requests to fix the inevitably approaching problems. That would be much appreciated, since I am in no position to install the W11 preview on any of my systems!

dlech, 2021-08-04:
I don't think this has anything to do with Win 11. It should be calling `write_value_with_result_async()` to return a `GattWriteResult` instead of `write_value_async()` which returns `GattStatus` (which is enum with base type of int).

dlech, 2021-08-04:
@theGENreel can you please test #616?

theGENreel, 2021-08-04:
@hbldh @dlech Thanks for answers. Checked at #616, write_gatt_descriptor works.

closed 2021-08-04 by theGENreel

dlech, 2021-08-04:
Great, thanks for testing.

hbldh, 2021-08-05:
Ah, I see. I should stop making comments on issues without checking the code first...

Good work!

PR #616 (2021-08-04): Use correct method when writing descriptor in WinRT backend.
We always use the "with result" variant of Windows API functions to get more information about failures. This was probably just overlooked when porting the Windows backend to WinRT.

Fixes #615.
commit d1c08dbe (2021-08-04): Use correct method when writing descriptor in WinRT backend.

We always use the "with result" variant of Windows API functions to get
more information about failures. This was probably just overlooked when
porting the Windows backend to WinRT.

Fixes #615.