hbldh/bleak #617: start_notify needs asyncio.sleep

Erhannis, 2021-08-05:
* bleak version: 0.12.1
* Python version: 3.8.0
* Operating System: Windows 10 Enterprise
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Not sure if this is a bug, undocumented quirk, or just something I missed, but:
When I use `start_notify`, it seems to only work when `asyncio.sleep` is occurring.  (As opposed to `time.sleep`, for instance, or waiting via another concurrency library.)  This is fine for demos and examples, but seems like usually you'd want to e.g. keep handling messages until some event signaled an end.  For instance, I'm using pycsp to coordinate different threads, and blocking to wait for the signal to stop causes no bluetooth messages to be handled.  Is this expected behavior?  I didn't see it mentioned in the docs.  What's the suggested fix?  I suppose I could `asyncio.sleep(1)` in a loop and check for stop every time, but that's kinda gross.

### What I Did

If you want to reproduce it, open the enable_notifications.py example and replace `asyncio.sleep` with `time.sleep` (and import time).  The messages don't get processed during the sleep, and instead pile up and are processed all at once at the end.

dlech, 2021-08-05:
Can you give a minimal full working example to reproduce the problem?

Erhannis, 2021-08-06:
Ah.  It's the "of course" bit that's the problem - I didn't know or expect that, and I don't see it mentioned in the docs.  I've not really used asyncio much before; perhaps the docs could more clearly state that that's what Bleak uses, and to be careful about blocking with non-asyncio calls.

I'll look into the three things you mention - given some black box method `block_until_stop_signal()`, would one of them be most suitable for blocking (and processing messages) until the method returned?

dlech, 2021-08-06:
Assuming that it is thread-safe, `run_in_executor()`.

Erhannis, 2021-08-06:
That works, thanks!

closed 2021-08-09 by hbldh