hbldh/bleak #623: Separating OS specific keyword arguments

hbldh, 2021-08-12:
* bleak version: 0.12.0
* Python version: All
* Operating System: All

Bleak has several OS specific keyword, all of which are integrated into the regular `**kwargs` input to methods. The primary means of making users aware of what they can do to affect their current environment is through documentation. The documentation does provide this, but it is evidently not enough, given the amount of issues that e.g. asks why `device="hci1` does not work in Windows to select another Adapter.

@dlech proposed two different ways of improving Bleak in this regards:

1. Using a `win_`, `macos_`, and `linux_` prefix to keyword arguments that are os specific.
2. Using a `win={"x"=1. "y"=2, ...}` style of OS specific grouping of parameters.

Both will render nicely with `black` and both relays the desired effect in my mind. @hbldh prefers the second way, but given that Bleak _currently_ has few OS specific parameters it will be almost equal. Given that I, creator of Bleak, has much less use of it that most everyone else that will read this will have, I am more that ready to yield to public demand. 

I vote 2, and everyone else that votes will have his or her's vote counted equally. At the first of September, I will start to implement the winning choice. IF someone beats me to the implementation of one of the choices, I will take that one.

dlech, 2021-08-20:
Option 2 seems like it could be adapted to use [typing.TypedDict](https://docs.python.org/3/library/typing.html#typing.TypedDict) in the future (requires Python 3.8) as well for better type hints.

dlech, 2022-10-03:
This has been implemented.

closed 2022-10-03 by dlech