hbldh/bleak #624: bleak.exc.BleakError: Multiple Characteristics with this UUID, refer to your desired characteristic by the `handle` attribute instead.

Andrei-Iosifescu123, 2021-08-13:
* bleak version: 0.12.1
* Python version: Python 3.9.5
* Operating System: Windows 10 21H1
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
I got the below error when trying to write 0x02 to a characteristic:
`Traceback (most recent call last):
  File "C:\Users\User\Desktop\test.py", line 36, in <module>
    loop.run_until_complete(run(ADDRESS, True))
  File "C:\python\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "C:\Users\User\Desktop\test.py", line 30, in run
    await client.write_gatt_char(ALERT_CHARACTERISTIC, b"%s" % (t))
  File "C:\python\lib\site-packages\bleak\backends\winrt\client.py", line 631, in write_gatt_char
    characteristic = self.services.get_characteristic(char_specifier)
  File "C:\python\lib\site-packages\bleak\backends\service.py", line 179, in get_characteristic
    raise BleakError(
bleak.exc.BleakError: Multiple Characteristics with this UUID, refer to your desired characteristic by the `handle` attribute instead.`

### What I Did

```
C:\Users\User\Desktop>python test.py
Connected: True
Alerting...
Traceback (most recent call last):
  File "C:\Users\User\Desktop\test.py", line 36, in <module>
    loop.run_until_complete(run(ADDRESS, True))
  File "C:\python\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "C:\Users\User\Desktop\test.py", line 30, in run
    await client.write_gatt_char(ALERT_CHARACTERISTIC, b"%s" % (t))
  File "C:\python\lib\site-packages\bleak\backends\winrt\client.py", line 631, in write_gatt_char
    characteristic = self.services.get_characteristic(char_specifier)
  File "C:\python\lib\site-packages\bleak\backends\service.py", line 179, in get_characteristic
    raise BleakError(
bleak.exc.BleakError: Multiple Characteristics with this UUID, refer to your desired characteristic by the `handle` attribute instead.
```

dlech, 2021-08-13:
Did you try what the error message suggests?

> bleak.exc.BleakError: Multiple Characteristics with this UUID, refer to your desired characteristic by the `handle` attribute instead.

Andrei-Iosifescu123, 2021-08-13:
> Did you try what the error message suggests?
> 
> > bleak.exc.BleakError: Multiple Characteristics with this UUID, refer to your desired characteristic by the `handle` attribute instead.

How do I get the handle attribute?

dlech, 2021-08-14:
First, get the `BleakGATTService` object for the service you want, then from the service, get the characteristic. The `BleakGATTCharacteristic` object has the `handle` attribute in question.

Andrei-Iosifescu123, 2021-08-14:
> First, get the `BleakGATTService` object for the service you want, then from the service, get the characteristic. The `BleakGATTCharacteristic` object has the `handle` attribute in question.

How do I do that?