hbldh/bleak #626: Receiving all advertisements from multiple sensors

fahol28, 2021-08-19:
* bleak version: 0.11.0
* Python version: 3
* Operating System: Raspberry pi
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.5

### Description

I try to gather all advertisement data from many ble Sensors (ruuvi tags). 
This data includes environmental data such as temperature, humidity and pressure.
I use bleak because i need to config sensors and need to connect to each of them for specific actions.

### What I Did
In my first approach i use:
```
import asyncio
from bleak import BleakScanner

mac = []

class Event_ts(asyncio.Event):
    def clear(self):
        self._loop.call_soon_threadsafe(super().clear)

    def set(self):
        self._loop.call_soon_threadsafe(super().set)

stopEvent = Event_ts()

def detection_callback(device, advertisement_data, adress):
    #parse data here
    print("%s" % hexlify(advertisement_data))
    print(adress)
    if _get_data_format_5(advertisement_data):
        decoder = gateway.AdvertisementDecoder.get_decoder(5)
        adv_data = decoder.decode_data(advertisement_data)

async def run():
    scanner = BleakScanner(filters={"UUIDs": ["6E400001-B5A3-F393-E0A9-E50E24DCCA9E"]})    
    scanner.set_scanning_filter()
    scanner.register_detection_callback(detection_callback)
    await scanner.start()
    await asyncio.sleep(80.0)
    await scanner.stop()
```

I get the advertisements but only every 10 seconds what is quite to low.

So I change my approach to:
```
import  asyncio

from bleak import BleakScanner
mac = []
class Event_ts(asyncio.Event):
    def clear(self):
        self._loop.call_soon_threadsafe(super().clear)

    def set(self):
        self._loop.call_soon_threadsafe(super().set)

stopEvent = Event_ts()

def detection_callback(device, advertisement_data, adress):
    #parse data here
    print("%s" % hexlify(advertisement_data))
    print(adress)
    if _get_data_format_5(advertisement_data):
        decoder = gateway.AdvertisementDecoder.get_decoder(5)
        adv_data = decoder.decode_data(advertisement_data)

def validate_mac(devices):
    print("validate")
    for i in devices:
        # self.logger.info('Device: %s with Address %s found!' % (i.name, i.address))
        if ("Ruuvi" in i.name) & (i.address not in mac):
            print("found")
            mac.append(i.address)
            print(mac)

async def find_tags():
    try:
        #find devices
        devices = await BleakScanner.discover(timeout=5.0)
        validate_mac(devices)
     if len(mac) > 0:
            for i in mac:
                async with BleakClient(i) as client:
                    #start notify advertisements. 
                    await client.start_notify("6E400003-B5A3-F393-E0A9-E50E24DCCA9E", detection_callback)
                    await stopEvent.wait()
    except Exception as e:
        print("Error: {}".format(e))
        client.stop_notify("6E400003-B5A3-F393-E0A9-E50E24DCCA9E")
    # When Ctrl+C is pressed execution of the while loop is stopped
    except KeyboardInterrupt:
        client.stop_notify()
        stopEvent.set()
        print('Exit')

```
This gets all the advertisement data, but i'm not able to identify from which sensor they are collected.

 How do I get all the advertisements with the macs or an equal identifier from the sensors?

I hope you can help me

dlech, 2021-08-19:
The `device` object in `detection_callback()` has an `address` attribute that is used to uniquely identify the device.

fahol28, 2021-08-19:
In the first approach this is correct. Here i need every data point not only once every 10 seconds.

In the second approach i use :
`start_notify`
and there it is only an int which identify the handler
```
async def start_notify(
        self,
        char_specifier: Union[BleakGATTCharacteristic, int, str, uuid.UUID],
        callback: Callable[[int, bytearray], None],
        **kwargs
    ) -> None:
```

dlech, 2021-08-19:
See https://github.com/hbldh/bleak/discussions/467 for an idea.

fahol28, 2021-08-20:
This solution works great. Thank you for your help.

closed 2021-08-20 by fahol28