hbldh/bleak #627: Notify 2 or more characteristics of the same BLE device.

pauloesteban, 2021-08-20:
* bleak version: 0.12.1
* Python version: 3.8.11
* Operating System: macOS 11.5.2
* BlueZ version (`bluetoothctl -v`) in case of Linux: N/A

### Description

Read 2 or more characteristics in the Bleak Client context for the same device.

### What I Did

Based on _enable_notifications.py_ example, I just simply added `start_notify` and `stop_notify` for another characteristic UUID. I also tried `async.gather` for both `start_notify`.

It is only printing for the first characteristic.

```
import sys
import logging
import asyncio
import platform

from bleak import BleakClient
from bleak import _logger as logger

CHARACTERISTIC_UUID_1 = "f000aa65-0451-4000-b000-000000000000"  # <--- Change to the characteristic you want to enable notifications from.
CHARACTERISTIC_UUID_2 = "e000aa65-0451-4000-b000-000000000000"  # <--- Change to the characteristic you want to enable notifications from.
ADDRESS = (
    "24:71:89:cc:09:05"  # <--- Change to your device's address here if you are using Windows or Linux
    if platform.system() != "Darwin"
    else "B9EA5233-37EF-4DD6-87A8-2A875E821C46"  # <--- Change to your device's address here if you are using macOS
)

def notification_handler(sender, data):
    """Simple notification handler which prints the data received."""
    print("{0}: {1}".format(sender, data))

async def run(address, debug=False):
    if debug:
        import sys

        l = logging.getLogger("asyncio")
        l.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        l.addHandler(h)
        logger.addHandler(h)

    async with BleakClient(address) as client:
        logger.info(f"Connected: {client.is_connected}")

        await client.start_notify(CHARACTERISTIC_UUID_1, notification_handler)
        await client.start_notify(CHARACTERISTIC_UUID_2, notification_handler)
        await asyncio.sleep(5.0)
        await client.stop_notify(CHARACTERISTIC_UUID_1)
        await client.stop_notify(CHARACTERISTIC_UUID_2)

if __name__ == "__main__":
    import os

    os.environ["PYTHONASYNCIODEBUG"] = str(1)
    loop = asyncio.get_event_loop()
    # loop.set_debug(True)
    loop.run_until_complete(run(ADDRESS, True))
```

dlech, 2021-08-20:
Can we be sure the problem is not with the device? Have you tried the suggested [troubleshooting](https://bleak.readthedocs.io/en/latest/troubleshooting.html)?

pauloesteban, 2021-08-20:
Hi @dlech,

Thanks for your suggestion. Running with `export BLEAK_LOGGING=1` for about 10 seconds.

```
...
2021-08-20 11:37:17,122 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device B9EA5233-37EF-4DD6-87A8-2A875E821C46: AM1V330 @ RSSI: -47 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fa469670040>
2021-08-20 11:37:17,123 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device B9EA5233-37EF-4DD6-87A8-2A875E821C46: AM1V330 @ RSSI: -47 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fa469670040>
2021-08-20 11:37:17,123 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
2021-08-20 11:37:17,124 bleak.backends.corebluetooth.client DEBUG: CentralManagerDelegate  at <CentralManagerDelegate: 0x7fa46966eb10>
2021-08-20 11:37:17,124 bleak.backends.corebluetooth.client DEBUG: Connecting to BLE device @ B9EA5233-37EF-4DD6-87A8-2A875E821C46
2021-08-20 11:37:17,421 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didConnectPeripheral_
2021-08-20 11:37:17,422 bleak.backends.corebluetooth.client DEBUG: Retrieving services...
...
...
2021-08-20 11:37:18,732 bleak.backends.corebluetooth.client DEBUG: Services resolved for BleakClientCoreBluetooth (B9EA5233-37EF-4DD6-87A8-2A875E821C46)
2021-08-20 11:37:18,768 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didUpdateNotificationStateForCharacteristic_error_
2021-08-20 11:37:18,769 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Character Notify Update
2021-08-20 11:37:18,806 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didUpdateNotificationStateForCharacteristic_error_
2021-08-20 11:37:18,806 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Character Notify Update
2021-08-20 11:37:18,807 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2021-08-20 11:37:18,863 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2021-08-20 11:37:18,919 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didUpdateValueForCharacteristic_error_
...
...
2021-08-20 11:37:28,841 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDisconnectPeripheral_error_
2021-08-20 11:37:28,841 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Peripheral Device disconnected!
```

dlech, 2021-08-20:
Sorry, this is not enough information to determine if the problem is with the device or with Bleak. You could log Bluetooth packets to see if the device is actually sending the notifications.