hbldh/bleak #631: Connecting after scanning and the dreaded "Software caused connection abort"

egnor, 2021-09-02:
* bleak version: 0.21.1
* Python version: 3.9.5
* Operating System: Ubuntu 21.04
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.56

### Description

First of all thanks :pray: to all of you for a very thoughtfully designed interface!

So, this may be more of a bluez issue, but it's surfacing here with bleak and may be related to how bleak communicates with bluez over dbus. I am attempting to communicate with a BTLE device (a little Bluetooth nametag). It's a very simple device that exposes one characteristic in one service. I am using Ubuntu 21.04, bluez 5.56, and bleak 0.21.1. The hardware is a Lenovo Thinkpad X1 Carbon 9th gen with some sort of Intel bluetooth controller (not sure how to get its precise identit).

Very frequently when connecting to the device, I get the infamous "Software caused connection abort" error relayed from bluez. (I gather that's something of a generic catch-all for connection failures?) However, I find that I can change the frequency of this error significantly by how I do things!

* create a `BleakClient` with a specified address (which I believe runs a discover process under the covers) -- I will get the error maybe 9 out of 10 times :frowning_face: 
* run my own `BleakScanner` until the device with the address of interest shows up, then stop scanning and pass that device to `BleakClient` -- I will get the error maybe 5 out of 10 times :neutral_face: 
* run my own `BleakScanner` until the device shows up, and _keep scanning_ while connecting to the device -- I will get the error maybe 1 out of 10 times :slightly_smiling_face: 

By adding some retry logic on top of this, this becomes mostly acceptable! (And once connected, I'm able to communicate with the device without further issues. It has a very strong signal, it's right next to me on the desk.)

I can actually recreate this variation with `bluetoothctl`! (This is where you say "go talk to the bluez folks then"! But please bear with me.) If I try to scan-on, then scan-off, then connect, I often get the error:

```
[bluetooth]# connect FF:FF:22:02:9D:75
Attempting to connect to FF:FF:22:02:9D:75
[CHG] Device FF:FF:22:02:9D:75 Connected: yes
Failed to connect: org.bluez.Error.Failed
[CHG] Device FF:FF:22:02:9D:75 Connected: no
```

Note that it reports as connected just before getting the error -- I see this in the messages with bleak, as well. However, if I keep scanning running while connecting, things usually work much better!

```
[bluetooth]# scan on
Discovery started
[CHG] Controller A8:64:F1:31:DB:10 Discovering: yes
[NEW] Device 57:6B:11:A6:AF:CC 57-6B-11-A6-AF-CC
  [... various other new-device reports elided for brevity ...]
[NEW] Device FF:FF:22:02:9D:75 CoolLED
[bluetooth]# connect FF:FF:22:02:9D:75
Attempting to connect to FF:FF:22:02:9D:75
[CHG] Device FF:FF:22:02:9D:75 Connected: yes
Connection successful
[NEW] Primary Service (Handle 0x4bb5)
	/org/bluez/hci0/dev_FF_FF_22_02_9D_75/service0001
	0000fff0-0000-1000-8000-00805f9b34fb
	Unknown
[NEW] Characteristic (Handle 0xb9c4)
	/org/bluez/hci0/dev_FF_FF_22_02_9D_75/service0001/char0002
	0000fff1-0000-1000-8000-00805f9b34fb
	Unknown
[NEW] Descriptor (Handle 0x02b4)
	/org/bluez/hci0/dev_FF_FF_22_02_9D_75/service0001/char0002/desc0004
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[CHG] Device FF:FF:22:02:9D:75 UUIDs: 0000fff0-0000-1000-8000-00805f9b34fb
[CHG] Device FF:FF:22:02:9D:75 ServicesResolved: yes
```

I notice that bluez seems to keep a table of devices, and once scanning is turned off, they age out pretty quickly -- 10 or 20 seconds later, I see the device records getting deleted. Maybe this is related somehow, even if the table entry is still there when connection is attempted, it's "stale" in some way?

Anyway, coming back to bleak, it seems that bleak's default M.O. is to run a discovery to get the device info, and then make a connection request, but this doesn't seem to align with what makes bluez happy. I'm actually having trouble finding official documentation on exactly what bluez expects from clients (pointers welcome!) so I'm not sure who's "in the right" here. I have my workaround for now but it feels fragile to me...

Tips and advice? Happy to go off to the bluez folks with a bug report, if their behavior is definitely not per spec.

egnor, 2021-09-18:
Adjacent and maybe related: https://github.com/hbldh/bleak/issues/377

dlech, 2021-10-06:
Could it be that this name tag device goes to sleep as soon as scanning stops? This could be why it isn't connectable after scanning stops.