hbldh/bleak #632: WinRT backend crashes on start_notify

rajesh-krishnan, 2021-09-04:
* bleak version:  0.12.1
* Python version:  3.9..6
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: N/A

### Description

I am trying to set up serial communications to the Bluno Mega 2560, which involves writing to a characteristic and subscribing to notifications on the same characteristic. Writing to the characteristic successfully sends characters to the peripheral. Calling start_notify to register a notification callback on the characteristic causes the WinRT backend to crash. 

The errors are reproduced by running  examples/enable_notifcations.py.  I have confirmed the characteristic supports notify by running examples/service_explorer.py. 

### What I Did

First I provide the results of running the service explorer:
```
> py service_explorer.py 30:E2:83:AD:99:01
Connected: True
[Service] 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
        [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2):  (read), Value: b'BlunoMega'
        [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4):  (read), Value: b'\x00\x00'
        [Characteristic] 00002a02-0000-1000-8000-00805f9b34fb (Handle: 6):  (read,write), Value: b'\x00'
        [Characteristic] 00002a03-0000-1000-8000-00805f9b34fb (Handle: 8):  (write), Value: None
        [Characteristic] 00002a04-0000-1000-8000-00805f9b34fb (Handle: 10):  (read), Value: b'P\x00\xa0\x00\x00\x00\xe8\x03'
[Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 12): Generic Attribute Profile
        [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 13):  (indicate), Value: None
                [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 15): Client Characteristic Configuration) | Value: b'\x02\x00'
[Service] 0000180a-0000-1000-8000-00805f9b34fb (Handle: 16): Device Information
        [Characteristic] 00002a23-0000-1000-8000-00805f9b34fb (Handle: 17):  (read), Value: b'\x01\x99\xad\x00\x00\x83\xe20'
        [Characteristic] 00002a24-0000-1000-8000-00805f9b34fb (Handle: 19):  (read), Value: b'DF Bluno'
        [Characteristic] 00002a25-0000-1000-8000-00805f9b34fb (Handle: 21):  (read), Value: b'0123456789'
        [Characteristic] 00002a26-0000-1000-8000-00805f9b34fb (Handle: 23):  (read), Value: b'FW V1.97'
        [Characteristic] 00002a27-0000-1000-8000-00805f9b34fb (Handle: 25):  (read), Value: b'HW V1.7'
        [Characteristic] 00002a28-0000-1000-8000-00805f9b34fb (Handle: 27):  (read), Value: b'SW V1.97'
        [Characteristic] 00002a29-0000-1000-8000-00805f9b34fb (Handle: 29):  (read), Value: b'DFRobot'
        [Characteristic] 00002a2a-0000-1000-8000-00805f9b34fb (Handle: 31):  (read), Value: b'\xfe\x00experimental'
        [Characteristic] 00002a50-0000-1000-8000-00805f9b34fb (Handle: 33):  (read), Value: b'\x01\r\x00\x00\x00\x10\x01'
[Service] 0000dfb0-0000-1000-8000-00805f9b34fb (Handle: 35): Vendor specific
        [Characteristic] 0000dfb1-0000-1000-8000-00805f9b34fb (Handle: 36): Serial (read,write-without-response,write,notify), Value: b'\n'
                [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 38): Characteristic User Description) | Value: b'Serial'
        [Characteristic] 0000dfb2-0000-1000-8000-00805f9b34fb (Handle: 39): AT Ctrl (read,write-without-response,write,notify), Value: b'\x02'
                [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 41): Characteristic User Description) | Value: b'AT Ctrl'
```
The characteristic of interest is '0000dfb1-0000-1000-8000-00805f9b34fb' which supports write and notify. Write and read work, but read only gets what the central wrote last. The notify is needed to get what the peripheral sends. This is their hack of a serial link over BLE with limited documentation.

Here is the error I get when I run the enable notifications script:

```
>py enable_notifications.py 30:E2:83:AD:99:01 0000dfb1-0000-1000-8000-00805f9b34fb
Traceback (most recent call last):
  File "C:\Users\r\workmirror\projects\sediment\design\seec\repo\seec\python\bleak\enable_notifications.py", line 49, in <module>
    loop.run_until_complete(run(ADDRESS, True))
  File "C:\Users\r\AppData\Local\Programs\Python\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "C:\Users\r\work\enable_notifications.py", line 35, in run
    async with BleakClient(address) as client:
  File "C:\Users\r\AppData\Local\Programs\Python\Python39\lib\site-packages\bleak\backends\client.py", line 61, in __aenter__
    await self.connect()
  File "C:\Users\r\AppData\Local\Programs\Python\Python39\lib\site-packages\bleak\backends\winrt\client.py", line 227, in connect
    await self.get_services(use_cached=use_cached)
  File "C:\Users\r\AppData\Local\Programs\Python\Python39\lib\site-packages\bleak\backends\winrt\client.py", line 424, in get_services
    characteristics_result = await service.get_characteristics_async(
RuntimeError
```
I have confirmed that `characteristic_obj.write_client_characteristi c_configuration_descriptor_async()` is getting passed ` GattClientCharacteristicConfigurationDescriptorValue.NOTIFY` . Skipping this step makes the program proceed without errors but the callback does not get called. 

This problem could be peripheral specific, and perhaps the descriptor does not want to be written to and the notify is always on.   I suspect many other peripherals that implement a modified BLE stack would have similar crashes on unauthorized writes. 

Will appreciate any pointers, advice, and any insights from anyone who is using Bleak/WinRT with Bluno Mega.

rajesh-krishnan, 2021-09-04:
Unfortunately the root cause appears to be the missing 2902 descriptor required by the BLE spec  as per this forum thread.
[https://www.dfrobot.com/forum/viewtopic.php?f=18&t=2035&start=10](https://www.dfrobot.com/forum/viewtopic.php?f=18&t=2035&start=10)

DFRobot claims that their notification is always on. Effectively it will only work in tolerant environments that will take a notification even if the client has not subscribed. With the exception of iOS and Android, most platforms do not provide a mechanism to set up a notification callback without using the 2902 descriptor to subscribe. 

Not clear what bleak can do about this, so closing the issue.  Will still appreciate advice from anyone who is using Bleak and Bluno.

closed 2021-09-04 by rajesh-krishnan

SuibianP, 2023-08-29:
For backends that are able to take notifications regardless of subscription status, the workaround is to ignore exceptions from `start_notify`/`stop_notify`. It works on my MacBook and should work on iOS and the Android backend as well.

SuibianP, 2023-09-13:
I also tested it working on Linux/BlueZ.
@dlech @hbldh Would it be good if this workaround is implemented as a parameter maybe for supported backends?

dlech, 2023-09-13:
Please see https://github.com/hbldh/bleak/issues/972#issuecomment-1491228424