hbldh/bleak #635: No results when scanning on MacOS 12

peterloron, 2021-09-08:
* bleak version: 0.12.1
* Python version: 3.9.6
* Operating System: MacOS 12 Beta (Monterey)

### Description

I installed bleak via pip, and used the scanning example from the docs. When the code is run, no devices are found. I can run a different BLE scanner and see many devices. Running as root doesn't change anything.

Happy to help investigate, but I'm not sure where to go from here.

dlech, 2021-09-08:
> Happy to help investigate, but I'm not sure where to go from here.

https://bleak.readthedocs.io/en/latest/troubleshooting.html

dlech, 2021-09-08:
https://github.com/hbldh/bleak/blob/develop/examples/detection_callback.py

peterloron, 2021-09-08:
Hello @dlech. Apologies, I should have mentioned I did read the troubleshooting section, open issues, etc. I enabled the debug env var, but it did not show anything interesting:
```
2021-09-08 16:00:37,577 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
2021-09-08 16:00:37,577 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
2021-09-08 16:00:37,578 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
2021-09-08 16:00:42,581 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
2021-09-08 16:00:42,581 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
...
```

The code I am running is exactly the detection_callback.py code you pointed to.

I did a packet capture, but I'm unable to determine if there is any useful information in there. I can post it somewhere if it would be useful.

dlech, 2021-10-06:
I can look at the packet capture if you attach it to the issue (just put it in a .zip file).

jonskywalker, 2021-10-23:
[bleakMacOS.pklg.zip](https://github.com/hbldh/bleak/files/7401912/bleakMacOS.pklg.zip)
Running into the same issue attached packet logger output if it helps. When enabling debug logging it's the same results reported above.
Bleak 0.13 
Python 3.10 
MacOS 12

michael-deng, 2021-10-31:
Just want to echo that I'm encountering the same issue in monterey.

oamado, 2021-11-10:
I see there are changes on https://github.com/ronaldoussoren/pyobjc, but not a new version since Jun 7, 2021. Probably this could help to solve the issue.

dlech, 2021-11-10:
What specific changes might be related to Bluetooth?

pedropombeiro, 2021-11-11:
> What specific changes might be related to Bluetooth? I have already tried https://pypi.org/project/pyobjc/8.0b1/.

@dlech pyobjc 8.0 was released 12 hours ago. In https://pyobjc.readthedocs.io/en/latest/changelog.html#new-features there's a mention to `Updated framework bindings for macOS 12`. Sounds promising...

dlech, 2021-11-12:
PyObjC v8.0 is out now and it doesn't fix the problem

kettenbach-it, 2021-11-22:
> PyObjC v8.0 is out now and it doesn't fix the problem

I can conform that.

This really needs to be fixed! It's a showstopper for me

dlech, 2021-11-22:
Please report the issue to Apple using the Feedback Assistant app. I'm not sure what else we can do since the problem seems to come from an undocumented change in the OS. If enough people report the issue, maybe they will prioritize it.

challiwill, 2021-11-23:
Can someone who understands this help me wrap my head around why I can still scan for devices using Chrome's Web Bluetooth, but can't with bleak? Is there somehow a different set of library functions exposed in c++? Is this a bug at the pyobjc level? I've been trying to read the chromium source code but must admit I'm struggling.

VladislavSmolyanoy, 2021-11-23:
> Can someone who understands this help me wrap my head around why I can still scan for devices using Chrome's Web Bluetooth, but can't with bleak? Is there somehow a different set of library [...]

I am unfamiliar with the framework that Web Bluetooth uses but bleak is built on top of Core Bluetooth which is currently broken in macOS Monterey...

kettenbach-it, 2021-11-23:
Since tools such as "LightBlue", which I like to use to develop in the BLE environment, and others work perfectly, the fault - in my humble opinion - is more likely to be with the [PyObjc Framework CoreBlueooth](https://github.com/ronaldoussoren/pyobjc/tree/master/pyobjc-framework-CoreBluetooth) or even within Bleak.

Stupid question: I digged into the code "behind" `BleakScanner.discover()`.
I see:

```
    @classmethod
    async def discover(cls, timeout=5.0, **kwargs) -> List[BLEDevice]:
        async with cls(**kwargs) as scanner:
            await asyncio.sleep(timeout)
            devices = scanner.discovered_devices
        return devices
```

and `scanner.discovered_devices`is:

```
    @property
    @abc.abstractmethod
    def discovered_devices(self) -> List[BLEDevice]:
        raise NotImplementedError()
```

How can this method ever run back a list of devices? It has no return. However, no exception is raised for me either.

kettenbach-it, 2021-11-23:
Ah, sorry. It's abstract. I need to look in the respective implementation

kettenbach-it, 2021-11-23:
I have been digging through the code on two Mac systems.
One system is an Intel Mac running Mac OS 11.6 and the other - where it doesn't work - is an M1 Mac running Mac OS 12.0.1.

In the end, it's all about the [here](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1519127-retrieveperipheralswithidentifie) documented function `retrievePeripheralsWithIdentifiers` from Apple's coreBluetooth package.
On Mac OS 11.6 this returns a list of devices and on 12.0.1 nothing. `retrieveConnectedPeripheralsWithServices`doesn't work either on 12.0.1.

What I noticed is that when I called the sample code with a BleakScanner.discover() on Mac OS 11.6, I was asked via a dialogue box if I wanted to authorise the application to use Bluetooth.
On Mac OS 12.0.1 nothing of the sort came up - **maybe we have just a permission problem here.**......

kettenbach-it, 2021-11-23:
I submitted feedback to Apple

arthurbiancarelli, 2021-11-23:
Hello, just want to share some findings because I encountered a similar problem on macOS 12 for a proprietary software I'm developing involving a C++ executable using Apple's CoreBluetooth
My executable is simply scanning Bluetooth devices and it asks the Bluetooth permissions via a dialog box (just by linking to CoreBluetooth - no code involved) on macOS 11/12.
 Permissions if granted are correctly reflected in the [Bluetooth manager authorization  status](https://developer.apple.com/documentation/corebluetooth/cbperipheralmanagerauthorizationstatus?language=objc)
 
 My problem is that on macOS 12, `didDiscoverPeripheral` is never called which seems similar to this issue
 
But if I create the executable as a [bundle](https://developer.apple.com/library/archive/documentation/CoreFoundation/Conceptual/CFBundles/BundleTypes/BundleTypes.html#//apple_ref/doc/uid/10000123i-CH101-SW1) that contains in the Info.plist manifest file the `NSBluetoothAlwaysUsageDescription` key, scanning does work.
(If I [embed the manifest in the binary file](https://reverse.put.as/2013/05/28/gimmedebugah-how-to-embedded-a-info-plist-into-arbitrary-binaries/) bluetooth scanning does not work which seems a regression from Apple's side)

So if my findings do applicate to your project maybe it could  work by adding this key in the Info.plist embedded in the python installation (for instance with the macOs system python: /usr/local/Cellar/python@3.9/3.9.7_1/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/Info.plist) 
 Creating a bundle executable using for instance py2app (not convenient of course) could also work but maybe the correct solution would be to operate directly on the pyobjc library and creating a bundle library containing a manifest instead of .so file  if Apple does not reply

Anyway this is a bug from Apple imho because even if the permission system hardens, the error should be clear (a crash for instance for Privacy violation like usual)  but not silently fails

psifertex, 2021-11-23:
Potential work-around: https://github.com/sonsongithub/SBMeter/commit/aabf32c0cbe3ff05305701339665d38cbd1597c9#diff-8e1d3fc4aff84e0803e9a73332cdf232fed0292d4ef493c00ad3a482d7b4baf9R45-R47

It appears that maybe filtering by specific services is another work-around for some use-cases (issue was referenced above but making it clear that is a work-around might help some folks)

Also, fwiw, the app I'm trying to get working (which uses a different btle library -- it uses https://github.com/deviceplug/btleplug/issues/224 which is already referenced in this thread -- but I'm tracking this issue to see about potential fixes) has that `NSBluetoothAlwaysUsageDescription` field in the Info.plist and it doesn't work still on Monterey so there might be multiple bugs here.

challiwill, 2021-11-23:
I haven't been able to get the workaround to work for me, but like I mentioned before I can connect to my same device via Web Bluetooth (which is also built on top of Core Bluetooth because as far as I understand everything on mac has to be). If anyone else also wants to look at the source of that, the Chromium implementation is here https://chromium.googlesource.com/chromium/src/+/refs/heads/main/device/bluetooth

dan1916, 2021-11-24:
I can also confirm that bleak scanning does not work on Mac OS 12.0.1. The Web Bluetooth still works.

challiwill, 2021-11-25:
> I haven't been able to get the workaround to work for me, but like I mentioned before I can connect to my same device via Web Bluetooth (which is also built on top of Core Bluetooth because as far [...]

I have actually been able to get it to work by setting both the `service_uuids` and the `options` to `{CBCentralManagerScanOptionAllowDuplicatesKey: objc.YES}`.

kettenbach-it, 2021-11-25:
> EDIT: I have actually been able to get it to work setting `service_uuids`. 

What does that mean? Can you please explain this to me in more detail!

hbldh, 2021-11-26:
This is indeed problematic but, as has been stated above, this is most proabably something Bleak cannot address without Apple fixing this and pushing an update for macOS 12. If there is a problem in macOS, I do not believe that there are any reliable workarounds, given that it seems like the detection events that Bleak are using are not sent by Core Bluetooth.

I do not have any access to any macOS 12 device, so I will not be able to do any troubleshooting here. But please, keep escalating the findings you make to Apple. They will have little effect in this issue I am afraid.

psifertex, 2021-11-26:
I would re-iterate the point made above that Chrome still works fine for discovering devices via the Web Bluetooth API without a service filter. Clearly there is _some_ path that works just fine on MacOS 12 even if the current APIs used by bleak are not functioning in the same way.

kettenbach-it, 2021-11-26:
I still believe, that this is a permission problem combined with the bug, that the Apple API doesn't give an error message if the caller does not have permission.

Is still do not get what @challiwill meant with his comment https://github.com/hbldh/bleak/issues/635#issuecomment-977147483.  Does some one know?

dlech, 2021-11-26:
> Is still do not get what @challiwill meant with his comment [#635 (comment)](https://github.com/hbldh/bleak/issues/635#issuecomment-977147483). Does some one know?

While I was researching this issue, I came across for forum post that suggested this as well. The suggestion is to provide a non-empty list of service uuids to [scanForPeripheralsWithServices:options:](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipheralswithservices?language=objc). However, I'm pretty sure I tried this without results. 

Chromium also passes a non-emtpy list here, so that is, in fact, one of the differences between Chromium and Bleak. The other difference is that Chromium is an app and therefor could have the suggested permissions in it's pinfo.list.

kettenbach-it, 2021-11-26:
When I digged into the bleak code, I think I found that bleak uses `retrievePeripheralsWithIdentifiers`but not `scanForPeripheralsWithServices`

See here:

https://github.com/hbldh/bleak/blob/e492699afa3b37b820eabc2640092d08db4da078/bleak/backends/corebluetooth/scanner.py#L117

Maybe it *should* use  scanForPeripheralsWithServices instead? (Besides that is actually something different....)

dlech, 2021-11-26:
It is here:
https://github.com/hbldh/bleak/blob/fc07589d8a4e91276e6b0331bdebb83fd87577a8/bleak/backends/corebluetooth/CentralManagerDelegate.py#L117-L119

kettenbach-it, 2021-11-26:
@dlech Are you sure, that this is the method that is called, when doing `BleakScanner.discover()`?
My backtrace leads to 

https://github.com/hbldh/bleak/blob/e492699afa3b37b820eabc2640092d08db4da078/bleak/backends/corebluetooth/scanner.py#L117

because this class implements the abstract class 

https://github.com/hbldh/bleak/blob/e492699afa3b37b820eabc2640092d08db4da078/bleak/backends/corebluetooth/scanner.py#L17

given in class

`BaseBleakScanner`:

as

```
    @property
    @abc.abstractmethod
    def discovered_devices(self) -> List[BLEDevice]:
        raise NotImplementedError()
```

I don't know what https://github.com/hbldh/bleak/blob/fc07589d8a4e91276e6b0331bdebb83fd87577a8/bleak/backends/corebluetooth/CentralManagerDelegate.py#L107 is good for, but I don't find it in the path if I simply trace back `BleakScanner.discover()` - which is the method that is not using in my case

dlech, 2021-11-26:
> Are you sure, that this is the method that is called, when doing `BleakScanner.discover()`?

100% sure. The `aysnc with` statement in`discover()` implicitly calls `__aenter__()` which calls `start()` which calls `start_scan()`.

challiwill, 2021-11-27:
> While I was researching this issue, I came across for forum post that suggested this as well. The suggestion is to provide a non-empty list of service uuids to [scanForPeripheralsWithServices:options:](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipheralswithservices?language=objc). Bleak currently passes `None` to that method. However, I'm pretty sure I tried providing a list that matched my device without results.

@dlech I only monkey patched this into bleak [at this line](https://github.com/hbldh/bleak/blob/develop/bleak/backends/corebluetooth/CentralManagerDelegate.py#L108), like this:
```
scan_options["service_uuids"] = ["6E400001-B5A3-F393-E0A9-E50E24DCCA9E".lower()]   
```

dlech, 2021-11-27:
Awesome, I must have done something wrong when I tried it. I've made a PR to make it easier to pass a list of service uuids as a `service_uuids` kwarg, e.g. `BleakScanner(service_uuids=[...])` or `BleakScanner.discover(service_uuids=[...])`.

Try it with:

    pip install https://github.com/hbldh/bleak/archive/refs/heads/macos12-scanning.zip

challiwill, 2021-11-28:
> The other difference is that Chromium is an app and therefor could have the suggested permissions in it's pinfo.list.

If I'm reading [the docs](https://developer.apple.com/documentation/bundleresources/information_property_list/nsbluetoothalwaysusagedescription) correctly, the `NSBluetoothAlwaysUsageDescription` plist key is not available on macOS, so unless that changed but wasn't updated in the docs it doesn't seem likely to me to be the culprit.

I'm wondering about this line here because it matches the behavior we're seeing, but there's nowhere that bleak sets bluetooth scanning to background mode as far as I can tell, unless somehow it happens implicitly:
```
Your app can scan for Bluetooth devices in the background by specifying the bluetooth-central
background mode. To do this, your app must explicitly scan for one or more services by specifying
them in the serviceUUIDs parameter. The CBCentralManager scan option has no effect while
scanning in the background.
```
[last paragraph here](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipheralswithservices?language=objc#discussion)

challiwill, 2021-11-28:
@dlech Then again, the `NSBluetoothAlwaysUsageDescription` is indeed present in a [chromium/content/shell/app/app-Info.plist](https://chromium.googlesource.com/chromium/src/+/refs/heads/main/content/shell/app/app-Info.plist#29).

challiwill, 2021-11-28:
I've now tried building my app using [py2app](https://github.com/ronaldoussoren/py2app) and setting `NSBluetoothAlwaysUsageDescription` in the plist and I cannot get it to work still. 

I have an Info.plist:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>NSBluetoothAlwaysUsageDescription</key>
        <string>reasons</string>
</dict>
</plist>
```
and my setup.py is:
```
from setuptools import setup

setup(
    app=["app.py"],
    data_files=[],
    options=dict(py2app=dict(
        plist="Info.plist",
    )),
    setup_requires=["py2app"],
)
```

@arthurbiancarelli If you're able to provide any more information about how you got this to work that would be helpful.

arthurbiancarelli, 2021-11-29:
I have created a minimal test application using pure Objective-C that scans Bluetooth devices. This application only works when compiling in the Apple Bundle format and with  `NSBluetoothAlwaysUsageDescription`  in the manifest (and also for Intel architecture, see below) Since the standard output is not simply accessible when executing a bundle application, I've made a simple GUI that displays the scanning results.
You can check out the code [here](https://github.com/arthurbiancarelli/test_ble_scanner_macOS)

I then created a minimal GUI application using `bleak`, `tkinter` and `py2app`  -> https://github.com/arthurbiancarelli/bleak/commits/ab/macOs12/1
Same results. The application works when built as a bundle for x86 architecture. Does not work when launched from the Python interpreter

I tried using a custom recompiled terminal (iTerm2) / Python Interpreter  that has the amended manifest but no luck so far.

I also tried the service filter trick that some reported but it does not work on my system. However it seems related to the iOS restrictions on background scanning
From the [Apple documentation](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipherals)
> Your app can scan for Bluetooth devices in the background by specifying the bluetooth-central background mode. To do this, your app must explicitly scan for one or more services by specifying them in the serviceUUIDs parameter

Maybe it does work with BLE devices that advertise their services without the need for an active scan as explained in this article (https://medium.com/@cbartel/ios-scan-and-connect-to-a-ble-peripheral-in-the-background-731f960d520d)  ?

-------------------------
_Note for Apple M1 the application must be signed with `codesign` if built natively (for arm architecture).  Compiling for x86_64 and executing within Rosetta emulator does not require this signing stage 
From the [release notes](https://developer.apple.com/documentation/macos-release-notes/macos-big-sur-11_0_1-universal-apps-release-notes)_
> New in macOS 11 on Macs with Apple silicon, and starting in macOS Big Sur 11 beta 6, the operating system enforces that any executable must be signed before it’s allowed to run.

dlech, 2021-11-29:
> I also tried the service filter trick that some reported but it does not work on my system

Can you share your raw advertising data (you can get this, for example, with the nRF Connect app for iOS or Android, or by logging Bluetooth packets on macOS). So far I've only tested with 128-bit UUIDs, so I wonder if we may need to make a tweak for 16-bit and 32-bit UUIDs.

arthurbiancarelli, 2021-11-29:
The device I'm working on has an advertising packet comprised strictly of the advertisement type, the manufacturer name and the shortened device name.
I have performed some more tests and if I add a service UUID into the advertisement packet I can scan the device both with bleak #692 and my executable (without the bundle + manifest and by manually specifying the expected Bluetooth service). 

So as I understand it would seem that macOS Monterey only performs a passive Bluetooth scan if the application is not executing in the "Apple context" (meaning a GUI compiled as a Bundle) similarly to the iOS restrictions previously discussed

This is kind of a good news but not completely satisfactory because it requires some firmware modifications for Bluetooth devices that do not advertise their services if one want to scan it via a CLI

lobodpav, 2021-12-05:
> Awesome, I must have done something wrong when I tried it. I've made a PR to make it easier to pass a list of service uuids as a `service_uuids` kwarg, e.g. `BleakScanner(service_uuids=[...])` or `BleakScanner.discover(service_uuids=[...])`.
> 
> Try it with:
> 
> ```
> pip install https://github.com/hbldh/bleak/archive/refs/heads/macos12-scanning.zip
> ```
> 
> You can also pass UUIDs to the detection callback example:
> 
> ```
> python examples/detection_callback.py 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
> ```
Scanning works when using your release. Connecting to the device fails, though:
```
Devices:  [37C6EE65-C04E-E15F-23E5-6BA7283DAD3A: Technic Hub]
Device details:  <CBPeripheral: 0x6000003b0000, identifier = 37C6EE65-C04E-E15F-23E5-6BA7283DAD3A, name = Technic Hub, mtu = 0, state = disconnected>
2021-12-06 00:13:45.489 Python[67881:7734022] -[CBCharacteristic handleCharacteristicsDiscovered:]: unrecognized selector sent to instance 0x6000010a85a0
2021-12-06 00:13:45.489 Python[67881:7734022] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[CBCharacteristic handleCharacteristicsDiscovered:]: unrecognized selector sent to instance 0x6000010a85a0'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007ff80f442e5b __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007ff80f1a3b9d objc_exception_throw + 48
	2   CoreFoundation                      0x00007ff80f4c7a2b -[NSObject(NSObject) __retain_OA] + 0
	3   CoreFoundation                      0x00007ff80f3a9d09 ___forwarding___ + 1406
	4   CoreFoundation                      0x00007ff80f3a96f8 _CF_forwarding_prep_0 + 120
	5   CoreBluetooth                       0x00007ff8210e3990 -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 144
	6   CoreBluetooth                       0x00007ff8210e3a81 -[CBPeripheral handleServiceEvent:serviceSelector:delegateSelector:delegateFlag:] + 113
	7   CoreBluetooth                       0x00007ff8210dfdd2 -[CBPeripheral handleMsg:args:] + 466
	8   CoreBluetooth                       0x00007ff8210ccd57 -[CBCentralManager handleMsg:args:] + 183
	9   CoreBluetooth                       0x00007ff821108141 -[CBManager xpcConnectionDidReceiveMsg:args:] + 177
	10  CoreBluetooth                       0x00007ff8210f8142 __30-[CBXpcConnection _handleMsg:]_block_invoke + 66
	11  libdispatch.dylib                   0x00007ff80f148ad8 _dispatch_call_block_and_release + 12
	12  libdispatch.dylib                   0x00007ff80f149cc9 _dispatch_client_callout + 8
	13  libdispatch.dylib                   0x00007ff80f14fcee _dispatch_lane_serial_drain + 696
	14  libdispatch.dylib                   0x00007ff80f1507fb _dispatch_lane_invoke + 417
	15  libdispatch.dylib                   0x00007ff80f14fb85 _dispatch_lane_serial_drain + 335
	16  libdispatch.dylib                   0x00007ff80f1507c8 _dispatch_lane_invoke + 366
	17  libdispatch.dylib                   0x00007ff80f15a7e1 _dispatch_workloop_worker_thread + 758
	18  libsystem_pthread.dylib             0x00007ff80f2fd08f _pthread_wqthread + 326
	19  libsystem_pthread.dylib             0x00007ff80f2fc01b start_wqthread + 15
)
libc++abi: terminating with uncaught exception of type NSException
```

dlech, 2021-12-05:
Can you please start a new issue for the connection problem?

Jakob-Koschel, 2021-12-07:
I tried https://github.com/hbldh/bleak/tree/macos12-scanning but I'm not sure where to get the correct service UUIDs.
I'm very inexperienced with this kind of bluetooth things, but I couldn't find any documentation on how to find service UUIDS. If I understood correctly you need to have already established a connection to that device to have a service UUID?
Thanks!

lobodpav, 2021-12-07:
> I tried https://github.com/hbldh/bleak/tree/macos12-scanning but I'm not sure where to get the correct service UUIDs.
> I'm very inexperienced with this kind of bluetooth things, but I couldn't find [...]

You can use the `nRF Connect` mobile app fo find UUIDs around you. Start scanning and make the BLE device advertise itself.

kettenbach-it, 2021-12-07:
Either I don't understand something, or this is a chicken-and-egg problem:

I want to use discover() to display the devices in the vicinity to see what services they offer.... If I have to use nRF Connect for this, I don't need discover().... in bleak.

Can someone clarify that!?

dlech, 2021-12-07:
As of right now, it is no longer possible to discover arbitrary devices using Bleak on macOS 12, so you will have to go with an alternate solution.

closed 2021-12-07 by dlech

kettenbach-it, 2021-12-08:
Okay, thx!

mkoderer, 2021-12-30:
> Can you share your raw advertising data (you can get this, for example, with the nRF Connect app for iOS or Android, or by logging Bluetooth packets on macOS). So far I've only tested with 128-bit [...]

The current implementation works also with 16-bit. For 16-bit the mac implementation will automatically extend it by the default 128-bit mask:

`> python examples/detection_callback.py 0xFE07`
`2021-12-30 08:14:49,474 __main__ INFO: ... service_uuids=['0000fe07-0000-1000-8000-00805f9b34fb`

dlech, 2021-12-30:
Thanks for sharing. I'm surprised it works even works from the command line. :+1:

mkoderer, 2021-12-30:
Not sure if this is already known here but Bluetility UI works on my mac without any specific service uuids.
I have no idea what's actually the difference since the code looks quite similar:

https://github.com/jnross/Bluetility/blob/2c91f93ba0715c354d5cdedbbf868cc2d5160d40/Bluetility/Scanner.swift#L36

dlech, 2021-12-30:
I think the difference is that it is an app (and possibly in addition to that it [requests permission](https://github.com/jnross/Bluetility/blob/ea374e8cfd760f1041259689bc70e5f43eec396f/Bluetility/Info.plist#L33)) and is not a command line tool like Python. There are several other comments in this issue that also noticed this that may be of interest.

dlech, 2022-01-10:
FYI, I've added a summary of the findings of this issue with additional insights at https://github.com/hbldh/bleak/issues/720#issuecomment-1009198158.

maximilian-kn, 2023-09-20:
Any updates on this issue? I'm currently experiencing problems trying to connect to a BLE device using a script with Bleak in Python.

dlech, 2023-09-21:
Apple fixed the bug in macOS 12.2. If you still have problems the best way to get help is to start a new discussion and include you code, error messages and logs.

PR #692 (2021-12-07): corebluetooth: work around macOS 12 scanner bug
macOS has a bug (feature?) where advertising data is no longer received unless one or more service UUIDs are given in `scanForPeripheralsWithServices:options:`.

This implements a new kwarg on `BleakScanner` to allow users to provide such a list of UUIDs. This commit only implements it in the CoreBluetooth backend with other backends to follow.

If macOS 12 is detected and this kwarg is not given, an error will be logged explaining how to fix the problem.

Fixes #635.