hbldh/bleak #639: Have to reconnect after connecting

sonus89, 2021-09-18:
* bleak version: 0.12.1
* Python version: 3.9
* Operating System: Linux ( 5.13.13-1-MANJARO )
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.61

### Description

How can I write UUID without reconnecting? 

### What I Did

1. Connecting to device ( nRF5280 Bluetooth USB dongle's connection indicator LED is on )
2. Writing UUID successful
3. Process finished.
4. Indicator LED still on ( connected )
5. Trying to write UUID again without reconnecting ( unsuccessful )

dlech, 2021-09-19:
Duplicate of #367?

sonus89, 2021-09-19:
Nope. This is not a duplicate of #367 !
I don't want to reconnect to the device 
I just want to be able to write the UUID without reconnecting.

dlech, 2021-09-19:
I'm pretty sure that is discussed as a solution in that issue as well.

closed 2022-07-26 by dlech