hbldh/bleak #641: assertion failed, "self._bus is None", bluez backend on Linux

egnor, 2021-09-21:
* bleak version: (github head)
* Python version: 3.7.3
* Operating System: Raspberry Pi OS "buster"
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.61

### Description

Running a stress test that consists of repeated discovery/connect/transmit cycles, after a couple hours I see this message, which seems to happen when closing a connection (BleakClient's __aexit__):

```
Traceback (most recent call last):
  ... various uninteresting bits of my code elided ...
  File "/home/pi/source/nametag/nametag/bluetooth.py", line 174, in __aexit__
    await self._exits.aclose()
  File "/usr/lib/python3.7/contextlib.py", line 593, in aclose
    await self.__aexit__(None, None, None)
  File "/usr/lib/python3.7/contextlib.py", line 652, in __aexit__
    raise exc_details[1]
  File "/usr/lib/python3.7/contextlib.py", line 635, in __aexit__
    cb_suppress = await cb(*exc_details)
  File "/usr/lib/python3.7/contextlib.py", line 539, in _exit_wrapper
    return await cm_exit(cm, exc_type, exc, tb)
  File "/home/pi/.local/share/virtualenvs/nametag-Zg6otAYu/src/bleak/bleak/backends/client.py", line 65, in __aexit__
    await self.disconnect()
  File "/home/pi/.local/share/virtualenvs/nametag-Zg6otAYu/src/bleak/bleak/backends/bluezdbus/client.py", line 480, in disconnect
    assert self._bus is None
AssertionError
```

### What I Did

I'm talking to a BTLE device (a cheap digital nametag). Unfortunately this is not reproducible on demand, though it usually crops up (and given that it's an assertion error is fairly damaging).

### Discussion

Here is [the code in question](https://github.com/hbldh/bleak/blob/develop/bleak/backends/bluezdbus/client.py#L480) with some commentary added:

```python
    async def disconnect(self) -> bool:
        """Disconnect from the specified GATT server.
        Returns:
            Boolean representing if device is disconnected.
        Raises:
            BleakDBusError: If there was a D-Bus error
            asyncio.TimeoutError if the device was not disconnected within 10 seconds
        """
        logger.debug(f"Disconnecting ({self._device_path})")

        if self._bus is None:
            # No connection exists. Either one hasn't been created or
            # we have already called disconnect and closed the D-Bus
            # connection.
            logger.debug(f"already disconnected ({self._device_path})")
            return True

        if self._disconnecting_event:
            # another call to disconnect() is already in progress
            logger.debug(f"already in progress ({self._device_path})")
            await asyncio.wait_for(self._disconnecting_event.wait(), timeout=10)
        elif self.is_connected:  # <== IF THIS IS FALSE, THE ASSERTION WILL FAIL
            self._disconnecting_event = asyncio.Event()
            try:
                # Try to disconnect the actual device/peripheral
                reply = await self._bus.call(
                    Message(
                        destination=defs.BLUEZ_SERVICE,
                        path=self._device_path,
                        interface=defs.DEVICE_INTERFACE,
                        member="Disconnect",
                    )
                )
                assert_reply(reply)
                await asyncio.wait_for(self._disconnecting_event.wait(), timeout=10)
            finally:
                self._disconnecting_event = None

        # sanity check to make sure _cleanup_all() was triggered by the
        # "PropertiesChanged" signal handler and that it completed successfully
        assert self._bus is None   # <== THE ASSERTION IN QUESTION

        return True
```

Since this function returns early if `self._bus` is already None, if the "`if self._disconnecting_event:`" and "`elif self.is_connected`" conditions both fall through, the assertion is guaranteed to fail!

Specifically, this could happen with a server side disconnection, with a race between
- `self._properties["Connected"]` being set False
- the call to `disconnect()`
- the `_cleanup_all()` task spawned by the property change handler

This would all be a lot more bulletproof if `_cleanup_all()` were synchronous so there was no intermediate state with `is_connected` false but `_bus` still non-None. This appears to be there to support `_remove_signal_handlers` which requires an async call to `remove_match`.

Assuming that all has to be that way, I'd propose
- establish the invariant that `_disconnecting_event` is triggered (and removed) exactly when `self._bus` is set to None
- rename it, perhaps, to `_bus_removal_event` to drive that point home
- change the logic in `disconnect()` to check for the event (and pile on it), then check for `self._bus` instead of checking `is_connected`?

dlech, 2021-10-07:
Great analysis of the issue, thanks for that.

>     * bleak version: (github head)

I'm guess that #621 didn't fix the issue then?

Looking at the bigger picture, I've been thinking that we should probably put locks on all methods that interact with hardware. This would prevent re-entrancy issues for users with parallel tasks as well (e.g. trying to connect to a device at the same time in two separate tasks). I think these sorts of issues are a problem on all backends, not just BlueZ.

alexanderadam, 2024-09-29:
This issue still happens when syncing with [omblepy](https://github.com/userx14/omblepy) (i.e. [here](https://github.com/userx14/omblepy/issues/31#issue-2435261728), [here](https://github.com/userx14/omblepy/issues/27), [here](https://github.com/userx14/omblepy/issues/22), [here](https://github.com/userx14/omblepy/issues/20), [here](https://github.com/userx14/omblepy/issues/16#issuecomment-1656747061), [here](https://github.com/userx14/omblepy/issues/12), [here](https://github.com/userx14/omblepy/pull/10), [here](https://github.com/userx14/omblepy/issues/7)) and people kinda got used to it.

But I guess it would be great if this would be fixed and some point.

I'm not a Python developer but I can also reproduce this easily in case I can help.

dlech, 2024-09-29:
This issue is closed, so I would suggest starting a new issue with a reproducible test case.

alexanderadam, 2024-09-29:
Thank you for the fast response.

Can you elaborate on the test case?

As mentioned before, I'm not a Python developer, so I can't contribute any code which would make any sense.
But I'm a [omblepy](https://github.com/userx14/omblepy) user, a tool that uses bleak. And [the `disconnect` method](https://github.com/userx14/omblepy/blob/main/omblepy.py#L387) reliably reproduces that state for all of its users.

So I can reproduce the issue by using a device that I own with the tool that uses bleak.

dlech, 2024-09-29:
If you can reproduce the error with one of the example Bleak scripts or if you can create a minimum version of the code from the tool to reproduce the issue, that would be useful. Also, any logs of error stack traces, debug messages and Bluetooth packet could be useful (see troubleshooting page in the docs).

commit f16f4024 (2021-10-08): bluez: syncronize disconnect and cleanup

When the following sequence of events occured, `assert self._bus is None`
would be hit.

- A "Connected": False property change signal is sent over D-Bus.
- Bleak handles the event and calls the `_cleanup_all()` method.
- The `_cleanup_all()` sends a D-Bus message to unsubscribe from signals and awaits it.
- While the previous method is still awaiting, the disconnect method is called.
- This method hits an assertation because the connected property is false
  but there is still a D-Bus bus connection since `_cleanup_all()` is still using it.

This fixe [... cut]