hbldh/bleak #642: Low data rate

billskar23, 2021-09-23:
I have successfully connected my BLE device (ESP32 board + IMU) to my windows pc using 
Bleak library and I can receive data from the IMU.. However, I receive the data with a sampling frequency <25 Hz. Using the same code on my ESP32 board I can receive data on my phone with a rate of 180 Hz.

Why does this happen? Do I have to modify the library?

Thank you

dlech, 2021-09-23:
Python is an interpreted language, so it can be a bit slow to start with, plus the bridge between Python and the WinRT API is probably not the most efficient. I would suggest starting with some profiling to see where the bottleneck is.

karlp, 2022-05-09:
FWIW, I'm receiving 5khz @ 16bit data via notifications with bleak, so it's not python per se.  However, it's absolutely critical that you batch these up into "bigger" notifications.  trying to send a notification for every single 16bit value will absolutely fail for instance.  You might have a shorter connection interval on the phone?

dlech, 2022-10-03:
Closing due to insufficient feedback (logs).

closed 2022-10-03 by dlech

karlp, 2022-10-04:
To clarify for anyone finding this, by "batch them up into bigger notifications" I mean, you absolutely can _not_ send 2 byte (16bit) notifications at 5kHz.  but sending 240byte notifications with multiple samples per packet lets me send my entire 16bit@5kHz sampled data without loss.