hbldh/bleak #646: Characteristic was not found

jakubkolton, 2021-10-03:
* bleak version: 2.5.2
* Python version: 3.8.2
* Operating System: Win 10 
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I wanted to read data from custom 128-bit UUID characteristic of my BLE Device (based on Nordic nRF development kit). So I used "Connect to a Bluetooth device and read its model number:" example - I changed the MAC and UUID for mine of course. I constantly had an "bleak.exc.BleakError: Characteristic c4d41401-a2b9-11eb-bcbc-0242ac130002 was not found!" despite it was visible for example in "NRF Connect" app for Android. Then I decided to check is it actually visible using "service_explorer.py" example - it seemed that my characteristic is not present. I'm sure it's able to read, because I'm able to do it via read and notifications through "NRF Connect" Android App. What might be a problem? "Run as admin" makes no difference. All that code works for example with "Device Name" characteristic.

Full traceback:
PS C:\Projekty_STM32\Praca_Inzynierska\Inzynierka\Python_BLE_UART_Interface> py ble_uart_iface.py
Traceback (most recent call last):
  File "ble_uart_iface.py", line 38, in <module>
    loop.run_until_complete(run(address))
  File "C:\Users\QB\AppData\Local\Programs\Python\Python38-32\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "ble_uart_iface.py", line 34, in run
    model_number = await client.read_gatt_char(MODEL_NBR_UUID)
  File "C:\Users\QB\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\backends\dotnet\client.py", line 593, in read_gatt_char
    raise BleakError("Characteristic {0} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic C4D41401-A2B9-11EB-BCBC-0242AC130002 was not found!

dlech, 2021-10-04:
>  Then I decided to check is it actually visible using "service_explorer.py" example

What is the output of this?

jakubkolton, 2021-10-04:
Connected: True
[Service] 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
        [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2): Device Name (read,write), Value: b'ECG_Device'
        [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4): Appearance (read), Value: b'\x00\x00'
        [Characteristic] 00002a04-0000-1000-8000-00805f9b34fb (Handle: 6): Peripheral Preferred Connection Parameters (read), Value: b'P\x00\xa0\x00\x00\x00\x90\x01'
        [Characteristic] 00002aa6-0000-1000-8000-00805f9b34fb (Handle: 8): Central Address Resolution (read), Value: b'\x01'
[Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 10): Generic Attribute Profile
[Service] 6e400001-b5a3-f393-e0a9-e50e24dcca9e (Handle: 11): Nordic UART Service
        [Characteristic] 6e400002-b5a3-f393-e0a9-e50e24dcca9e (Handle: 12): Nordic UART RX (write-without-response,write), Value: None
        [Characteristic] 6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 14): Nordic UART TX (notify), Value: None
                [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 16): Client Characteristic Configuration) | Value: b'\xb8\x00'

It's interesting that it doesn't discover all the client's characteristics - there should be custom C4D41401-A2B9-11EB-BCBC-0242AC130002 and 0x2A01, 0x2AA6 from Generic Access Service, 0x2A05 from Generic Attribute Service. And what is even more interesting, it discovers a Nordic UART Service that isn't discovered by NRF Connect mobile app.

dlech, 2021-10-04:
Is it possible that Windows is using cached values? https://bleak.readthedocs.io/en/latest/troubleshooting.html#id1

In most places this defaults to `False` but it looks like there is one place where it is `True`. We should probably change that to `False`.

jakubkolton, 2021-10-04:
I believe it might use cached values. I've changed one line according to your tip:
# async with BleakClient(address) as client:
    async with BleakClient(address, use_cached=False) as client:

Now it seems to have yet another .NET issue:
Traceback (most recent call last):
  File "service_explorer.py", line 88, in <module>
    loop.run_until_complete(run(ADDRESS, True))
  File "C:\Users\QB\AppData\Local\Programs\Python\Python38-32\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "service_explorer.py", line 43, in run
    async with BleakClient(address, use_cached=False) as client:
  File "C:\Users\QB\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\backends\client.py", line 61, in __aenter__
    await self.connect()
  File "C:\Users\QB\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\backends\dotnet\client.py", line 250, in connect
    await self.get_services(use_cached=use_cached)
  File "C:\Users\QB\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\backends\dotnet\client.py", line 443, in get_services
    services_result = await wrap_IAsyncOperation(
  File "C:\Users\QB\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\backends\dotnet\utils.py", line 79, in wrap_IAsyncOperation
    raise BleakDotNetTaskError(op.ErrorCode.ToString())
bleak.exc.BleakDotNetTaskError: System.Exception: Katastrofalny błąd. (Wyjątek od HRESULT: 0x8000FFFF (E_UNEXPECTED))

dlech, 2021-10-04:
This is a Windows error, so I would suggest:
- Reboot your computer.
- If the device is listed under Bluetooth devices in Windows settings, remove it.
- Search the web for the error (translation from Polish: Catastrophic Error).
- Log Bluetooth packets while reproducing the error.

jakubkolton, 2021-10-04:
First step was what had helped me. Thank you a lot!

dlech, 2021-10-04:
I would like to leave this open so that we change the default to not use cached.

fjmolinas, 2021-12-01:
Hi, I seem to have the same issue on Linux Ubuntu21.04. I update the characteristics of a service but when I scan I always get the same, I've also rebooted my computer in between with the same result.

dlech, 2021-12-01:
Please open a new issue with sample code to reproduce the issue and logs. This issue was for Windows, but it seem like you have a problem on Linux.

fjmolinas, 2021-12-01:
> Hi, I seem to have the same issue on Linux Ubuntu21.04. I update the characteristics of a service but when I scan I always get the same, I've also rebooted my computer in between with the same result.

Sorry for the noise, this seems to be an issue with bluez and unrelated to bleak.

commit 1f8f1705 (2021-10-06): winrt: default to not use cache for services

All other platforms already default to not use cache. Also, using cache
is problematic for DIY devices that don't announce that services have
changed.

Fixes #646.