hbldh/bleak #647: main.py: Unexpected Error: 'NoneType' object has no attribute 'add_connection_status_changed'

abeldsouza, 2021-10-04:
Hello, I recently came across this ble library (https://github.com/Jakeler/ble-serial) and started exploring it to see if its suitable for me to use in my project. While connecting to my ble peripheral device i get the following error. I dont understand this error and hence seeking help

main.py: Unexpected Error: 'NoneType' object has no attribute 'add_connection_status_changed'

System Version information below:

Windows 10 Pro x64
Laird’s BL 652 peripheral device, Bluetooth v5.0 - Single mode with nRF52832 chipset

C:\Windows\system32>pip freeze
ble-serial==2.4.1
bleak==0.12.1
coloredlogs==15.0.1
humanfriendly==10.0
pypiwin32==223
pyreadline3==3.3
pyserial==3.5
pywin32==301
winrt==1.0.21033.1

C:\Windows\system32>python -V
Python 3.9.7

### Description
I tried to use the ble-serial to connect to my peripheral. The connection failed with an exception.

### What I Did

It is preferable if an issue contains a [Miminal Workable Example](https://stackoverflow.com/help/minimal-reproducible-example).
This will otherwise be one of the first questions you will get as a response.

It is also preferable if that example is not dependent on a specific peripheral device, but can be run and
reproduced with other BLE peripherals as well.

```
C:\Windows\system32>python -m ble_serial.scan
Started BLE scan

D5:D5:25:ED:2C:7C (RSSI=-53): AWAK_?008-D5D525ED2C7C
24:E7:8A:2E:95:E5 (RSSI=-60):
24:F5:BE:B5:95:CC (RSSI=-67):
1B:93:ED:13:A2:D3 (RSSI=-67):
30:73:0F:6C:A4:CF (RSSI=-70):
25:56:C7:AD:96:FF (RSSI=-70):
39:19:E6:79:69:C0 (RSSI=-72):
16:7D:C4:AA:4D:C7 (RSSI=-72):
7C:55:65:F8:34:BB (RSSI=-73):
2F:1A:FE:60:AF:AB (RSSI=-73):
31:32:63:36:65:EF (RSSI=-73):
09:24:F8:A1:3F:ED (RSSI=-75):
BF:D1:5A:0B:80:A5 (RSSI=-75):
2A:D8:A3:91:83:45 (RSSI=-76):
37:EE:45:66:D5:57 (RSSI=-77):
49:43:E8:DA:30:DB (RSSI=-77):
6E:37:C9:44:F4:3E (RSSI=-77):
50:67:5F:ED:FA:CC (RSSI=-78):
70:54:43:ED:ED:16 (RSSI=-78):
D9:04:B8:61:42:24 (RSSI=-79): AWAK_e015-D904B8614224
52:36:7D:88:8D:B2 (RSSI=-80):
2F:95:2D:CC:29:51 (RSSI=-81):
D1:46:71:2F:5B:92 (RSSI=-83):
64:E7:D8:46:BA:B3 (RSSI=-84):
41:8D:32:35:79:B0 (RSSI=-84):
72:D1:F3:3E:D8:4C (RSSI=-85):
7A:2A:C2:3E:61:12 (RSSI=-85):
72:E9:E3:CB:17:AE (RSSI=-85):
53:1F:38:82:92:6D (RSSI=-86):
7B:2B:71:39:A2:55 (RSSI=-86):
77:7C:25:1B:66:5D (RSSI=-87):
F4:07:80:A3:C4:3B (RSSI=-87):
6A:39:BB:CB:EE:58 (RSSI=-87): [LG] webOS TV UJ632T
32:13:1C:E5:E5:CB (RSSI=-90):
4E:EE:FE:86:EB:34 (RSSI=-90):
45:5B:37:64:28:39 (RSSI=-91):
4C:2F:30:91:E8:C4 (RSSI=-91):
28:02:D2:41:54:E7 (RSSI=-92):
28:F0:96:75:F8:8A (RSSI=-92):
46:27:72:4D:01:03 (RSSI=-92):
44:73:D6:86:B1:05 (RSSI=-92): MeetUp Soft Remote
78:EC:31:08:A8:DA (RSSI=-92):
DB:C2:56:97:AC:55 (RSSI=-92):
22:D4:3D:46:B5:6B (RSSI=-92):
75:97:FB:CC:25:BA (RSSI=-93):
04:BA:13:28:A3:89 (RSSI=-94):
5C:C1:D7:6F:12:93 (RSSI=-94):
4A:35:16:5E:4C:8F (RSSI=-95):
03:E3:3C:3B:1E:39 (RSSI=-96):
5F:FF:05:3D:47:BA (RSSI=-96):
44:73:D6:86:BF:A9 (RSSI=-97):
5F:B6:DF:D9:06:16 (RSSI=-98):
4E:BC:B5:E6:CD:9C (RSSI=-100):
9D:1A:F4:65:BB:86 (RSSI=-100):
40:C2:9E:BC:DD:60 (RSSI=-100):

Finished BLE scan

C:\Windows\system32>python -m ble_serial.scan -d D5:D5:25:ED:2C:7C
Started deep scan of D5:D5:25:ED:2C:7C

SERVICE 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
     CHARACTERISTIC 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2):  ['read']
     CHARACTERISTIC 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4):  ['read']
     CHARACTERISTIC 00002a04-0000-1000-8000-00805f9b34fb (Handle: 6):  ['read']
     CHARACTERISTIC 00002aa6-0000-1000-8000-00805f9b34fb (Handle: 8):  ['read']
SERVICE 00001801-0000-1000-8000-00805f9b34fb (Handle: 10): Generic Attribute Profile
     CHARACTERISTIC 00002a05-0000-1000-8000-00805f9b34fb (Handle: 11):  ['indicate']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 13): Client Characteristic Configuration
SERVICE 569a1101-b87f-490c-92cb-11ba5ea5167c (Handle: 14): Unknown
     CHARACTERISTIC 569a2000-b87f-490c-92cb-11ba5ea5167c (Handle: 15):  ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 17): Client Characteristic Configuration
     CHARACTERISTIC 569a2001-b87f-490c-92cb-11ba5ea5167c (Handle: 18):  ['write-without-response', 'write']
     CHARACTERISTIC 569a2002-b87f-490c-92cb-11ba5ea5167c (Handle: 20):  ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 22): Client Characteristic Configuration
     CHARACTERISTIC 569a2003-b87f-490c-92cb-11ba5ea5167c (Handle: 23):  ['write-without-response', 'write']
SERVICE 0000180a-0000-1000-8000-00805f9b34fb (Handle: 25): Device Information
     CHARACTERISTIC 00002a29-0000-1000-8000-00805f9b34fb (Handle: 26):  ['read']
     CHARACTERISTIC 00002a24-0000-1000-8000-00805f9b34fb (Handle: 28):  ['read']
     CHARACTERISTIC 00002a27-0000-1000-8000-00805f9b34fb (Handle: 30):  ['read']
     CHARACTERISTIC 00002a26-0000-1000-8000-00805f9b34fb (Handle: 32):  ['read']
     CHARACTERISTIC 00002a28-0000-1000-8000-00805f9b34fb (Handle: 34):  ['read']

Completed deep scan of D5:D5:25:ED:2C:7C

C:\Windows\system32>python -m ble_serial -d D5:D5:25:ED:2C:7C
12:18:50.664 | INFO | ble_interface.py: Receiver set up
12:18:50.664 | INFO | ble_interface.py: Trying to connect with D5:D5:25:ED:2C:7C
12:18:52.076 | ERROR | main.py: Unexpected Error: 'NoneType' object has no attribute 'add_connection_status_changed'
12:18:52.076 | WARNING | main.py: Shutdown initiated
12:18:52.076 | INFO | main.py: Shutdown complete.
```

As seen above, i can do a scan and also a deep scan. However attempting to connect to the device fails. 
I can connect and receive data from the peripheral device with many popular ble apps like light blue (on android and iOS). On windows i have tried 2 ble apps. Both can connect and receive / write data successfully.

https://github.com/sensboston/BLEConsole
Noble (https://github.com/noble/noble)
My custom project currently works with noble but I am looking to get away from noble as there are some teething issues that cannot be resolved. hence i choose ble-serial.

Thanks,
Abel.

Note: This query was posted in ble-serial and based on the response received was advised to check on this forum
https://github.com/Jakeler/ble-serial/issues/45

dlech, 2021-10-04:
> 12:18:52.076 | ERROR | main.py: Unexpected Error: 'NoneType' object has no attribute 'add_connection_status_changed'

Can you get a full stack trace for this error?

abeldsouza, 2021-10-05:
@dlech Thank you for your reply. Since i am bit new to this, would you help me with some information on how the stack trace is to be collected. I assume that updating the code is not the correct way to go about it. So i was trying to run the ble connect operation by hooking up to python debugger pdb. But i did not get what i was looking for, Below is what i tried

[image]

Please let me know.

Thanks,
Abel.

dlech, 2021-10-05:
How about enabling Bleak logging instead? https://bleak.readthedocs.io/en/latest/troubleshooting.html

abeldsouza, 2021-10-05:
@dlech please see attached. is the log what you expected? Else please let me know what i did wrong :) The installations for capturing bluetooth traffic is in progress. will try that tomorrow.

Thanks,
Abel
[LogWithBleakLogging.txt](https://github.com/hbldh/bleak/files/7287668/LogWithBleakLogging.txt)
.

dlech, 2021-10-05:
For some reason during the command `python -m ble_serial -d D5:D5:25:ED:2C:7C`, the Bleak logging is not working. Maybe the `ble_serial` library is disabling Bleak logging in this case?

abeldsouza, 2021-10-05:
@dlech yeah that did look odd. I was also wondering why there was no debug traces.

Jakeler, 2021-10-05:
> Maybe the `ble_serial` library is disabling Bleak logging in this case?

Yes, ble-serial sets it indeed always to `INFO` log level, in [this line](https://github.com/Jakeler/ble-serial/blob/v2.4.1/ble_serial/log/console_log.py#L4). Please follow the from source installation and just remove this line, then try again and it should print much more.

abeldsouza, 2021-10-05:
@Jakeler Thanks for the help.
@dlech Log attached. Pls have a look.

Thanks,
Abe
[LogWithBleakLogging_Updated.txt](https://github.com/hbldh/bleak/files/7288117/LogWithBleakLogging_Updated.txt)
l.

dlech, 2021-10-05:
It looks like we aren't handling the case that `BluetoothLEDevice.FromBluetoothAddressAsync()` can return null/None. We can check this and raise a `BleakError` when this happens to get a better error message.

Here is a comment from a MSFT engineer explaining what this could cause this: https://github.com/microsoft/Windows-universal-samples/issues/1089#issuecomment-487586755

Does your device require pairing or use a random address?

closed 2021-10-06 by dlech

abeldsouza, 2021-10-08:
@dlech Thank you again. I did as you say. I can see the new error message. Ill hop on back to the ble-serial page and get help from @Jakeler

abeldsouza, 2021-10-08:
BLE devices should generally connect. It should not need pairing as a prerequisite for connection based on my work in ble so far. Pairing is only done post connection.

On my peripheral device, the peripheral closes the connection if a central device that is unbonded does not initiate pairing within a few seconds post connection. This is the logic I have to kick out unauthenticated central devices.

However in this case my laptop and the peripheral devices are bonded. The pairing is compete. I can communicate with my device using a few other ble lib.

[image]

commit db3ee6d0 (2021-10-06): winrt: improve error message when connecting fails

BluetoothLEDevice.FromBluetoothAddressAsync() can return null, so we
need to handle this case. The link in the comment provides some insight
in to what might cause this failure.

Fixes #647.