hbldh/bleak #652: proposed breaking change: Bleak init fuction

dlech, 2021-10-06:
I've collected a number of cases for needing a Bleak init function. But first I should say that this would be a breaking change because it would require users to modify their code like this:

```python
import asyncio

from bleak import init as init_bleak

async def main():
    await init_bleak()
    ...

asyncio.run(main())
```

So why do we need such a function?

- Allows detecting the presence of Bluetooth hardware before using other Bleak API:
  - Is a possible solution for #320
  - Would provide a place to start a global D-Bus instance that could help address #500
  - Is a better way to handle: https://github.com/hbldh/bleak/blob/41aa0e261277201aad057fe2a77ddf26df2fe0f9/bleak/backends/corebluetooth/CentralManagerDelegate.py#L77-L82
- Allows a place to customize Bleak startup:
  - Could select default adapter for BlueZ backend here
  - Could address STA vs. MTA issues in WinRT backend here (needed for UI integration).
- Could be used to avoid import side effects (well-behaving Python module should not have side effects on import, like spawning processes, which we are currently doing):
  - This is also relevant for STA vs. MTA issues.
  - Would avoid the need hacks like: https://github.com/hbldh/bleak/blob/41aa0e261277201aad057fe2a77ddf26df2fe0f9/bleak/__init__.py#L19-L20

dlech, 2021-10-09:
I suppose we could make this less breaking by issuing a warning and calling the init function the first time a scanner is started if it has not already been called.

dlech, 2021-10-16:
Thanks for the suggestion. However, some of the features that need to be added to this are async and we can't call async functions from `__new__()` or `__init__()`.

dlech, 2021-10-16:
This is why I was thinking we could make a new global async init function. Users can call it explicitly if they need custom options or if they want to control exactly when it runs, but we could implicit call it if hasn't be run yet the first time we start scanning (from the async `BleakScanner.start()` method.

elupus, 2021-10-16:
I think a breaking change with async factory functions make more sense then. You would still need to do detections to figure out which backend yo use right?

dlech, 2021-10-16:
> You would still need to do detections to figure out which backend yo use right?

Yes, that would not change.

dlech, 2022-10-03:
Closing since most of the problems this would solve have been fixed other ways. There is still potentially a need for a `BleakAdapter` class as described in https://github.com/hbldh/bleak/issues/992#issuecomment-1242760927

closed 2022-10-03 by dlech