hbldh/bleak #666: [org.bluez.Error.Failed] Software caused connection abort

Alexandre2006, 2021-10-11:
* bleak version: 0.12.1
* Python version: 3.9.5
* Operating System: Pop OS 21.04 (Ubuntu 21.04)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.56

### Description

Code errors out while trying to connect to device (in this case, a Bluetooth light bulb) for no apparent reason. Most common error is "RuntimeError: Event loop is closed" but the most informative error is "bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Software caused connection abort". They each have a about 50/50 chance of appearing when running the code. If it isn't the runtime error, it's a bleak error, and vice versa.

### What I Did

Here is my code: https://pastebin.com/r4M4LecZ

I am initializing a new "Backend" (which is what I'm calling the lower level code of the Bluetooth light) with the mac address to a light, which I am able to connect to using bluetoothctl.

```
Traceback (most recent call last):
  File "/home/thinkalex/Desktop/PycharmProjects/DanceFloor/main.py", line 12, in <module>
    test = Backend("A4:C1:38:9E:AD:9E")
  File "/home/thinkalex/Desktop/PycharmProjects/DanceFloor/LED/Backend.py", line 22, in __init__
    asyncio.run(self._connect())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/thinkalex/Desktop/PycharmProjects/DanceFloor/LED/Backend.py", line 34, in _connect
    await self.backend.connect(use_cached=False)
  File "/home/thinkalex/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/client.py", line 298, in connect
    assert_reply(reply)
  File "/home/thinkalex/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Software caused connection abort

```

OR

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/thinkalex/Desktop/PycharmProjects/DanceFloor/LED/Backend.py", line 54, in _ping_loop
    asyncio.run(self._ping())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/thinkalex/Desktop/PycharmProjects/DanceFloor/LED/Backend.py", line 57, in _ping
    await self._send_custom(CHARACTERISTIC, b'\xAA\x01\x00\x00\x00\x00\x00\x00\x00\x00'
  File "/home/thinkalex/Desktop/PycharmProjects/DanceFloor/LED/Backend.py", line 61, in _send_custom
    await self.backend.write_gatt_char(characteristic, data)
  File "/home/thinkalex/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/client.py", line 843, in write_gatt_char
    reply = await self._bus.call(
  File "/home/thinkalex/.local/lib/python3.9/site-packages/dbus_next/aio/message_bus.py", line 303, in call
    self._call(msg, reply_handler)
  File "/home/thinkalex/.local/lib/python3.9/site-packages/dbus_next/message_bus.py", line 588, in _call
    self.send(msg)
  File "/home/thinkalex/.local/lib/python3.9/site-packages/dbus_next/aio/message_bus.py", line 326, in send
    self._writer.schedule_write(msg, future)
  File "/home/thinkalex/.local/lib/python3.9/site-packages/dbus_next/aio/message_bus.py", line 85, in schedule_write
    self.loop.add_writer(self.fd, self.write_callback)
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 346, in add_writer
    self._add_writer(fd, callback, *args)
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 295, in _add_writer
    self._check_closed()
  File "/usr/lib/python3.9/asyncio/base_events.py", line 510, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
Exception ignored in: <function Backend.__del__ at 0x7f8a77538a60>
Traceback (most recent call last):
  File "/home/thinkalex/Desktop/PycharmProjects/DanceFloor/LED/Backend.py", line 26, in __del__
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
  File "/home/thinkalex/Desktop/PycharmProjects/DanceFloor/LED/Backend.py", line 45, in _disconnect
  File "/home/thinkalex/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/client.py", line 459, in disconnect
  File "/home/thinkalex/.local/lib/python3.9/site-packages/dbus_next/aio/message_bus.py", line 303, in call
  File "/home/thinkalex/.local/lib/python3.9/site-packages/dbus_next/message_bus.py", line 588, in _call
  File "/home/thinkalex/.local/lib/python3.9/site-packages/dbus_next/aio/message_bus.py", line 326, in send
  File "/home/thinkalex/.local/lib/python3.9/site-packages/dbus_next/aio/message_bus.py", line 85, in schedule_write
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 346, in add_writer
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 295, in _add_writer
  File "/usr/lib/python3.9/asyncio/base_events.py", line 510, in _check_closed
RuntimeError: Event loop is closed
```

dlech, 2021-10-12:
Bleak is not thread-safe. Try rewriting you code without threads.

Alexandre2006, 2021-10-12:
Sorry if the answer to many of these questions are obvious, I don't have any experience with BLE and Bluetooth and I'm quite rusty in python

dlech, 2021-10-12:
see #631 for some ideas

if a connect fails with that error, maybe just retry connecting?

Alexandre2006, 2021-10-16:
Thank you for your assistance! It worked!

closed 2021-10-16 by Alexandre2006

dlee-bd, 2022-04-14:
@Alexandre2006  Hello, I have been having the same issue with the `[org.bluez.Error.Failed] Software caused connection abort
` error occurring every once in awhile. (For me, it felt like 30~40%). As I can see "it worked!" for you, may I please ask what the solution was? Was it simply just retrying until it works? or was there a code fix that you were able to make?

Thank you.

igo, 2023-11-05:
After an hour of googling with no success to solve this issue I restarted raspberry pi that had this issue and it started working.

ianchanning, 2024-03-14:
@dlee-bd https://xkcd.com/979/

ianchanning, 2024-12-03:
I eventually got this working, first a level of dbus filtering seemed to help when scanning as with #631, but not significantly. However having up to 5 retries was enough.

```python
    retries = 5
    for attempt in range(1, retries + 1):
        logger.info(
            f"Attempt {attempt} of {retries} to connect to {get_device_id(device)}..."
        )
        try:
            # Initialize the Bluetooth client
            async with BleakClient(
                device,
                disconnected_callback=disconnected_callback,
                timeout=CONNECT_TIMEOUT_S,
            ) as client:
                logger.info(f"Connected: {client.is_connected}")
                # code for once connected...

            logger.info(f"Connected: {client.is_connected}")
            return True

        except BleakError as e:
            logger.error(f"Configure connect fail because: '{e!r}'")

        except Exception as e:
            logger.error(f"Configuration connect failed for some other reason: {e!r}")

        # retry delay
        await asyncio.sleep(1.0)