hbldh/bleak #668: Same script working only on some computers

Lukwrt, 2021-10-15:
* bleak version: 0.12.1
* Python version: 3.8.5 (or 3.9)
* Operating System: Windows 10

### Description

I'm using bleak to connect a medical device (Heart Rate Monitoring mainly) in BLE to computers and receive the measurements.

I wrote a small script to do this and this worked on my computer. 
My model is a ASUS Vivobook 14 X405UA.

After this, I tried the script on another computer, this model is a Dell Inspiron 15 3505.
On this computer the script worked well for 5 days after which it started to missbehave and then almost not work at all.
I suspect a windows update changed something to this computer maybe related to the Bluetooth driver.
I rebooted the computer and reinstalled visual studio, python, bleak from scratch and I'm still getting errors.

My issue is to understand why the script is not working anymore on the Dell.

### What I Did

This is not exactly the script I'm using as I'm in fact reading from two characteristics and writing to another one but the structure in terms of Bleak and asyncio usage is the same, if needed I can provide a more complete code.

```
import asyncio
from bleak import BleakClient, BleakScanner
from bleak.exc import BleakError

# Variables
hr_uuid = "0aad7ea0-0d60-11e2-8e3c-0002a5d5c51b"
ADDRESS = "AA:AA:AA:AA:AA:AA"

def handler(sender, data):
    """Simple notification handler which prints the data received."""
    print(data)

async def run(address):
    device = await BleakScanner.find_device_by_address(ADDRESS, timeout=30.0)
    if not device:
        raise BleakError(f"A device with address {ADDRESS} could not be found.")
    async with BleakClient(device, timeout=30.0) as client:
        await client.start_notify(hr_uuid, handler)
        await asyncio.sleep(10.0)
        await client.stop_notify(hr_uuid)
        await client.disconnect()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(ADDRESS))
```

So on my computer this still works fine, I can connect to the device, receive notifications, read them and disconnect properly.
On the dell computer I got mainly one error (I tried the script in pyzo and visual studio code) , here's the traceback in Pyzo : 

```
Traceback (most recent call last):
  File "C:\Users\Script.py", line 178, in <module>
    loop.run_until_complete(run(ADDRESS))
  File "C:\Program Files\pyzo\source\pyzo\pyzokernel\guiintegration.py", line 209, in stub_run_until_complete
    return self.app._original_run_until_complete(*args)
  File "c:\users\user\appdata\local\programs\python\python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "C:\Users\Script.py", line 134, in run
    await client.start_notify(hr_uuid, handler)
  File "c:\users\user\appdata\local\programs\python\python39\lib\site-packages\bleak\backends\winrt\client.py", line 773, in start_notify
    status = await characteristic_obj.write_client_characteristic_configuration_descriptor_async(
RuntimeError
```

The device connects to the computer but has an issue with the notification function.
So the error is in start notify, and particularly in the line 773 of winrt\client.py (I slightly modified the paths of the traceback but the important is the one mentioned before). I suppose this is linked to the windows wrt backend but I am not sure how to fix this.

Lukwrt, 2021-10-18:
Hello I'm unsure how to try the develop branch. I cloned it on my computer but when I want to import bleak, as the folder is not a package, I get "module Bleak not found", how should I import the functions coming from the cloned repo ?

lrq3000, 2021-11-25:
> I'm not sure it will fix the issues here, but could you please try #680 to see if it makes a difference?

I have a similar issue (`BleakError: Could not get GATT services: Unreachable` when trying to instantiate a `BleakClient`). I just tried to pip install the develop branch with the recently merged PR #680, but this didn't change the issue.

One thing to note is that the device I am trying to connect to is broadcasting advertisement messages every 100ms for 400µs, so it's not infrequent that it doesn't even display in the list of devices. Maybe this contributes to this issue?

lrq3000, 2021-11-25:
Fixed the issue, it was because Windows was paired with the device, it works now that I unpaired them. I also used [this free Microsoft tool](https://github.com/Microsoft/BluetoothLEExplorer) to debug, maybe this can be helpful to OP. If this tool cannot connect either, then the issue is with the device or the pairing between the computer and the device, but not Bleak, since the tool I linked to is developed by Microsoft itself, so that's how I troubleshot my own issue.

closed 2025-05-16 by dlech