hbldh/bleak #682: spotty connection on Windows for long advertisement intervals?

kazola, 2021-10-31:
* bleak version: 0.12.x - 0.14
* Python version: 3.7 - 3.9
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.61

### Description

Problems connecting to a peripheral advertising every 1.3 seconds.

### What I Did

On Linux with bluepy, no problem.
On Linux with bleak, no problem.
On Windows with C#, no problem.
On Windows with bleak, more than half of the times connection fails.

Can someone indicate me if this also happens for them?
Thanks!

dlech, 2021-11-01:
Can you share the Windows Python code and C# code so we can see the difference?

kazola, 2021-11-13:
These are vanilla codes from Visual Studio example and bleak. Simple connections.
Going to try putting some sleep(1) somewhere (maybe it's my firmware's fault) and get back here.
:+1:

hbldh, 2021-11-26:
@kazola Did it solve anything? Or is it Windows Bleak specific?

kazola, 2021-11-30:
Hi @hbldh 
Sorry for the delay in answering.
The connection under Windows is still way slower for my device.
However, I re-did a little bit of my code:

`           cli = BleakClient(mac)
            cli._mtu_size = MAX_MTU_SIZE
            if await cli.connect(timeout=10):
                await cli.start_notify(uuid_c, _nh)
`

to
```
            cli = BleakClient(mac, timeout=30)
            if await cli.connect():
                cli._mtu_size = MAX_MTU_SIZE
                await asyncio.sleep(1.1)
                await cli.start_notify(uuid_c, _nh)
```

And now, while still slow, I have way more successful connection attempts.
Am I making any sense?