hbldh/bleak #684: Notify require additional write before data is returned

takeitforwardsolutions, 2021-11-01:
* bleak version: bleak-0.13.0.dist-info
* Python version: 3.9.1
* Operating System:windows 10 64bit

### Description

I am sending commands via Bluetooth to move a device up and down via step-motor. We are measuring reach via a cable pulley and I need to capture the x,y when they lift pulley up and down. All that is working, we send commands over Bluetooth to the Arduino. Our machine developer has provided us commands to move device and request data capture(record data). The issue I am facing is that the data isn't returned immediately. 

### What I Did
Device connection I add notify:await self.client.start_notify("0000ffe1-0000-1000-8000-00805f9b34fb", self.notification_handler)
I send await current_client.write_gatt_char("0000ffe1-0000-1000-8000-00805f9b34fb", data=bytes_to_send) to enable data recording.
I move pulley up and down. No data returned.
I send await current_client.write_gatt_char("0000ffe1-0000-1000-8000-00805f9b34fb", data=bytes_to_send)  again and data is returned.

I am expecting data to be real-time, returning data as I move pulley up and down after first write command. I confirmed via android emulator that it works, data will come back in real-time. For some reason from python project it requires additional write command and data is returned as one lump of data.

Any help would be appreciated.

takeitforwardsolutions, 2021-11-04:
Hi, wondering if anyone else has faced similiar issue?

dlech, 2021-11-04:
Can you share the Python/Bleak code and the Android code so that we can see the difference?

takeitforwardsolutions, 2021-11-08:
> 
> 
> Can you share the Python/Bleak code and the Android code so that we can see the difference?

Sure thing, let me check with our machine developer to ensure no concerns with sharing details.

dlech, 2022-10-03:
Closing due to lack of feedback (although this sounds like a misconfigured gatt server similar to #972).

closed 2022-10-03 by dlech