hbldh/bleak #686: `BleakClientWinRT.get_services` takes longer time in Windows 11 than in Windows 10

hbldh, 2021-11-06:
* bleak version: 0.14.0a1 (`feature/wtd` branch)
* Python version: 3.7.8
* Operating System: Windows 11 ('10', '10.0.22000', 'SP0', 'Multiprocessor Free')

### Description

`get_services` call take a lot longer time in Windows 11 than in Windows 10.

### What I Did

A near 10 second connection delay in Windows 11 comparing to 1-5 seconds in Windows 10 was surprising to me. This issue is more of a mental note than an error report.  Based on memory, it was faster than 10 seconds earlier.

If anyone else using Bleak and noticing the same, please verify that it takes longer time in Windows 11 if you have experienced the same, or refute it if you have experienced the opposite.

```
import os

os.environ["BLEAK_LOGGING"] = "1"

import asyncio
from bleak import BleakClient, BleakScanner

address = "24:71:89:cc:09:05"
MODEL_NBR_UUID = "00002a24-0000-1000-8000-00805f9b34fb"

async def run(address):
    device = await BleakScanner.find_device_by_address(address)
    async with BleakClient(device) as client:
        model_number = await client.read_gatt_char(MODEL_NBR_UUID)
        print("Model Number: {0}".format("".join(map(chr, model_number))))

asyncio.run(run(address))
```

Output:

```powershell
2021-11-06 21:54:40,931 bleak.backends.winrt.scanner DEBUG: Received 24:71:89:CC:09:05: Unknown.
2021-11-06 21:54:40,933 bleak.backends.winrt.scanner DEBUG: 2 devices found. Watcher status: 3.
2021-11-06 21:54:40,933 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ 24:71:89:CC:09:05
2021-11-06 21:54:41,551 bleak.backends.winrt.client DEBUG: _ConnectionStatusChanged_Handler: 1
2021-11-06 21:54:41,551 bleak.backends.winrt.client DEBUG: Get Services...
2021-11-06 21:54:53,966 bleak.backends.winrt.client INFO: Services resolved for BleakClientWinRT (24:71:89:CC:09:05)
2021-11-06 21:54:54,085 bleak.backends.winrt.client DEBUG: Read Characteristic 000C : bytearray(b'CC2650 SensorTag')
Model Number: CC2650 SensorTag
2021-11-06 21:54:54,086 bleak.backends.winrt.client DEBUG: Disconnecting from BLE device...
2021-11-06 21:54:57,205 bleak.backends.winrt.client DEBUG: _ConnectionStatusChanged_Handler: 0
```

takeitforwardsolutions, 2021-11-08:
I'll spin up windows 11 and give a try.

dlech, 2021-12-02:
Discovering services takes only one second for me on Windows 11. 

```
2021-12-02 16:53:56,187 bleak.backends.winrt.client DEBUG: Get Services...
2021-12-02 16:53:57,127 bleak.backends.winrt.client INFO: Services resolved for BleakClientWinRT (F0:F8:F2:48:90:50)
```

commit 3576cacd (2021-12-07): Adding use_cached_services keyword to WinRT backend

This allows Bleak to use the Windows Bluetooth cache for services, characteristics and descriptors. Since Windows 11 took 10 seconds to fetch services on each connect, this will provide a faster connect on devices that do not change.

Fixes #686

Also modified Characteristic class in WinRT to use description property from super class in case of empty string description from Windows object.

The SensorTag example is a bit expanded to also light up a LED.
PR #695 (2022-07-25): Use Cached Services option for WinRT backend
This allows Bleak to use the Windows Bluetooth cache for services, characteristics and descriptors. Since Windows 11 took 10 seconds to fetch services on each connect, this will provide a faster connect on devices that do not change.

Fixes #686

Also modified Characteristic class in WinRT to use description property from super class in case of empty string description from Windows object.

The SensorTag example is a bit expanded to also light up a LED.
commit a5b98c7a (2022-07-29): v0.15.0

Added
-----

* Added new ``assigned_numbers`` module and ``AdvertisementDataType`` enum.
* Added new ``bluez`` kwarg to ``BleakScanner`` in BlueZ backend.
* Added support for passive scanning in the BlueZ backend. Fixes #606.
* Added option to use cached services, characteristics and descriptors in WinRT backend. Fixes #686.
* Added ``PendingDeprecationWarning`` to use of ``address_type`` as keyword argument. It will be moved into the
  ``winrt`` keyword instead according to #623.
* Added better error message when adapter is not present in BlueZ backend. Fixes #889.

Changed
-------

 [... cut]
PR #911 (2022-07-29): Release/v0.15.0

Added
-----

* Added new ``assigned_numbers`` module and ``AdvertisementDataType`` enum.
* Added new ``bluez`` kwarg to ``BleakScanner`` in BlueZ backend.
* Added support for passive scanning in the BlueZ backend. Fixes #606.
* Added option to use cached services, characteristics and descriptors in WinRT backend. Fixes #686.
* Added ``PendingDeprecationWarning`` to use of ``address_type`` as keyword argument. It will be moved into the
  ``winrt`` keyword instead according to #623.
* Added better error message when adapter is not present in BlueZ backend. Fixes #889.

Changed
- [... cut]
closed 2022-07-25 by dlech