hbldh/bleak #688: BleakClient: Timeout issue observed with BleakClient function

RohanDicholkar, 2021-11-15:
### Discussed in https://github.com/hbldh/bleak/discussions/687

<div type='discussions-op-text'>

<sup>Originally posted by **RohanDicholkar** November 15, 2021</sup>
I am trying to run a example code given on github to discover BLE devices and then get connected to my device. My sample code is as below. Performance of the code is very unstable. Out of about 10 tries, I am able to connect to my device only once for remaining 9 times I get timeout error. Timeout error is also given below, Can someone please suggest if any change is required in the program below. Is there any already known issue for this function

```python
"""
Connect by BLEDevice
"""

import asyncio
import platform
import sys

from bleak import BleakClient, BleakScanner
from bleak.exc import BleakError

ADDRESS = (
    "6E:01:8F:08:37:71"
    if platform.system() != "Darwin"
    else "B9EA5233-37EF-4DD6-87A8-2A875E821C46"
)

async def main(ble_address: str):
        device = await BleakScanner.find_device_by_address(ble_address, timeout=5.0)
        if not device:
            raise BleakError(f"A device with address {ble_address} could not be found.")
        async with BleakClient(device) as client:
            svcs = await client.get_services()
            print("Services:")
            for service in svcs:
                print(service)

if __name__ == "__main__":
    asyncio.run(main(sys.argv[1] if len(sys.argv) == 2 else ADDRESS))
```

**Timeout error**
```
==================================================== RESTART: C:\Bluetooth\bleak-develop\bleak-develop\examples\connect_by_bledevice.py ====================================================
Traceback (most recent call last):
  File "C:\Python39\lib\asyncio\locks.py", line 226, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Python39\lib\asyncio\tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Bluetooth\bleak-develop\bleak-develop\examples\connect_by_bledevice.py", line 32, in <module>
    asyncio.run(main(sys.argv[1] if len(sys.argv) == 2 else ADDRESS))
  File "C:\Python39\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "C:\Bluetooth\bleak-develop\bleak-develop\examples\connect_by_bledevice.py", line 24, in main
    async with BleakClient(device) as client:
  File "C:\Python39\lib\site-packages\bleak\backends\client.py", line 61, in __aenter__
    await self.connect()
  File "C:\Python39\lib\site-packages\bleak\backends\winrt\client.py", line 249, in connect
    await asyncio.wait_for(event.wait(), timeout=timeout)
  File "C:\Python39\lib\asyncio\tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```</div>

hbldh, 2021-11-26:
Please fill out this information first:

* bleak version:
* Python version:
* Operating System:
* BlueZ version (`bluetoothctl -v`) in case of Linux:

kaedenbrinkman, 2021-12-01:
Similar issue here. I am using the same code as @RohanDicholkar and it finds the device every time but freezes on trying to connect (`async with BleakClient(device) as client:`). It appears that `BleakScanner` works fine but not `BleakClient`. No errors are thrown.

Bleak version:  0.13.0
Python version: 3.9.9
OS: Windows 11

RohanDicholkar, 2021-12-01:
@kaedenbrinkman - My issue got resolved by following below steps. Please try,

https://github.com/hbldh/bleak/issues/604#issuecomment-892691565

kaedenbrinkman, 2021-12-03:
It looks like my issue was simply caused by the fact that Bleak does not yet support pairing devices. After pairing to the device in another app, Bleak could connect just fine.

zapta, 2022-04-30:
I am also hitting this timeout. Any known workaround?

kaedenbrinkman, 2022-04-30:
I can't comment on this issue, but I should say I ended up switching to https://github.com/OpenBluetoothToolbox/python_simpleble for my Python project and eliminated all my pairing troubles.

zapta, 2022-04-30:
Thanks kaedenbrinkman, I will take a look. Would definitely appreciate a simpler API without the heavy relying on asyncio.

As for the issue here, I found that if I pair the device with Windows, then the Bleak connection works consistently well. I had to change the code of my device to enable authentication to make the Windows paring process happy. Otherwise it failed with an error, though Nordic BLE app on the iPhone had no problem connecting to it.

As far as this issue is concerned, I am OK.

dlech, 2022-07-25:
This seems like a duplicate of #604, so if there are additional comments, please post on that issue.

closed 2022-07-25 by dlech