hbldh/bleak #689: HRESULT:0x800704C7

Lance-Tin, 2021-11-20:
* bleak version: 0.12.1
* Python version: 3.8.8
* Operating System: Windows11,22000.318
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I am trying to scan the BLE devices and connect them. than raising a bug like this: (Before I update the system, I never had such a bug.)
'''
System.ObjectDisposedException: 该对象已关闭。 (异常来自 HRESULT:0x80000013)
System.Exception: 操作已被用户取消。 (异常来自 HRESULT:0x800704C7)
System.Exception: 操作已被用户取消。 (异常来自 HRESULT:0x800704C7)
System.ObjectDisposedException: 该对象已关闭。 (异常来自 HRESULT:0x80000013)
System.Exception: 操作已被用户取消。 (异常来自 HRESULT:0x800704C7)
'''

### What I Did

You can go to the [Lance_Tin](https://github.com/Lance-Tin/Synchronization-Tool-for-Xsens-Dot.git) to down the code which i used.

[image]

Lance-Tin, 2021-11-20:
I have solved this problem by returning Windows11 to 10, But if someone wants to use windows 11 to do some works, it's a bug to solve in the future.

dlech, 2021-11-20:
> * bleak version: 0.12.1

Did you try Bleak v0.13.0 on Windows 11? We have dropped support for the dotnet backend.

hbldh, 2021-11-26:
Closing this, since it uses an old version of Bleak and the now deprecated `dotnet` backend instead of the `winrt`. I believe the `BleakBridge` that was used in the `dotnet` backend did not play well with Windows 11 in this case. It did use the oldest UWP Bluetooth available in Windows 10 and it is very likely that this produced problems in 11.

closed 2021-11-26 by hbldh