hbldh/bleak #690: No disconnect event when interface being removed

aryeko, 2021-11-22:
* bleak version: 0.13.0
* Python version: 3.7.3 and 3.9.8
* Operating System: raspian
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.58

### Description

I'm connecting to an Android device, after few seconds (up to 1 minute) I get a disconnection on my Android device but not on my python app (using bleak). 

### What I Did

1. Started BLE scanning with BleakScanner 
2. Started advertising on my Android device (see mac address 76:2D:35:40:73:C7 in the attached log)
3. BleakScanner found my Android device and connected successfully (see time in the log 2021-11-22 09:11:10.550)
4. At 2021-11-22 09:11:42,537 (attached log time) Bleak received "InterfaceRemoved" event without disconnection event  

The Android device got a disconnection event with reason "Peer disconnected".

Bleak didn't called disconnect.

I called disconnect from app side at 2021-11-22 09:11:56.127 (attached log time) and Bleak raised an exception that the device is not connected.

The property `client.is_connected` returns `True` before calling disconnect.

[log.log](https://github.com/hbldh/bleak/files/7580359/log.log)

dlech, 2021-11-22:
> can you fire a disconnect event for my case? The device was disconnected after all.

Yes, that should be working in Bleak already. I've tested it many times using the disconnection example.

aryeko, 2021-11-22:
I don’t get a disconnection event from Bleak - this is the reason I submitted the issue

dlech, 2021-11-22:
Can you supply the code to reproduce the problem?

aryeko, 2021-11-22:
well it is just a UART peripheral using your code example for Bleak as central: [uart_service](https://github.com/hbldh/bleak/blob/develop/examples/uart_service.py)

For the peripheral I'm using Android app - for example [nRF connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en&gl=US).

Note that the issue is not reproducing on all the phones, so far I reproduced it with high reproduction rate using Samsung Galaxy S8 or Huawei P40.

Isn't the attached log describe the issue enough? I pinned out the important timesamps in it as I see it.

dlech, 2022-09-03:
The log starts with:

>     2021-11-22 09:10:51.741 | INFO     | __main__:main:114 - started main - version=0.0.50

Which is clearly not from the uart_service.py example. Please provide a minimal reproducible test case and reopen the issue if this is still a problem.

closed 2022-09-03 by dlech