hbldh/bleak #691: Timeouts for read_gatt_char/write_gatt_char

cflavio, 2021-11-25:
* bleak version: 0.12
* Python version: 3.8
* Operating System: Ubuntu 20.04
* BlueZ version: 5.52

### Description

Hi, I have observed that sometimes I have some invocations of read_gatt_char/write_gatt_char that never return. So, my application freezes. Is it the expected behavior? Is there a best practice for adding a timeout to them (so, if they don't return then something like an exception is raised and the control returns to the caller)?
Thank you very much!

hbldh, 2021-11-26:
There is no timeout capability in the `dbus-next` library that Bleak uses, but one could possibly use a `await asyncio.wait_for(self._bus.call(...), timeout=5.0)` (See [Python docs](https://docs.python.org/3/library/asyncio-task.html#asyncio.wait_for)) to stop waiting for a reply. I have no idea if this messes up the `MessageBus` that represents Bleak's connection to the DBus system which in turn communicates with BlueZ which in turn communicates with your device...

You can try it yourself by modifying in [`read_gatt_char`](https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/client.py#L631) and You can try it yourself by modifying in [`write_gatt_char`](https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/client.py#L787). You should raise a `BleakError` if timeout happens and catch that in you application, instead of just returning `None`, though.

Try it and see if it works. Feel free to make a PR if you get it into a state that you are satified with, and we will review it!

dlech, 2023-07-19:
There is also now `asyncio.timeout()` which works better than `asyncio.wait_for()`.