hbldh/bleak #694: BluezDBus: infinite loop on connect+scan in parallel task

devbis, 2021-11-30:
* bleak version: 0.13.0
* Python version: 3.9
* Operating System: Linux, OpenWrt
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.56

### Description

I'm running two separate tasks: one is starting scanning and the second one is trying to connect to a device.
The problem is that despite of timeout device is not connecting and the code doesn't return from `connect()` coroutine.

### What I Did

The first iteration: 
I ran the `client.connect()` with wait_for timeout and it fails in the first run and succeeds in other tries.

The second iteration: 
I put debug `prints` in the bleak backend code and found the code that `hangs`.
The problem is inside this code: https://github.com/hbldh/bleak/blob/v0.13.0/bleak/backends/bluezdbus/scanner.py#L148-L155

Here is the output (from both tasks):

```
-- call bus GetManagedObjects
-- call bus GetManagedObjects
-- call bus SetDiscoveryFilter
-- call bus SetDiscoveryFilter
-- call bus StartDiscovery
-- call bus StartDiscovery
-- StartDiscovery done

(only a single print after StartDiscovery)
```

Replacing the mentioned code to the following worked as well

```py
        reply = await asyncio.wait_for(
          self._bus.call(
            Message(
                destination=defs.BLUEZ_SERVICE,
                path=self._adapter_path,
                interface=defs.ADAPTER_INTERFACE,
                member="StartDiscovery",
            )
          ), 
          timeout=3,
        )
```

I suppose somehow bluez responses to the first StartDiscovery only. 
I guess a timeout can solve freezing if there is no more elegant solution.

dlech, 2021-12-06:
Same as #361?

devbis, 2021-12-06:
It may relate. I need to check simultaneous connections to a couple of devices.

dlech, 2023-07-19:
Closing since the BlueZ backend has changed significantly since this was reported.

closed 2023-07-19 by dlech