hbldh/bleak #697: Connections to Lego Technic Hub fail on macOS 12

lobodpav, 2021-12-06:
* bleak version: https://github.com/hbldh/bleak/archive/refs/heads/macos12-scanning.zip
* Python version: 3.9
* Operating System: macOS 12.0.1 Monterey
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

When using the #635 `macos12-scanning.zip` fix to connect to my Lego technic Hub on macOS 12, it works nicely and I'm able to obtain an instance of my `BLEDevice`.

Connecting to the device fails, though.

### What I Did

```python
import asyncio
from bleak import BleakClient
from bleak import BleakScanner

device_uuid = "00001623-1212-EFDE-1623-785FEABCD123"

async def main():
    scanner = BleakScanner(service_uuids=[device_uuid])
    devices = []

    while len(devices) == 0:
        devices = await scanner.discover(service_uuids=[device_uuid])

        print("Devices: ", devices)
        if len(devices) > 0:
            device = devices[0]
            print("Device details: ", device.details)

            client = BleakClient(address_or_ble_device=device)
            conn = await client.connect()
            print("Connected: ", conn)

asyncio.run(main())
```

```
Devices:  [37C6EE65-C04E-E15F-23E5-6BA7283DAD3A: Technic Hub]
Device details:  <CBPeripheral: 0x6000003b0000, identifier = 37C6EE65-C04E-E15F-23E5-6BA7283DAD3A, name = Technic Hub, mtu = 0, state = disconnected>
2021-12-06 00:13:45.489 Python[67881:7734022] -[CBCharacteristic handleCharacteristicsDiscovered:]: unrecognized selector sent to instance 0x6000010a85a0
2021-12-06 00:13:45.489 Python[67881:7734022] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[CBCharacteristic handleCharacteristicsDiscovered:]: unrecognized selector sent to instance 0x6000010a85a0'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007ff80f442e5b __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007ff80f1a3b9d objc_exception_throw + 48
	2   CoreFoundation                      0x00007ff80f4c7a2b -[NSObject(NSObject) __retain_OA] + 0
	3   CoreFoundation                      0x00007ff80f3a9d09 ___forwarding___ + 1406
	4   CoreFoundation                      0x00007ff80f3a96f8 _CF_forwarding_prep_0 + 120
	5   CoreBluetooth                       0x00007ff8210e3990 -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 144
	6   CoreBluetooth                       0x00007ff8210e3a81 -[CBPeripheral handleServiceEvent:serviceSelector:delegateSelector:delegateFlag:] + 113
	7   CoreBluetooth                       0x00007ff8210dfdd2 -[CBPeripheral handleMsg:args:] + 466
	8   CoreBluetooth                       0x00007ff8210ccd57 -[CBCentralManager handleMsg:args:] + 183
	9   CoreBluetooth                       0x00007ff821108141 -[CBManager xpcConnectionDidReceiveMsg:args:] + 177
	10  CoreBluetooth                       0x00007ff8210f8142 __30-[CBXpcConnection _handleMsg:]_block_invoke + 66
	11  libdispatch.dylib                   0x00007ff80f148ad8 _dispatch_call_block_and_release + 12
	12  libdispatch.dylib                   0x00007ff80f149cc9 _dispatch_client_callout + 8
	13  libdispatch.dylib                   0x00007ff80f14fcee _dispatch_lane_serial_drain + 696
	14  libdispatch.dylib                   0x00007ff80f1507fb _dispatch_lane_invoke + 417
	15  libdispatch.dylib                   0x00007ff80f14fb85 _dispatch_lane_serial_drain + 335
	16  libdispatch.dylib                   0x00007ff80f1507c8 _dispatch_lane_invoke + 366
	17  libdispatch.dylib                   0x00007ff80f15a7e1 _dispatch_workloop_worker_thread + 758
	18  libsystem_pthread.dylib             0x00007ff80f2fd08f _pthread_wqthread + 326
	19  libsystem_pthread.dylib             0x00007ff80f2fc01b start_wqthread + 15
)
libc++abi: terminating with uncaught exception of type NSException
```

### What works

When running the same code to connect t my Bluetooth headset, I'm able to obtain a connection.

```
Devices:  [D1D552D6-4AC1-924B-B39A-7E44957D1C30: LE-Pavel Bose QC35 II]
Device details:  <CBPeripheral: 0x600002fec0b0, identifier = D1D552D6-4AC1-924B-B39A-7E44957D1C30, name = LE-Pavel Bose QC35 II, mtu = 0, state = disconnected>
Connected:  True
```

dlech, 2021-12-28:
The example program works for me (macOS 12.1, 2021 MacBook Pro, M1). Maybe Apple fixed the crash?

lobodpav, 2021-12-28:
I have tried the same program today on the same macOS 12.0.1 version and it worked.

The only difference I've spotted is my Python got upgraded from 3.9.8 to 3.9.9. The release notes do not mention any BLE changes so now I'm a little confused 😞.

dlech, 2021-12-28:
The crash is in CoreBluetooth and there isn't any Python or Bleak functions in the backtrace. Maybe rebooting the computer fixed it? Anyhow, nothing we can fix in Bleak anyway, so I'll go ahead an close the issue and hope it keeps working for you.

closed 2021-12-28 by dlech